openpnp/openpnp #475: Altium pick and place board import

Misfittech, 2017-03-22:
Currently the Altium pick and place board import uses the Reference X and Y location.  In Altium the user can select the reference location for each part. For example some people use pin #1 of a part as the reference location.   It appears that by default OpenPnP would like the center of the part, not the reference point.  Altium does/can export the center of the part and thus it is recommend that the Altium importer try to use the Center location by default and the Reference point if Center is unavailable.

issus, 2017-03-22:
The importer should use the reference point. Your altium library should
have the reference point be the pick location for the part - this is
important for odd packages and connectors where the pick point is not the
part centre.

My altium library is setup like this if you need a reference:
https://github.com/issus/altium-library

cri-s, 2017-03-22:
This is example header line from AD:
"Designator","Footprint","Mid X","Mid Y","Ref X","Ref Y","Pad X","Pad Y","Layer","Rotation","Comment"
It is correct behavior to use Ref X as coordinate, Pad X is pin 1 coordinate, on openpnp unused. It can be used to correct rotation if needed.
If you need to use center coordinates because the ref coordinates are bad, change "Mid X" to "X" and then center coordinate are used. You need to do the same with Y.

robermeyer, 2017-03-22:
The center of the part is not always the correct location to PNP.  Depends
on part shape.

Pin 1 is almost never the right place.

Ref would always give the PNP the right place as defined when the component
was created.  Assuming the part was created correctly.  Since Altium
described those as "user-defined reference point' correctly could be in the
users perception of correct.

I use the same definition as Mark.

The location information is expressed in three ways, corresponding to the
three methods by which a component footprint's reference point can be
specified in the source PCB library:

   - *Mid X* , *Mid Y* - coordinates for the component's center point
   - *Ref X* , *Ref Y* - coordinates for the component's user-defined
   reference point
   - *Pad X* , *Pad Y *- coordinates for Pad 1 of the component.

closed 2017-03-22 by vonnieda