openpnp/openpnp #476: OpenCV TemplateMatch doesn't work with rotationed objects

sp-apertus, 2017-03-22:
In theory comparing components bottom vision image to a predefined reference template would be THE ideal method to recognize and orient any package of any shape or pin layout even with index pins or recognizing labels/prints orientation.

Unfortunately the current MatchTemplate OpenCV functions does not work when your object is rotated.

I googled around a bit and while there are very sophisticated approaches: OpenCV Moments or SIFT, FLANN, etc. the more I read the more I got the impression these are overkills. 
Someone just suggested rotating the reference image in a loop and check if the matching treshhold is peaking at any specific angle - might be worth a try....

Thoughts?

vonnieda, 2017-03-22:
Can you explain why this is needed? Is the current algorithm not working under some specific circumstances?

I prefer the current system because it requires no training for new parts.  That being said, I have considered alternate algorithms so that we could handle pin 1 finding, but in general that is only needed for loose part pickup since the likelihood of picking a part from any feeder with more than a 45 degree angle error is very low.

cri-s, 2017-03-22:
Do logpolar mapping before doing template match, othen the resulting position is rotation invariant. It return the position, not the angle of rotation.

sp-apertus, 2017-03-22:
cri-s please elaborate on the mentioned logpolar mapping procedure.

I am trying to create an OpenCV Pipeline using the MatchTemplate stage to find center and rotation of this chip:
[image]

in this demo bottom vision image:
[image]

vonnieda, 2017-03-23:
@sp-apertus These images look like they would work very well with the default pipeline. Is there a reason you are interested in using template matching instead?

cri-s, 2017-03-23:
In this case I probable use cog for knowing the rotation. However if
you want do template match, do standard recognition, unrotate it to 0
degree and then using 2 or 4 template match to determinate the
quadrant.

Code to convert rotation and scale to translation:

Mat Source=new Mat(source.rows(),source.cols(),CvType.CV_32FC1);
  Imgproc.logPolar(source,Source,new Point(Source.cols() /
2,Source.rows() / 2),0.1,Imgproc.CV_WARP_INVERSE_MAP +
Imgproc.CV_WARP_FILL_OUTLIERS);

sp-apertus, 2017-03-23:
> @sp-apertus These images look like they would work very well with the default pipeline. Is there a reason you are interested in using template matching instead?

Because I am interested in general in making the templace matching method useful :)

cri-s, 2017-03-23:
Maybe it's that what you search. http://www.lfd.uci.edu/%7Egohlke/code/imreg.py.html

cri-s, 2017-03-23:
Generally blobs (contours) are more flexible, if you have understand it, and faster.
Example of resistor location 
on bad lightning
[image]
If you have better lightning, as example this , example taken from manual,
[image]
Then one just match the metallic parts getting right and correct angle.
In both cases, templatematch for one reson or annother perform lesser.

cri-s, 2017-03-23:
back to orginal question. 

optional, do canny operator on images.
convert images to logpolar.
convert logpolar images to floating point.
Point pt = phaseCorrelate(Source,Template);
pt.X is scale, pt.Y is rotation, example (16.2986, 36.9105)
If you need to match multiple targets, it's the same as templatematch result with multiple targets.
Now you know rotation and scale of object, rotate template/source and scale it.
do templatematch to find out the position.

Scale = exp( pt.x / 40) = exp(16.2986 / 40) = 1.503 
Rotation = pt.y*360/(Template.cols()) = 44.29
In this example a logpolar magnitude scale parameter of 40 was used. 

ask if you have further questions.

vonnieda, 2017-03-26:
Closing this as it's not an issue, but a discussion. You are welcome to continue the discussion here if you like, or on the mailing list.

closed 2017-03-26 by vonnieda

yusuf8421, 2024-09-03:
you can use below code sample find if pic rotated
```
public bool MatchTemplateRotated(Mat image, Mat template, out Point matchLocation, out double foundAngle, out double foundScale, out Mat rtemplate, double threshold = 0.5)
{
    var angles = new[] { 0, 45, 90, 135, 180,270,360 }; // Kullanmak istediğiniz açılar
    var scales = new[] { 0.5, 1.0, 1.5 }; // Kullanmak istediğiniz ölçekler
    //var angles = Enumerable.Range(-90, 181).ToArray(); // -90 ile 90 derece arasında 1 derecelik artışlarla
    //var scales = Enumerable.Range(75, 51).Select(x => (double)x / 100).ToArray(); // 0.75 ile 1.25 arasında 0.01'lik artışlarla
    foreach (var angle in angles)
    {
        foreach (var scale in scales)
        {
            var rotatedTemplate = RotateAndScaleTemplate(template, angle, scale);
            var result = new Mat();
            Cv2.MatchTemplate(image, rotatedTemplate, result, TemplateMatchModes.CCoeffNormed);

            double minVal, maxVal;
            Point minLoc, maxLoc;
            Cv2.MinMaxLoc(result, out minVal, out maxVal, out minLoc, out maxLoc);

            if (maxVal > threshold)
            {
                matchLocation = maxLoc;
                foundAngle = angle;
                foundScale = scale;
                rtemplate = rotatedTemplate;
                return true;
            }
        }
    }
    matchLocation = new Point(0, 0);
    foundAngle = 0;
    foundScale = 1.0;
    rtemplate = new Mat();
    return false;
}
```
```