openpnp/openpnp #478: No defined transitions from Preflight to Initialize.

vonnieda, 2017-03-23:
Discussion in https://groups.google.com/d/msgid/openpnp/62323514-27c4-4c49-a527-371046c7fa6d%40googlegroups.com?utm_medium=email&utm_source=footer

This error (and others like it) have been reported several times since adding the FSM. I have not been able to reliably reproduce it. Need to see the info in that thread and see if it can be reproduced.

vonnieda, 2017-03-24:
I tried to reproduce this in a variety of ways and can't make it happen. If anyone can come up with a reproducible version of this it would really help me a lot.

tracklogic, 2017-03-24:
I've seen this a lot but i've never noted the steps ... all i know is if the job has failed in the past then you can't seem to start a new job as you get this message. Next time i see it i'll try and make a note of exactly whats gone on.

I do think one time is when i had a nullptr exception in some custom code i wrote (feeder class)... so job processor was trying to do a feed on feeder... nullptr happened... clicked ok.. tried to restart the job and then got the preflight error... maybe its something to do if an exception happens mid-job?

vonnieda, 2017-03-24:
That's how I was trying to reproduce it. I configured a strip feeder to be out of parts, which causes an exception on feed. I ran a job which failed on the first feed but after that I was still able to pause, start, restart, etc.

trampas, 2017-03-24:
I just reproduced this bug, well a similar error of " No Defined
transitions from the Align for Initialize"

The bottom camera is black, and I did a placement as the placement showed
the vision holes for the strip feeder I clicked stop.  The machine was
still running trying to use up camera for vision I hit the start button for
the job again. The vision for the up camera failed and ask if I wanted to
pause job. I clicked to pause job. Then I click run job and got the error.

In the eclipse console :

Mar 23, 2017 3:40:07 PM java.util.prefs.WindowsPreferences <init>
WARNING: Could not open/create prefs root node Software\JavaSoft\Prefs at
root 0x80000002. Windows RegCreateKeyEx(...) returned error code 5.
Don't call wizardCompleted
Don't call wizardCompleted
OpenCV Error: Assertion failed (points.checkVector(2) >= 0 &&
(points.depth() == CV_32F || points.depth() == CV_32S)) in cv::minA

Not sure if this is helpful, if you need to remote into the machine and try
yourself I am happy to provide TeamViewer for you.

Trampas

vonnieda, 2017-03-24:
Thanks @trampas, that might be enough for me to figure it out. It sounds like there is a state mismatch between the JobPanel and the JobProcessor. Sounds like the JobProcessor has not yet finished it's cleanup process and the JobPanel is allowing a new start command to be sent. I'll test this more tonight and see if I can get it reproduced. Thanks!

vonnieda, 2017-03-26:
Reproduced this in the test setup by starting the pnp-test job and immediately hitting stop and then start. The job is still finishing the fid checks in the background (which is annoying and related to #278) and we issue a new start. I think the fix is to disable the job controls until the job is actually stopped.

vonnieda, 2017-04-23:
Okay, the root of the problem here is that this code is a disaster.

When you hit one of the job action buttons such as start, stop or step, this sends a message to the JobPanel FSM which calls one of jobAbort(), jobRun(), jobStart(), etc. These in turn do a machine submit with a call to the JobProcessor.

The problem is that since the call to the JobProcessor gets queued (via the submit) the state of the JobPanel FSM updates immediately, which allows the buttons to be pressed again, but the actual JobProcessor call has not been executed yet.

Investigating whether we can submit the tasks as a Future and then wait on it, effectively making the buttons synchronous. This will lock the UI until the task is complete, but that's better than screwing up all the state.

This whole mess will be rewritten without the FSM in the near future.

vonnieda, 2017-04-24:
So I don't lose this in notes; after a long motorcycle ride and lots of thinking, here is how this will be refactored:

JobPanel FSM replaced with a simple state enum:
* Running
* Paused
* Stopped
* Pausing
* Stopping

A step method will be added which runs a single step and handles error display.
The Start/Run job action runs a machine task in a loop, checking state == Running, calling step.
The Step action runs a machine task, no loop, calling step.

Clicking Step, Pause or Stop will change the state to the *ing version (Pausing, Stopping) and allow the current step to finish, then will change to Paused, Stopped. Whenever the state is *ing then all buttons will be disabled until the state changes.

trampas, 2017-04-24:
Might be easier to have current state and "next state".

ldpgh, 2018-10-27:
Just FYI ... some more context.

Got message too while trying a 2-device job for testing.
1) Pressing start button ... but nothing happened.
2) Pressed stop
3) Pressed start again ... found the error message. 

OpenPnP log-window has been cleared shortly before :
```
2018-10-27 17:34:56 SystemLogger INFO: isAllPlaced false
2018-10-27 17:34:56 MessageBoxes DEBUG: Error: java.lang.Exception: No defined transitions from PreFlight for Initialize
2018-10-27 17:35:15 SystemLogger INFO: isAllPlaced false
2018-10-27 17:35:15 MessageBoxes DEBUG: Error: java.lang.Exception: No defined transitions from PreFlight for Initialize
2018-10-27 17:45:55 SystemLogger INFO: isAllPlaced false
2018-10-27 17:45:55 MessageBoxes DEBUG: Error: java.lang.Exception: No defined transitions from PreFlight for Initialize
```

Eclipse console window:
```
...
2018-10-27 17:33:41 ReferenceHead DEBUG: H1.moveToSafeZ(0.74)
2018-10-27 17:33:41 ReferenceNozzle DEBUG: N1.moveToSafeZ(0.74)
2018-10-27 17:33:41 ReferenceCamera DEBUG: OpenPnpCaptureCameraDown.moveToSafeZ(0.74)
2018-10-27 17:33:41 ReferenceActuator DEBUG: ZZZ_ReferenceActuator_Dummy_01.moveToSafeZ(0.74)
2018-10-27 17:33:41 ReferenceActuator DEBUG: ZZZ_ReferenceActuator_Dummy_02.moveToSafeZ(0.74)
2018-10-27 17:33:41 ReferencePnpJobProcessor INFO: Job finished 2 parts in 43.2 sec. This is 166.6 pph
2018-10-27 17:33:41 Scripting TRACE: Scripting.on Job.Finished
2018-10-27 17:33:41 FiniteStateMachine TRACE: Next => Cleanup -> Stopped
2018-10-27 17:33:41 FiniteStateMachine TRACE: Reset => Stopped -> Uninitialized
2018-10-27 17:33:42 FiniteStateMachine TRACE: Finished => Running -> Stopped
2018-10-27 17:33:45 ReferenceHead DEBUG: H1.moveToSafeZ(0.74)
2018-10-27 17:33:45 ReferenceNozzle DEBUG: N1.moveToSafeZ(0.74)
2018-10-27 17:33:45 ReferenceCamera DEBUG: OpenPnpCaptureCameraDown.moveToSafeZ(0.74)
2018-10-27 17:33:45 ReferenceActuator DEBUG: ZZZ_ReferenceActuator_Dummy_01.moveToSafeZ(0.74)
2018-10-27 17:33:45 ReferenceActuator DEBUG: ZZZ_ReferenceActuator_Dummy_02.moveToSafeZ(0.74)
2018-10-27 17:33:45 ReferenceCamera DEBUG: moveTo((61.482349, 132.194991, NaN, -0.479163 mm), 0.74)
2018-10-27 17:33:45 GcodeDriver DEBUG: sendCommand(moveto X61.4823 Y132.1950   F740, 5000)...
2018-10-27 17:33:45 GcodeDriver TRACE: [serial://COM220] >> moveto X61.4823 Y132.1950   F740
2018-10-27 17:33:46 GcodeDriver TRACE: [serial://COM220] << ok
2018-10-27 17:33:46 GcodeDriver DEBUG: sendCommand(serial://COM220 moveto X61.4823 Y132.1950   F740, 5000) => [ok]
2018-10-27 17:33:46 ReferenceCamera DEBUG: moveTo((61.482349, 132.194991, 0.000000, -0.479163 mm), 0.74)
2018-10-27 17:34:18 SystemLogger INFO: isAllPlaced false

2018-10-27 17:34:18 FiniteStateMachine TRACE: Initialize => Uninitialized -> PreFlight
2018-10-27 17:34:18 FiniteStateMachine TRACE: StartOrPause => Stopped -> Running
2018-10-27 17:34:30 FiniteStateMachine TRACE: Abort => Running -> Stopped
2018-10-27 17:34:56 SystemLogger INFO: isAllPlaced false

2018-10-27 17:34:56 MessageBoxes DEBUG: Error: java.lang.Exception: No defined transitions from PreFlight for Initialize
2018-10-27 17:35:15 SystemLogger INFO: isAllPlaced false

2018-10-27 17:35:15 MessageBoxes DEBUG: Error: java.lang.Exception: No defined transitions from PreFlight for Initialize
2018-10-27 17:45:55 SystemLogger INFO: isAllPlaced false

2018-10-27 17:45:55 MessageBoxes DEBUG: Error: java.lang.Exception: No defined transitions from PreFlight for Initialize

2018-10-27 17:33:46 GcodeDriver TRACE: [serial://COM220] << ok
2018-10-27 17:33:46 GcodeDriver DEBUG: sendCommand(serial://COM220 moveto X61.4823 Y132.1950   F740, 5000) => [ok]
2018-10-27 17:33:46 ReferenceCamera DEBUG: moveTo((61.482349, 132.194991, 0.000000, -0.479163 mm), 0.74)
2018-10-27 17:34:18 SystemLogger INFO: isAllPlaced false
2018-10-27 17:34:18 FiniteStateMachine TRACE: Initialize => Uninitialized -> PreFlight
2018-10-27 17:34:18 FiniteStateMachine TRACE: StartOrPause => Stopped -> Running
2018-10-27 17:34:30 FiniteStateMachine TRACE: Abort => Running -> Stopped
2018-10-27 17:34:56 SystemLogger INFO: isAllPlaced false
2018-10-27 17:34:56 MessageBoxes DEBUG: Error: java.lang.Exception: No defined transitions from PreFlight for Initialize
2018-10-27 17:35:15 SystemLogger INFO: isAllPlaced false
2018-10-27 17:35:15 MessageBoxes DEBUG: Error: java.lang.Exception: No defined transitions from PreFlight for Initialize
2018-10-27 17:45:55 SystemLogger INFO: isAllPlaced false
2018-10-27 17:45:55 MessageBoxes DEBUG: Error: java.lang.Exception: No defined transitions from PreFlight for Initialize
```

ldpgh, 2018-10-27:
Sending commands   `disable / enable / home`   works. Machine is doing a home-run.

But further activities (e.g press home-button) are ignored. No printouts to the log-file of OpenPnP. Will restart the GUI.

commit 545d9aa0 (2019-02-22): Update changes for recent changes.
Fixes #478.
Fixes #278.
commit c0b5b012 (2019-06-01): Merge branch 'develop' into test

* develop: (103 commits)
  Added a bare minimum pipeline example.
  Fixes override sounds not working for SoundSignaler. Added configuration wizard for SoundSignaler.
  Fixes #814. Home action's icon is now tied to a property which is bound to the machine's "homed" property. Added a few notes for future improvements in the same vein.
  Fix a minor issue in the test driver where getLocation did not apply head offsets. Fix some lint in the ReferenceHead which caused tests failures when the driver was updated at runtime.
  Improved documentation on new planner.
  [... cut]
closed 2019-02-22 by vonnieda