openpnp/openpnp #481: Add Contrast Limited Adaptive Histogram Equalization (CLAHE) to the opencv pipeline arsenal

dzach, 2017-03-24:
Lighting plays an important role in openpnp when computer vision is being used. In general, uniform lighting is desired, but it is not easily achievable. In particular, openpnp `Threshold` pipeline operation suffers greatly from non uniform light distribution, as shown below:

[image]

Trying to correct the obvious lighting problem by using histogram equalization results to this:

[image]

For a particular threshold value (here 12) this is the result from a 'Threshold' pipeline operation:

[image]

As expected, a large area at the periphery of the image is under-illuminated and this results to loss of information.

[image]

A better suited equalization algorithm would adapt equalization levels to the local content.

From [wikipedia](https://en.wikipedia.org/wiki/Adaptive_histogram_equalization) :

> Adaptive histogram equalization (AHE) is a computer image processing technique used to improve contrast in images. It differs from ordinary histogram equalization in the respect that the adaptive method computes several histograms, each corresponding to a distinct section of the image, and uses them to redistribute the lightness values of the image. It is therefore suitable for improving the local contrast and enhancing the definitions of edges in each region of an image.
> 
> However, AHE has a tendency to overamplify noise in relatively homogeneous regions of an image. A variant of adaptive histogram equalization called contrast limited adaptive histogram equalization (CLAHE) prevents this by limiting the amplification.

Here is the result of a CLAHE equalization (clip limit 32) on the original image:

[image]

and the resulting threshold operation (here done in Gimp). Thresholds where adjusted to provide about the same detail in IC pins outlines:

[image]

OpenCV code to produce the results, adapted from [an answer to a stackoverflow.com question](http://stackoverflow.com/questions/24341114/simple-illumination-correction-in-images-opencv-c) (credit goes to [user Bull](http://stackoverflow.com/users/2151446/bull)). It takes as arguments an image and a optional integer for `clip limit`:

`
#include <stdio.h>
#include <stdlib.h>
#include <opencv2/core/core.hpp>
#include <vector>       // std::vector
#include "opencv2/core/utility.hpp"
#include "opencv2/highgui.hpp"
#include <opencv2/imgproc/imgproc.hpp>

int main(int argc, char** argv)
{
    int clim = 4;
    if (argc = 3)
      clim = atoi(argv[2]);
    
  // READ RGB color image and convert it to Lab
    cv::Mat bgr_image = cv::imread(argv[1]);
    cv::Mat lab_image;
    cv::cvtColor(bgr_image, lab_image, CV_BGR2Lab);

    // Extract the L channel
    std::vector<cv::Mat> lab_planes(3);
    cv::split(lab_image, lab_planes);  // now we have the L image in lab_planes[0]

    // apply the CLAHE algorithm to the L channel
    cv::Ptr<cv::CLAHE> clahe = cv::createCLAHE();
    clahe->setClipLimit(clim);
    cv::Mat dst;
    clahe->apply(lab_planes[0], dst);

    // Merge the the color planes back into an Lab image
    dst.copyTo(lab_planes[0]);
    cv::merge(lab_planes, lab_image);

   // convert back to RGB
   cv::Mat image_clahe;
   cv::cvtColor(lab_image, image_clahe, CV_Lab2BGR);

   // display the results  (you might also want to see lab_planes[0] before and after).
   cv::imshow("image original", bgr_image);
   cv::imshow("image CLAHE", image_clahe);
   cv::imwrite("out.png",image_clahe);
   cv::waitKey(); 
}
`

Could something like this be implemented as a new stage in an OpenPnP OpenCV pipeline?

vonnieda, 2017-03-24:
Certainly. Adding stages is pretty easy. Here's the current Threshold stage: https://github.com/openpnp/openpnp/blob/develop/src/main/java/org/openpnp/vision/pipeline/stages/Threshold.java

You'd just make a new class in that package or just add it as an option to the existing Threshold one.

dzach, 2017-03-24:
Thanks @vonnieda ! Will try it and see what happens.

vonnieda, 2017-03-24:
Great! Note that if you add a whole new Stage you also need to add it to the list so it will show up in the GUI: https://github.com/openpnp/openpnp/blob/develop/src/main/java/org/openpnp/vision/pipeline/ui/CvPipelineEditor.java#L65

dzach, 2017-03-24:
Thanks again! You've made it so easy, that even an ignorant with the language, as I am, can be able to try something like this. Great stuff.

dzach, 2017-03-24:
How is `ContrastLimitedAdaptiveHistogramEqualization` for a class name?

I'm getting this error

```
[ERROR] COMPILATION ERROR : 
[INFO] -------------------------------------------------------------
[ERROR] [...]/openpnp/src/main/java/org/openpnp/vision/pipeline/stages/ContrastLimitedAdaptiveHistogramEqualization.java:[24,30] cannot find symbol
  symbol:   method createCLAHE()
  location: class org.opencv.imgproc.Imgproc
```

The docs say:
 ```
org.opencv.imgproc
Class CLAHE

java.lang.Object
  extended by org.opencv.core.Algorithm
      extended by org.opencv.imgproc.CLAHE

```

Does OpenPnP's OpenCV iclude class CLAHE? It has been introduced in OpenCV 3.2 recently.

Here is my `infinite monkey' code,  `ContrastLimitedAdaptiveHistogramEqualization.java` which I cannot yet get to compile:

```
package org.openpnp.vision.pipeline.stages;

import org.opencv.core.Mat;
import org.opencv.imgproc.Imgproc;
import org.opencv.imgproc.CLAHE;
import org.openpnp.vision.pipeline.CvPipeline;
import org.openpnp.vision.pipeline.CvStage;
import org.simpleframework.xml.Attribute;

public class ContrastLimitedAdaptiveHistogramEqualization extends CvStage {
    @Attribute
    private double clim = 4.0;

    public double getClipLimit() {
        return clim;
    }

    public void setClipLimit(double clim) {
        this.clim = clim;
    }
    
    @Override
    public Result process(CvPipeline pipeline) throws Exception {
        CLAHE clahe = Imgproc.createCLAHE();
        Mat mat = pipeline.getWorkingImage();
        clahe.setClipLimit(clim);
        clahe.apply(mat,mat);
        return null;
    }
}

```

vonnieda, 2017-03-25:
Ah, yes, unfortunately we're still using OpenCV 2.4 in OpenPnP. I have an upgraded codebase ready to go but the new version of OpenCV seems to be a lot more crashy so I have released it yet. Have been hoping they will release a new point release that addresses some of that. My apologies for that, I didn't realize CLAHE was that new.

dzach, 2017-03-25:
No problem, the code will be waiting for the OvenCV upgrade, whenever it happens.

cri-s, 2017-03-25:
Clahe was not that new, the java wrapper just forget to add createCLAHE because of pointer issue. There is no way to create the required CLAHE variable. Javacpp byteco have added a create method to algorithm class for resolve this general java problem with pointers. 
Clahe don't are useful in this type of application and adaptive threshold are more indicated. The problem with clahe is that it produce a lot of artifacts and noise that cannot be filtered out. It is a lot useful in other applications, not here.
It seems that you just experiment with a lot of wrong constraints or you setup need some improvements.
If you need example of adaptive threshold with this image, and clahe, I could supply that.
In case you decide to go further with clahe I could you supply the stage code, hovewer I see not any utility including it on official branch.

dzach, 2017-03-25:
@cri-s Thank you for the comment. By no means is my testing complete. Partial testing so far, by using GIMP for threshold operation on images acquired by ImageWriteDebug indicates that my setup might benefit from a CLAHE pipeline stage, however, without being able to test the CLAHE stage with OpenPnP I cannot really tell if it is any good.

I've seen the code for createCLAHE but didn't want to go that deep right now.

Coincidentally, I was just testing a new AdaptiveThreshold stage for OpenPnPto see if there are benefits in using it in place of Threshold. I was planning to issue a pull request later today. I'll create another issue and show the code there.

cri-s, 2017-03-25:
What i mean is that the lightning is far from usable. If you have it, that should be checked again.
Using you'r source:
[image]
this is the output of clahe
[image]
I have it not processed further, but from looking over image it seems to be not usable.
This is the output of adaptive threshold, mean with parameters 11 2
[image]
For image checking purpose for you i have cleaned it a bit applying	imclearborder and bwareaopen .
For purpose of recognizing the devices, it's not necessary as the filtering for valid contours remove it automatically. The result is this:
[image]
Just as note for your other issue, if you use gaussian instead of median then the resistors get not properly detected or the
center is shifted.

dzach, 2017-03-25:
@cri-s Thanks for looking into this. If you increase your block size in adaptive threshold the empty interiors get filled. You can also use simple Threshold and get filled shapes. As with everything, there are some settings that are not good and some that are better.

Lighting was at about 10/255 when taking the original image above, because at that moment, that was the best compromise with the stripfeeder detection. Now I have it at about 100/255, after changing some other settings.

Setting of my machine though is a different issue than the one I'm trying to tackle here. Here I was looking into ways to correct non-uniform lighting through software, instead of hardware. CLAHE seems to be doing it to a certain extent.

cri-s, 2017-03-25:
Non uniform lightning, shoot a reference image on white paper, invert
it, add it to image, normalize. You should do it on bigger image space
as CV_8U to avoid clipping.
Annother possibility is to use grain merge from gimp and emulate it on
opencv, always with the
same white paper reference image with gaussian blur applied.

Lighting on strip feeder should be that on threshold 70-80 the empty
pockets disappears.

dzach, 2017-03-25:
>Non uniform lightning, shoot a reference image on white paper, invert it, add it to image, normalize.

That's a good tip!

vonnieda, 2017-03-25:
Note: Enhancement blocked by release of OpenCV 3.2.

cri-s, 2017-05-16:
Hi, where tu put clahe java source code? Vision?

vonnieda, 2017-05-16:
@cri-s If it's an implementation of the algorithm, yes, please put it in the vision package. If it's a stage it should go in the stages package.

cri-s, 2017-05-16:
It is a implementation of CLAHE class on java, but on PipelineEditor
the time is 3500ms
in contrast with <5 ms the javacv code. I have seen this only after
have posted the question,
so i don't really know if it should be merged.

vonnieda, 2017-05-16:
@cri-s No, probably not. That's pretty bad.

rresol, 2017-06-29:
@dzach Why it is that when we increase our block size in adaptive gaussian the empty interiors get filled? As far as I understand gaussian threshold calculates destination pixel values by weighted average of the adjacent pixels. So, as adjacent pixels are black in the interiors the destination pixel values should be black irrespective of the block size.
Please correct me if I am missing something. It would be great if you can help me with understanding it.

dzach, 2017-06-29:
Hi @rresol 

> destination pixel values should be black irrespective of the block size.

I'm not an expert in OpenCv algorithms, but this is the way I understand it, in layman's terms:

Assuming a sliding window with a block of size 3, and an image with a single white pixel and all neighboring black, then the averaging window will cover the single white pixel for at least 3 pixel steps, in both the `mean` and `gaussian` methods. Gaussian is different in that the averaging 2D window is not a "square" block of '1's but rather a pointy "hill" of a gaussian curve. So, the larger the averaging window, the wider the outskirts will be, depending also on the cParm value.

markmaker, 2022-02-03:
> No problem, the code will be waiting for the OvenCV upgrade, whenever it happens.

@dzach I guess this is no longer blocked ;-)

tonyluken, 2022-02-03:
I believe this functionality is available in [HistogramEqualizeAdaptive](https://github.com/openpnp/openpnp/blob/develop/src/main/java/org/openpnp/vision/pipeline/stages/HistogramEqualizeAdaptive.java).  Recommend closing this issue