openpnp/openpnp #482: Add an Adaptive Threshold stage to the OpenCV pipeline arsenal

dzach, 2017-03-25:
From Wikipedia:

> The simplest thresholding methods replace each pixel in an image with a black pixel if the image intensity is less than some fixed constant T, or a white pixel if the image intensity is greater than that constant. 
[...]
To make thresholding completely automated, it is necessary for the computer to automatically select the threshold T.

> [thresholding methods include:]
> [...]
> - Local methods adapt the threshold value on each pixel to the local image characteristics. In these methods, a different T is selected for each pixel in the image.

Enter **Adaptive Thresholding**.

OpenCV 2.4 already includes an [adaptiveThreshold](http://docs.opencv.org/2.4/modules/imgproc/doc/miscellaneous_transformations.html) image transformation. Preliminary code for testing such an openpnp pipeline stage is like the following:

File _AdaptiveTrasform.java_:

```
package org.openpnp.vision.pipeline.stages;

import org.opencv.core.Mat;
import org.opencv.imgproc.Imgproc;
import org.openpnp.vision.pipeline.CvPipeline;
import org.openpnp.vision.pipeline.CvStage;
import org.openpnp.vision.pipeline.Property;
import org.openpnp.vision.pipeline.Stage;
import org.simpleframework.xml.Attribute;
import org.simpleframework.xml.Root;

@Stage(category="Image Processing", description="Performs adaptive thresholding on the working image.")
public class AdaptiveThreshold extends CvStage {

    private static final int ADAPTIVE_THRESH_MEAN_C = 0;
    private static final int ADAPTIVE_THRESH_GAUSSIAN_C = 1;
    private static final int CV_THRESH_BINARY = 0;
    private static final int CV_THRESH_BINARY_INV = 1;

    @Attribute
    @Property(description="Size of a pixel neighborhood that is used to calculate a threshold value for the pixel")
    private int blockSize = 127;

    @Attribute
    @Property(description="Constant subtracted from the mean or weighted mean")
    private int cParm = 80;

    public void setBlockSize(int blocksize) {
        this.blockSize = blocksize;
    }

    public int getBlockSize() {
        return blockSize;
    }
    public void setCParm(int cparm) {
        this.cParm = cparm;
    }
    public int getCParm() {
        return cParm;
    }

    @Override
    public Result process(CvPipeline pipeline) throws Exception {
        Mat mat = pipeline.getWorkingImage();
        blockSize = 2 * (blockSize / 2) + 1;
        blockSize = blockSize < 3 ? 3 : blockSize;
        Imgproc.adaptiveThreshold(mat, mat, 255, ADAPTIVE_THRESH_GAUSSIAN_C, CV_THRESH_BINARY, blockSize, cParm);
        return null;
    }
}
```
If this is ok, I can issue a pull request as soon as I find how to present the user with a menu to choose between _ADAPTIVE_THRESH_MEAN_C_ and _ADAPTIVE_THRESH_GAUSSIAN_C_

cri-s, 2017-03-25:
You don't need pull menu for MEAN or GAUSSIAN as only MEAN is usable. 
Gaussian washes out the contours and leave them open, with mean you have closed contours.
Escpecially on lookdown when using it on strip having open contours results in unusable results.

dzach, 2017-03-25:
You get the same open contours also with MEAN if your block size is small. To close them, increase block size, maybe up to 200 or more, depending on the size of the contours you might want to detect. This of course makes adaptive threshold more like simple Threshold, but differences can still be beneficial.

Here is a sample CPP demo to try with live video and see the results:

File _adaptiveThreshold.cpp_

```
#include "opencv2/imgproc.hpp"
#include "opencv2/highgui.hpp"

#include <iostream>
#include <stdlib.h>

using namespace std;
using namespace cv;

/** Function Headers */
void blksz_trackbar(int, void *);
void c_trackbar(int, void *);

/** Global Variables */
int blksz=103, c_parm=84;

/** @function main */
int main()
{
    Mat frame, frame_threshold;
    VideoCapture cap(0);

    namedWindow("Video Capture", WINDOW_NORMAL);
    namedWindow("Object Detection", WINDOW_NORMAL);

    //-- Trackbars to set thresholds for RGB values
    createTrackbar("Block size","Object Detection", &blksz, 255, blksz_trackbar);
    createTrackbar("c_parm","Object Detection", &c_parm, 512, c_trackbar);

    while((char)waitKey(1)!='q'){
        cap>>frame;
        if(frame.empty())
            break;
        //-- Detect the object based on RGB Range Values
        cvtColor(frame, frame_threshold, CV_RGB2GRAY);

        adaptiveThreshold(frame_threshold,frame_threshold,255,ADAPTIVE_THRESH_MEAN_C, CV_THRESH_BINARY,blksz,c_parm); 
        //-- Show the frames
        imshow("Video Capture",frame);
        imshow("Object Detection",frame_threshold);
    }
    return 0;
}
void blksz_trackbar(int, void *)
{
    int bsz = blksz < 3 ? 3 : blksz;
    setTrackbarPos("Block size","Object Detection", 2 * (bsz/2) + 1);
}
void c_trackbar(int, void *)
{
    setTrackbarPos("c_parm","Object Detection", c_parm);
}

```

vonnieda, 2017-03-25:
@dzach See FindContours for an example of how to use a dropdown with values that come from OpenCV enumerations: https://github.com/openpnp/openpnp/blob/develop/src/main/java/org/openpnp/vision/pipeline/stages/FindContours.java#L14

By declaring a Java enum and specifying that a property uses it a dropdown will be created from the enum values automatically.

A few other notes:
* Change the class name to ThresholdAdaptive. This puts it with the other Threshold class in the list and makes it easier to find when a user is thinking "I want a threshold of some type."
* You don't have to redeclare the enum values at the top like ADAPTIVE_THRESH_MEAN_C, those are in Imgproc already. Sometimes the names are slightly different. In this case you can use Imgproc.ADAPTIVE_THRESH_GAUSSIAN_C, Imgproc.THRESH_BINARY, etc.
* As for @cri-s's comment "You don't need pull menu for MEAN or GAUSSIAN as only MEAN is usable.", while that might be true, I am good with including additional functionality. The pipeline system is for experimentation with vision systems, and I think it's good to have more options even if they aren't ideal for a specific application. I use the pipeline editor all the time to learn more about how an algorithm works and what it's good for.

Hope that helps!

dzach, 2017-03-26:
Done it! Thanks @vonnieda !

vonnieda, 2017-03-26:
Merged, thanks!

closed 2017-03-26 by vonnieda