openpnp/openpnp #484: Jog controls: Z-movement depends on context

ldpgh, 2017-03-26:
While using the "Machine Controls" widget (Jog controls) the movement of the z-axis seems to depend on previous actions and state of the zaxis. All actions were made in the Machine Controls widget (Jog Controls) respective in the Camera window.

-------------------

Setup
- OpenPnp: version from 2017-03-05 ... snapshot build 00604b
- OS: Win7
- Camera: using Zoom (Heaad: H1) from OpenPnp-Test (no camera attached)

-------------------

Prepare machine.xml
- Start with deleted machine.xml. OpenPnP recreates a fresh machine.xml. Close the program.
- Add GCodeDriver section to machine.xml. At least ` CONFIRM_REGEX`, ` CONNECT`, `ENABLE`, `MOVETO`, `HOME` commands need to be added. To see the issue just `MOVETO` is of importance. Other are just to provide communication with OpenPnP.

GCoderDriver section inserted in machine.xml (... or see [machine.xml.0.fresh_withGCode](https://github.com/ldpgh/rasenmaeher/blob/master/machine.xml.0.fresh_withGCode))

```
<driver class="org.openpnp.machine.reference.driver.GcodeDriver" port-name="COM220" baud="57600" flow-control="Off" data-bits="Eight" stop-bits="One" parity="None" set-dtr="false" set-rts="false" units="Millimeters" max-feed-rate="1000" backlash-offset-x="-1.0" backlash-offset-y="-1.0" non-squareness-factor="0.0" backlash-feed-rate-factor="0.1" timeout-milliseconds="5000" connect-wait-time-milliseconds="0">
   <homing-fiducial-location units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
   <command type="COMMAND_CONFIRM_REGEX">
      <text><![CDATA[^ok.*]]></text>
   </command>
   <command type="CONNECT_COMMAND">
      <text><![CDATA[connect]]></text>
   </command>
   <command type="ENABLE_COMMAND">
      <text><![CDATA[enable]]></text>
   </command>
   <command type="DISABLE_COMMAND">
      <text><![CDATA[disable]]></text>
   </command>
   <command type="HOME_COMMAND">
      <text><![CDATA[home]]></text>
   </command>
   <command type="MOVE_TO_COMMAND">
      <text><![CDATA[moveto {X:X%.4f} {Y:Y%.4f} {Z:Z%.4f} {Rotation:E%.4f} {FeedRate:F%.0f}]]></text>
   </command>
   <command type="PICK_COMMAND">
      <text><![CDATA[pick]]></text>
   </command>
   <command type="PLACE_COMMAND">
      <text><![CDATA[place]]></text>
   </command>
   <sub-drivers class="java.util.ArrayList"/>
   <axes class="java.util.ArrayList">
      <axis name="x" type="X" home-coordinate="0.0">
         <head-mountable-ids class="java.util.HashSet">
            <string>*</string>
         </head-mountable-ids>
      </axis>
      <axis name="y" type="Y" home-coordinate="0.0">
         <head-mountable-ids class="java.util.HashSet">
            <string>*</string>
         </head-mountable-ids>
      </axis>
      <axis name="z" type="Z" home-coordinate="0.0">
         <head-mountable-ids class="java.util.HashSet">
            <string>N1</string>
         </head-mountable-ids>
      </axis>
      <axis name="c1" type="Rotation" home-coordinate="0.0">
         <head-mountable-ids class="java.util.HashSet">
            <string>N1</string>
         </head-mountable-ids>
      </axis>
   </axes>
</driver>
```
-------------------
Case / Context 1
- connect
- home
- set Distance 10
- move Z up one step using arrow-up button
- move Y up one step using arrow-up button
- press Park-XY
   - moves Z to 0
   - moves Y to 0
   - moves Z to 10
- press Park-XY again
   - moves Z to 0 . . . . . this is an unexpected move
   - moves Z to 10 . . . . . this is an unexpected move
- press Park-XY again
   - moves Z to 0 . . . . . this is an unexpected move
   - moves Z to 10 . . . . . this is an unexpected move

-------------------
Case / Context 2
- connect
- home
- set Distance 10
- move Z up one step using arrow-up button
- move Y up one step using arrow-up button
- move XY using mouse in the Cameras-widget (e.g. to top/right)
   - moves Z to 0
   - moves to new XY
   - <no_move> . . . . . different to Context 1 . . . . . Z-axis remains at park position
- move XY using mouse in the Cameras-widget (e.g. to top/right)
   - <no_move> . . . . . different to previous command . . . . . Z-axis remains at park position
   - moves to new XY
   - <no_move> . . . . . different to Context 1 . . . . . Z-axis remains at park position

-------------------
Case / Context 3
- connect
- home
- set Distance 10
- move Z up one step using arrow-up button
- move XY using mouse in the Cameras-widget (e.g. to top/right)
   - moves Z to 0
   - moves to new XY
- press Park-XY
   - <no_move> . . . . . different to Context 1 . . . . . but OK, for my opinion

vonnieda, 2017-03-26:
Hi @ldpgh, please set logging to TRACE and include the resulting log file from these tests: https://github.com/openpnp/openpnp/wiki/FAQ#where-are-configuration-and-log-files-located

ldpgh, 2017-03-26:
... log data were added.

vonnieda, 2017-03-26:
Thanks @ldpgh, here is what is going on:

Case 1: The unexpected moves are just the head moving to Safe Z before an XY move. The first time you press park notice the Z=0, Y=0, Z=10. This can be interpreted as:
* Move Z to safe Z (Z=0)
* Move XY to Park (Y=0, X is already at 0)
* Move Z back to where it was before park (Z=10)

In the case of the second and third parks, the same thing happens, except X and Y are already at 0, so the middle move is skipped.

In OpenPnP, whenever there is a move in XY we first make sure Z is at it's safe position so that we don't drag the nozzle across the bed. Then, in most cases, we put it back where it was before we moved. Z moving to Safe-Z is performed by telling all devices on the head to move to their Safe-Z position. This can be noted in the logs as `ReferenceHead DEBUG: H1.moveToSafeZ(1.0)`

Case 2: The difference here is that you are moving the camera, and not the nozzle. You have your Z axis tied only to the nozzle, so moves of the camera do not affect the Z axis. So, what happens is:
* move XY using mouse in the Cameras-widget (e.g. move to FID1)
    * move Z to 0: Head is moving to Safe-Z
    * move to new XY: Normal move
    * <no_move> . . . . . different to Context-1 . . . . . Z-axis remains at park position: Z remains because camera was moved. Camera has no Z axis so no movement happens in Z. 
* move XY using mouse in the Cameras-widget (e.g. to top/right)
    * <no_move> . . . . . different to previous command . . . . . Z-axis remains at park position: No move required because all devices already at Safe Z.
    * move to new XY: Normal move.
    * <no_move> . . . . . different to Context-1 . . . . . Z-axis remains at park position: No move required because camera is not attached to Z.

Case 3: Basically same as above. Nothing happens in Z during park because it's already at 0.

closed 2017-03-26 by vonnieda