openpnp/openpnp #485: Auto Update failing to close OpenPnP on Windows

vonnieda, 2017-03-26:
LA3PNA reports that auto update is now working (hooray!) but is failing to close OpenPnP before installing (boo.) Message is: http://imgur.com/a/yleLE

la3pna, 2017-03-26:
Workround is to kill the OpenPnP process in the task manager. Look for the
version with the OpenPnP logo, not the one with the installer logo.

vonnieda, 2017-03-27:
Made some changes in https://github.com/openpnp/openpnp/commit/044fc48fcb46a2dfa56d9971675104084be29f07 to try to address this. Waiting on feedback.

vonnieda, 2017-03-28:
No luck, still locked up. Next step is to attempt to try to get a JVM dump with thread info as that will tell us which thread is blocking.

vonnieda, 2017-03-28:
@la3pna found that by starting with a fresh machine.xml the problem went away. Now debugging the machine.xml that he sent me.

vonnieda, 2017-03-28:
Another user noted that @la3pna previously said that one of the cameras was not working. I bet the issue is in the close/shutdown code for the cameras. https://github.com/openpnp/openpnp/blob/develop/src/main/java/org/openpnp/machine/reference/camera/OpenCvCamera.java#L195 should probably have a timeout, rather than waiting forever.

vonnieda, 2017-04-02:
I believe this should now be resolved. @la3pna, can you check? You'll need to install the current version and then do an update when I next push a release.

commit be475a25 (2017-04-02): Temporarily removed system logging to log window as it was causing OpenPnP to fail to shutdown cleanly. This is an effort to fix #485 and #498 and is related to the discussion in #288.
closed 2017-04-13 by vonnieda