openpnp/openpnp #486: Does a move to the Up Camera honor the safe-Z setting?

dzach, 2017-03-26:
In my current configuration the up looking camera's top is at a level higher than safe-Z for the nozzle. When the head moves towards the camera, the nozzle travels at its safe-Z height. I thought that the safe-Z setting of the Up Camera would be followed during a move to the Up Camera, but apparently this is not taken into consideration with my current setup:

```
      <cameras>
         <camera class="org.openpnp.machine.reference.camera.OpenCvCamera" id="CAM1489158164459" name="UpCam" looking="Up" settle-time-ms="250" rotation="0.0" flip-x="false" flip-y="true" offset-x="0" offset-y="0" crop-width="0" crop-height="0" scale-width="0" scale-height="0" deinterlace="false" device-index="1" preferred-width="0" preferred-height="0" fps="24">
            <units-per-pixel units="Millimeters" x="0.052" y="0.052" z="0.0" rotation="0.0"/>
            <vision-provider class="org.openpnp.machine.reference.vision.OpenCvVisionProvider"/>
            <head-offsets units="Millimeters" x="48.993" y="11.441" z="15.0" rotation="0.0"/>
            <safe-z value="15.0" units="Millimeters"/>
            <calibration enabled="false">
               <camera-matrix length="9">511.35529749877384, 0.0, 331.74307694552664, 0.0, 507.98404985294843, 268.69560159792445, 0.0, 0.0, 1.0</camera-matrix>
               <distortion-coefficients length="5">0.14496614276925227, -0.38438655431146146, 0.0017563342971549127, -0.001999721447235985, 0.3300097296385976</distortion-coefficients>
            </calibration>
            <properties class="java.util.ArrayList"/>
         </camera>
      </cameras>

```
The head does indeed move to the camera's set location and height, but the Z axis move is unfortunately performed last.

Is the UpCam's safe-Z setting honored? How can I make the nozzle raise to UpCam's safe-Z (or location height) **BEFORE** the move is made?

vonnieda, 2017-03-26:
No, Safe-Z is only applied to objects that move. The fact that there is a Safe-Z setting for the up camera is actually a side effect of it being a camera. It's a bug, and the field will be removed.

You should adjust your configuration so that Safe-Z for the nozzle is truly safe. When the nozzle is at Safe-Z it should be safe from hitting anything on the machine. This is the only way OpenPnP can know it's safe to move in XY.

closed 2017-03-26 by vonnieda

dzach, 2017-03-26:
I cannot adjust it to a higher level because the design is a dual head and so, if I did, the second head would be lower.

vonnieda, 2017-03-26:
You will need to move the camera lower then. It's imperative that all fixed objects be mounted below the Safe-Z plane or crashes are very likely. It's not just when moving to the camera, but even just moving past it during normal operations. OpenPnP assumes that it can move anywhere in XY "safely" when it is at Safe-Z.

dzach, 2017-03-26:
Can the move to the camera start with the Z axis and then proceed to the XY?

vonnieda, 2017-03-26:
No, sorry. I won't add this special case. Having any object be above the Safe Z plane is a machine design flaw and needs to be fixed in the machine design. Otherwise I can't guarantee that there won't be other head crashes.

dzach, 2017-03-26:
Ok. The safe-Z for the Up looking camera is misleading then. It gives the impression that there is something special there, and I thought it was just for my case. Fortunately the Juki nozzle seems to be able to withstand some abuse.

vonnieda, 2017-03-26:
Agreed, it's misleading. Already removed: https://github.com/openpnp/openpnp/issues/487 :)