openpnp/openpnp #488: gcode driver dispense missing

cri-s, 2017-03-27:
Can you implement dispense command, where the first location is dispense point and the second is the SafeZ.
If however there is xy distance between the two locations the time parameter changes to F, speed.
I assume the gcode driver can  already read back PCB position using g38 command.

vonnieda, 2017-03-27:
@tracklogic is working on dispense in https://github.com/openpnp/openpnp/pull/464

cri-s, 2017-03-27:
Just looked on #464 dispense() implementation, that don't work for pneumatic dispenser(including auger type) that dispense dots. Further there is a lot hardcoded, used actuator, speeds, ... not really suitable for
gcode driver.

vonnieda, 2017-04-13:
I'll accept a PR for dispense if it's complete and working, but until then I am closing dispense issues since we've decided to deprecate this feature.

closed 2017-04-13 by vonnieda