openpnp/openpnp #490: Executing external commands

dzach, 2017-03-27:
Is there any way to execute external commands from within openpnp, e.g, from a javascript script?

I tried the `$EXEC( ... )` Nashorn global function but it seems  shell scripting features are not enabled.

vonnieda, 2017-03-27:
I'm not aware of anything off the top of my head, but I also haven't tried to do this. If you discover something please update the Wiki at https://github.com/openpnp/openpnp/wiki/Scripting. Have you tried just calling the standard Java process stuff in Runtime? https://docs.oracle.com/javase/7/docs/api/java/lang/Runtime.html#exec(java.lang.String)

cri-s, 2017-03-27:
I have done it a lot using ostermillers utils, from java and bsh. See no reason why it should not work from js.

dzach, 2017-03-27:
Runtime.exec  executes the command in a separate process, e.g.

```
Runtime = Java.type("java.lang.Runtime").getRuntime();
Runtime.exec("firefox");
```

works, but 

```
Runtime.exec("ls -l");
```
does not print anything to the output (or to the Log window).

dzach, 2017-03-27:
It seems that [something like](https://wiki.openjdk.java.net/display/Nashorn/Nashorn+jsr223+engine+notes):
```
NashornScriptEngineFactory factory = new NashornScriptEngineFactory();
ScriptEngine engine = factory.getScriptEngine(new String[] { "-scripting" });
```

would allow shell scripting. From the examples in [Shell Scripting](https://docs.oracle.com/javase/8/docs/technotes/guides/scripting/nashorn/shell.html#sthref25) I gather that shell scripting runs the script in the same process (not a separate one, as does `Runtime.exec()`) and returns the output to the calling function.

This is a partial solution. As a workaround, output can be returned through a temporary file, but that's not so elegant.

dzach, 2017-03-27:
Update the Wiki at https://github.com/openpnp/openpnp/wiki/Scripting with a small example.

vonnieda, 2017-03-27:
Thanks @dzach. Should we close this ticket or do you feel we need to look further into enabling scripting in the engine?

dzach, 2017-03-27:
I'm currently trying to find a way to get the output of a system command to javascript. Let's wait a little more and see if we can get something.

dzach, 2017-03-27:
Ok, solved. 
I updated the examples in the Wiki (https://github.com/openpnp/openpnp/wiki/Scripting) with code to read stdout from a system command found [here](http://alvinalexander.com/java/edu/pj/pj010016). It would be better to add a helper function, but for the time being this can be closed.

closed 2017-03-27 by dzach

vonnieda, 2017-03-28:
Looks good, thank you!