openpnp/openpnp #494: Return values from a SriptRun stage to the OpenCV pipeline

dzach, 2017-03-29:
I have not yet found a way to return results to the pipeline from a ScriptRun script. When I try a return statement I get:

`javax.script.ScriptException: <eval>:7:0 Invalid return statement`

no matter what the return value is. As a workaround I now manipulate the model from the `results` stage. 

What is this the proper way of working with ScriptRun?

vonnieda, 2017-03-29:
I'm seeing the same behavior. I could have sworn this worked when I wrote it :) I'll do some more testing and check my notes and get back to you.

vonnieda, 2017-03-29:
Got it, example:
```
var Result = Packages.org.openpnp.vision.pipeline.CvStage.Result;
var result = new Result(pipeline.getWorkingImage(), "I am a model object");
result;
```

You can't use return because it's not a function. You just let the return value dangle there at the end. Found at http://stackoverflow.com/questions/38034497/java-returning-an-object-from-scriptengine-javascript. I'll update the examples tonight.

dzach, 2017-03-29:
Cheers!

closed 2017-03-29 by dzach

vonnieda, 2017-03-30:
Documentation updated at https://github.com/openpnp/openpnp/wiki/CvPipeline#scriptrun