openpnp/openpnp #498: win registry issue

cri-s, 2017-04-01:
At start there is a registry issue and sometime it blocks the start of openpnp too generating thousands of log errors until stack overflow occur. deleting the registry keys resolve the problem. It was noted when using the standalone pipeline editor. Then launching a older openpnp that exited because of package.XML config read error. When launching the newer version, it don't started giving out only registry errors.
On testing it occurred 2-3 times launching only the new version, the case was infrequent.

vonnieda, 2017-04-02:
Note: The registry error has been around a long time, but I didn't think it actually caused any problems. Seems like it does. The failure to start is likely related to the stack overflow being caused by the new logging code #288

vonnieda, 2017-04-02:
@cri-s I think this is fixed now. Can you check it out?

vonnieda, 2017-04-08:
Found the registry error: http://stackoverflow.com/questions/16428098/groovy-shell-warning-could-not-open-create-prefs-root-node

Looks like it's not a concern.

closed 2017-04-13 by vonnieda