openpnp/openpnp #501: Auto Panelization Improvements

vonnieda, 2017-04-02:
This is a follow up to #456 with some suggestions for improvements to the new feature.

* Move DlgPanelXOut and DlgAutoPanelize out into their own files. JobPanel is already too long and these can be cleanly broken out.
* Bug: Add two unique boards. Click the panelize button. Get an error that only one board can be used but dialog opens anyway. After this the state of the add board and delete board buttons is broken.
* Would like to see the single board requirement removed. Everything is in place now for multiple panels, so even if it's not used, there's no reason to have all the special cases for a single board and it just makes it confusing.
* Opening the panel dialog and then closing it without hitting Ok leaves the job in an unloadable state due to the part-id in the panel not being populated.

vonnieda, 2017-04-02:
FYI: I am currently working on moving the dialogs into their own files.

BendRocks, 2017-04-04:
For the "Opening the panel dialog and then closing it..." issue you note, is that as simple as adding the require=false attribute on partId field? Or is something bigger needed? I tried it and it seems to work but I want to make sure I'm not missing something larger

    @Element(required = false)
    private String partId;

tonyluken, 2023-08-02:
@vonnieda,

Can we close this issue as I believe it was address by [PR1507](https://github.com/openpnp/openpnp/pull/1578)?

vonnieda, 2023-08-03:
Yes, looks good to me. Thank you @tonyluken!

closed 2023-08-03 by vonnieda