openpnp/openpnp #505: ReferenceFeeder and Feed

BendRocks, 2017-04-04:
I'm using YamahaCL feeders. For background, these feeders have a two stage feed. The first stage occurs when you turn on air to the feeder. The feeder pulls back a protective cover and exposes the next part on the tape. At this stage, you can pick the part. During this time, the air is hissing in the feeder and it's fairly loud. 

When the air is turned off, the tape advances and the protective finger slides back into place and everything goes quiet again.

I've derived from ReferenceSlotAutoFeeder

For normal operation, it's OK to turn on the air in the feed() method, and then turn off air in the postPick().

The UI has a button (black with green arrows) that "commands the selected feeder to perform a feed operation". This button only calls feed(). And it seems wrong to have it call postPick() as there's a lot of logic in postPick().

What I am thinking is adding a manualFeed(...) method to Feeder.java for the UI to call (today it just calls feed()). This would allow a two stage feeder to work for manual actuation, and the existing logic everywhere else could work as-is (which means 2-stage feeder would get the air turned on in feed() and turned off in postPick().

Default behavior would be that manualFeed() calls feed(). Overridden behavior on CL feeders would allow manualfeed to turn-on air, wait 250 mS, and then turn off air.

Is there a better way to reliably allow UI code to manually feed a 2-stage stage feeder? Calling postPick() form UI code seems fraught with potential issues. I've also tried a push-push approach, where the first call to feed(), and the second call to feed() turn off the air. But this has issue if someone turns on air and then decides to run a job. 

Finally, it looks like UI code calls feed() in just the feeders panel and the parts panel. These calls would be changed to manualFeed(). I don't think other feeder code would need to change EXCEPT in YamahaCL feeders derived from ReferenceSlotAutoFeeder. Those would override the default manualFeed().

Thoughts? I'd be happy to knock this up and submit for review if it's easier to look at diffs

vonnieda, 2017-04-04:
Hi @BendRocks, I understand the proposal, but I don't really understand the problem you need to solve. 

feed() is intended to do just that: feed the feeder; which, in my mind, is "prepare the next part to be picked". I think this is what happens with the CL feeder. Alternately you can use the "feed and pick" button which does the whole feed, pick, postPick sequence, and I think this is the most common operation. 

So, before I blab on a bunch more, can you tell me what the actual use case is that you want solved? Is it purely the case where someone hits feed, doesn't perform a pick, and then hits feed again? I feel like this would be better solved with some state for this particular feeder so that it knows "I fed, but didn't postPick, I need to postPick if you want me to feed again"

BendRocks, 2017-04-04:
The entire issue is the feed button in the UI. It's a needed button, no question.

But inside the feed() routine, what should the code do? A feed operation on the CL feeder is 3 parts: Open the air, wait a bit, then close the air. 

Since the feed() code doesn't know who is calling (user pressing button or running a job), it has no way of knowing in the feed() routine how many of those steps should be done OR if the postPick() will ever be called.

I think the best you can do currently is turn the air on when the feed button is pressed. And then turn the air off when the feed button is pressed again. That is what I'm currently doing. But two clicks for one  feed feels kind of hacky. What if there's a feeder in the future that needs 3 steps to actuate? Does that need 3 pushes in the UI to make it feed once?

To argue the other side of this, it's very common for there to be buttons to let you advance the feeder step by the step. But I think those buttons might belong in the UI for the feeder. Thus, the feeder panel "feed" button should perform a full feed operation (all steps), and if you need some additional granularity, those buttons could live in the feeder-specific UI.

vonnieda, 2017-04-04:
I would argue that the CL feeder's feed operation is actually just "open the cover". The other two parts of the operation you described are the pick and postPick operations, in OpenPnP parlance.

What about just changing your feed command to: turn off air, wait a few ms, turn on the air? Then a feed will always finish any previous unfinished operation. This could be done without maintaining any state. If you want to add in a little state, I think it makes sense to just keep state that says "The feeder is open" and if it's open, and someone does a feed(), close it first.

That said, I'd like to better understand in what case you are using the feed button? Is it just to advance the feeder as you load it, or something like that?

BendRocks, 2017-04-05:
But open the cover doesn't actually feed...when you release the air it actually feeds. 

> What about just changing your feed command to: turn off air, wait a few ms, turn on the air? 

It's possible. The TVM920 could do almost 3600 pph out of the box (no vision) and I'll likely be less than half of that on OpenPNP. I'd hate to add another 50-100 mS of delay on each pick for no reason. But for early work maybe it's the right way.

> That said, I'd like to better understand in what case you are using the feed button? Is it just to advance the feeder as you load it

Yes, that and to see if something is stuck, or clear a part that is crooked in the pocket....

vonnieda, 2017-04-05:
@BendRocks I'd really like to find a solution to this that doesn't modify the feeder interface. If it turns out that that's what we need to do, that's fine, but currently I feel I don't fully understand the problem enough.

I think that maybe I misunderstand how these feeders are intended to be used. So, can you verify something for me?

Is the correct order of operations:
1. Apply air, cover opens.
2. Pick.
3. Retract nozzle.
4. Release air, feeder feeds and cover closes.

Or is it:
1. Apply air, cover opens.
2. Release air, feeder feeds and cover closes.
3. Pick.
4. Retract nozzle.

Or perhaps something else?

I was under the impression that the cover covered the previously fed part until just before the pick so that the part doesn't bounce around, but perhaps not.

Thanks,
Jason

BendRocks, 2017-04-05:
The correction sequence is this:

1. Apply air, cover opens.
2. Pick.
3. Retract nozzle.
4. Release air, feeder feeds and cover closes.

Right now feed() handles (1) and post-pick handles (4). The UI, however, only calls feed().

Fixes summary options (detailed above in various forms)
Fix 1:
UI could call post-pick. Not sure side effects.

Fix 2:
Logic in Feed would need two pushes of UI button to handle full feed cycle. Clumsy for user

Fix 3: 
Close all feeders + some delay in in feed() would make sure state was always maintained. This slows down cycle by 10-20%

Fix 4:
manualFeed() added to feeders. This would be called by UI only and handle full feed cycle. 

Fix 5:
add flag to feed() that indicates a full feed cycle is to be performed. UI would set this.

vonnieda, 2017-04-05:
Thanks @BendRocks, that is very helpful.

The purpose of the feed button, and the feed() call is to prepare the feeder for pick. For a CL feeder, the feeder is prepared for a pick by applying the air, so this meets the goal. Except that the CL feeder is unique in that it needs postPick before it can feed again.

I just realized, after typing a long response, that we have another fundamental problem and I think that if we solve that we'll solve both:

Currently, there is a situation where a CL feeder can get "stuck" during a job run. The scenario is:
1. Job processor performs a feed()
2. Job processor performs a pick(). pick() fails for some reason - maybe vacuum sense.
3. Job processor attempts to retry, performing a feed(). 

Since postPick() is not called when a pick fails, the feeder is never going to recover. Your suggestion of manualFeed would have this same problem.

So, to me, this indicates that we need a mechanism where we guarantee the sequencing of (feed -> postPick -> feed) and never allow (feed -> feed).

How about if we just add a boolean to ReferenceAutoFeeder and ReferenceSlotAutoFeeder that we check in feed, and if it's set we postPick before feed? feed set it, postPick clears it.

BendRocks, 2017-04-06:
The last paragraph will fix the aborted feed...but on the TVM920 you can't have an aborted feed because there's no part sensor (ie vacuum). The factory SW uses vision to find a bad feed if needed. But the feed/pick is so reliable it's not really needed.

The reason I raised this issue was because I wanted to fix the two button pushes required to feed the feeder. But as I noted above, I can live with this no problem as-is and can bring it up again down the road if it gets really bothersome.

Thanks for the guidance and suggestions

closed 2017-04-06 by BendRocks