openpnp/openpnp #509: Add on Hard Start Button From Outside

Thangpham2507, 2017-04-07:
Add on the hard start, step and stop button from outside

Discussion
https://groups.google.com/forum/#!topic/openpnp/NAP7KN_-tCE

vonnieda, 2017-04-09:
We don't currently have a way to raise an event from a driver or hardware, so that makes this a bit tricky. A few ideas:

1. Poll an actuator's read(). This would create a lot of driver chatter and the performance would not be very good.
2. Add a driver event system, or add it on to actuator. This would allow a driver to raise a generic event.
3. Instead of trying to do this in a generic fashion, just add first class support for these three specific functions.

Not really sure what the best option is right now.

cri-s, 2017-04-11:
I use Keyboard hotkeys for that, ctrl-break for stop, ctrl-F5 for start and ctrl-F10 for step/pause.
ctrl-shift-F5 start with snema board loading/unloading, ctrl-shift-break unloads the snema board
without loading and starting a new board. With firmata, the java robot class is used to inject keyboard shortcuts, otherwise PIC18F14K50 as Keyboard emulator for 10 keys. If the mapping is different, using
only F keys without modifiers, standard keyboard can be used with wires from the front buttons to the keyboard soldered.

yaddatrance, 2017-07-15:
Hi Cri-S, did you add those hotkeys? Not seeing those in the source (Looking in the gui dir)

cri-s, 2017-07-15:
I have added it inside private source tree including joypad support.
Key event processing require some changes and times ago when I have
proposed this for barcode readers there was no interest in including it and
especially reserving/assigning functions keys for particular functions
 that are callable globally.

yaddatrance, 2017-07-15:
Any chance you could share how you implemented that? Global hotkeys sounds so important, especially given the lack of a hardware event system. My hope is to add start/stop/step as well as a switch that sets jog increments directly vs up-down. Your original post had me so excited there for a moment. :) I thought all my work was done for me...

vonnieda, 2017-07-15:
The job controls including increments already have global hotkeys. See the
keyboard shortcuts section in
https://github.com/openpnp/openpnp/wiki/User-Manual

vonnieda, 2017-07-15:
Sorry, jog, not job.

yaddatrance, 2017-07-17:
@vonnieda I saw those keys mapped across several sections:

MainFrame.java, has the jog controls, 
MachineControlsPanel.java, has Stop (startStopMachineAction) and Home (homeAction)
JobPanel.java has Open/New/Save

Do you see any issues if I add something like 

```
putValue(MNEMONIC_KEY, KeyEvent.VK_F5);
putValue(ACCELERATOR_KEY, KeyStroke.getKeyStroke("F5",
                    Toolkit.getDefaultToolkit().getMenuShortcutKeyMask())); 
```

into startPauseResumeJobAction in JobPanel.java?

So the code block would look like:

```
    public final Action startPauseResumeJobAction = new AbstractAction() {
        {
            putValue(SMALL_ICON, Icons.start);
            putValue(NAME, "Start");
            putValue(SHORT_DESCRIPTION, "Start processing the job.");
            putValue(MNEMONIC_KEY, KeyEvent.VK_F5);
            putValue(ACCELERATOR_KEY, KeyStroke.getKeyStroke("F5",
                    Toolkit.getDefaultToolkit().getMenuShortcutKeyMask())); 
        }

        @Override
        public void actionPerformed(ActionEvent arg0) {
            UiUtils.messageBoxOnException(() -> {
                fsm.send(Message.StartOrPause);
            });
        }
    };
```

or is it a better idea to create a new action block that takes that key directly calls

`UiUtils.messageBoxOnException(()->{ fsm.send(Message.StartOrPause);});`

vonnieda, 2017-07-17:
Adding that mnemonic and accelerator is fine with me if it does the trick.
I don't think it will work if you don't have that tab open and the panel
focused, but I haven't tried.

Jason

cri-s, 2017-07-18:
Code for F key, remove what is not needed, ,
You can have several functions like that inside Job and inside Feeder and ...

KeyboardFocusManager.getCurrentKeyboardFocusManager()
  .addKeyEventDispatcher(new KeyEventDispatcher() {
      @Override
      public boolean dispatchKeyEvent(KeyEvent e) {
        if (KeyEvent.KEY_PRESSED == e.getID()) {
    if(e.isAltDown() || e.isAltGraphDown() ||  e.isControlDown() ||
e.isMetaDown()  || e.isShiftDown()) { return false; }
            switch(e.getKeyCode()) {
                            case KeyEvent.VK_F1:  //
              ;return true; case KeyEvent.VK_F2:  //
              ;return true; case KeyEvent.VK_F3:  //
              ;return true; case KeyEvent.VK_F4:  //
              ;return true; case KeyEvent.VK_F5:  //
              ;return true; case KeyEvent.VK_F6:  //
              ;return true; case KeyEvent.VK_F7:  //
              ;return true; case KeyEvent.VK_F8:  //
              ;return true; case KeyEvent.VK_F9:  //
              ;return true; case KeyEvent.VK_F10:  //
              ;return true; case KeyEvent.VK_F11:  //
              ;return true; case KeyEvent.VK_F12:  //
            }
        }
          return false;
      }
});

yaddatrance, 2017-07-19:
@cri-s I like your idea.

Anybody see any problems implementing that inside MainFrame.java?

vonnieda, 2017-07-19:
@yaddatrance We already have a system for setting global hotkeys in MainFrame at https://github.com/openpnp/openpnp/blob/develop/src/main/java/org/openpnp/gui/MainFrame.java#L437. Is there a reason that won't work?

yaddatrance, 2017-07-21:
Ah so those are global! Ok, I'll implement my changes there then, thanks a bunch.

Benjamin45, 2020-11-24:
Hello,

Do you have some news about implement a way to raise an event on push button ?

markmaker, 2020-11-24:
Hi @Benjamin45 

I guess you could use an MCU that emulates a USB keyboard and then send one of the hotkeys that were mentioned [here](https://github.com/openpnp/openpnp/pull/609).

One example:
https://www.pjrc.com/teensy/td_keyboard.html

_Mark

markmaker, 2021-04-25:
I think the [solution proposed above](https://github.com/openpnp/openpnp/issues/509#issuecomment-733090095) is viable, i.e. this does not have to be solved within OpenPnP. 

It would be difficult in G-code i.e. it would involve polling at rates that are prohibitive.

vonnieda, 2021-04-26:
I agree that the easiest method would be a keyboard wedge.

This has me thinking, though. We already have a read thread in the Gcode(Async)Driver that is always attempting to read from the serial port. Any message that comes in will get read immediately. We already check every line read for COMMAND_CONFIRM_REGEX, COMMAND_ERROR_REGEX, and POSITION_REPORT_REGEX - what if we added ACTUATOR_EVENT_REGEX? 

Anything matching ACTUATOR_EVENT_REGEX would get passed to a new callback on Actuator (`Actuator.event(String)`) and Actuator subclasses could handle these however they wanted. This would give us a way to handle out of band events such as button presses, jog controllers, board presence sensors, conveyor sensors, etc.

markmaker, 2021-04-26:
Yes, if we leave the classical imperative G-code model and have controllers that report state changes without polling, then this could be done. 

I would argue that the new event concept be coupled with the read concept i.e. use the same ACTUATOR_READ_REGEX. 

Because actuator reads spoil continuous motion, I was thinking about what I called "monitoring actuators", where the read value is obtained in the background and being continually stored on the actuator, so the actual code driven read is fast and alarm values (or closures) can be configured to handle incoming values. It would be nice to e.g. keep vacuum monitored throughout, while a part is on the nozzle and to avoid having to query the controller at specific waypoints (PartOn/PartOff checks), which means still-stand i.e. spoiling optimized continuous motion. 

See here:
https://github.com/openpnp/openpnp/blob/e4f983f76f716593b7fedf72c9b75c68c9c95fc5/src/main/java/org/openpnp/machine/reference/driver/GcodeAsyncDriver.java#L62-L82

In this, I was thinking about (shared) polling, but of course this would work even better with custom controllers that send unsolicited reports.