openpnp/openpnp #513: Improved Actuator Panel

dzach, 2017-04-10:
# Feature Request
## Describe the Feature
Improve the `Actuators` panel to reduce the number of clicks required for actuator operations. Eliminate the settings modal dialog and replace it with a frame inside the panel.

## How Will It Look?
Here is a mockup of the proposed changes:

[image]

## Why Do We Need It?
The actuator panel provides buttons and a dialog for selecting and setting the value of a defined actuator. However, this panel requires an unnecessary large number of clicks for just a simple on/do work/off operation, mainly because the dialog is modal and needs to be closed and reopened, between operations. Reducing the number of clicks required will make user experience better and will reduce the time required for setting up a job.

## Provide Examples

 For example, e.g. to turn on and off he lights of the camera one needs to:

1. go to the `Actuators` panel,
2. click on the relevant actuator button,
3. press the `On` button
4. press the `Close` button

then go about doing some operation with the lights on and then, just to turn the lights off, 
the above 4 clicks have to be repeated. This is necessarily complicated and can be simplified to a minimum.

vonnieda, 2017-04-10:
I like the idea of eliminating the dialog, but I don't think we can do it with a grid of buttons. Some people use an extremely large number of actuators for their feeders, and this would make the panel become too crowded. How about a dropdown to select the actuator instead?

dzach, 2017-04-10:
Drop down menus require at least two clicks at all times: one to bring the menu up and another to chose the item.

I intended to show a vertical scroll bar in the mock up, but forgot about it. Scroll bars require none, one or more clicks or a slide to bring the item in view, but then the item should stay in view and the same operation will always be only one click away.

One can also have the boolean actuator items as toggle buttons, so that, once selected, one can click them repeatedly without changing position, or even hit `Space`, to toggle their value.

The entry box for setting the analog (double) value should also accept the `Enter` key for setting its value.

vonnieda, 2017-04-10:
@dzach I'm against the idea of using buttons for this. It makes the UI very cluttered. This is how it worked previously and a big part of the reason I changed it was that I saw screenshots of users' installations where the buttons were huge (due to actuator names), or getting cut off, or just expanding the UI too much. You can try the old version out by checking out a version previous to e82a13344c9f2263149460c35323236602e5adc5.

In addition, toggle buttons were also found to be a UI problem. Many users did not understand how they worked and in Java the rendering was somewhat poor, so it wasn't always clear what their state was. On top of that, we don't currently have a way to say that an actuator is a "booleaen" or "double" actuator, so we can't use a different control depending on that.

I don't personally feel that scrolling, possibly multiple times, is any worse than two clicks but if you feel it's a real issue then I'd be open to alternatives - just not buttons. For instance, we could use a JList or JTable, which gets you uniformity of widgets, and scrolling to handle long lists. 

In general, I consider the actuators panel to be more of a one time use during setup thing than something someone is going to be using regularly. If you find you are using it often enough for the UI to be an issue I'd be interested in what the use case is so we can see if we can improve that.

dzach, 2017-04-10:
If  button toggle state is not clearly shown with the button, then this idea will obviously not work. Forgoing the modal dialog may be a great improvement on its own merit, thought.

Indeed I use the actuator panel a lot : while testing pipeline stages for turning on / off lights to position the head/camera to different locations marked as `Loose feeders` etc. For turning on/off the vacuum pump e.t.c. But that's just me, doing eternal testing, but not work. Unfortunately I can't speak for others, neither do I have the experience of pnp machines in production, to draw some ideas from them, so my comments are of limited value.

I started fiddling with the `Macro` idea, starting at the UI level, and have created an extra panel next to "Machine" with some fake buttons in it. However, a separate panel with buttons, even if they have icons instead of text, is in no way as good as user defined icons/toolbars that can be present on any panel without forcing changes to the usual work flow, but instead enhancing it with extra functionality. It would be really nice to be able to put some user defined toolbars, or single icons, wherever extra tools are needed. That would make tailoring the UI to one's needs very easy.

But that's getting in deep water for my infinite monkey doing Java programming.

cri-s, 2017-04-13:
If you have this problem, here a helper for you, it generate light.txt on config dir with all actuators 
and you remove that what you don't want have. shift click is for disabling the actuator.
it's intended where the actuators are exclusive and on/off have different settings mapped.

[lights.bsh.txt](https://github.com/openpnp/openpnp/files/919711/lights.bsh.txt)

dzach, 2017-04-14:
@cri-s Thank you for the code. 
It's an introduction to BeanShell for me, so please forgive my ignorance. I'm getting the error:
``` 
$ bsh lights.bsh
Evaluation Error: Sourced file: lights.bsh : Attempt to resolve method: 
getConfigurationDirectory() on undefined variable or class name: config [...]
```

I tried 
```
import org.openpnp.model.Configuration;
```
but that didn't help. Any suggestions?

cri-s, 2017-04-14:
It works without modification and it's intended for running inside
OpenPnP ! as script.
copy it inside you config directory into the scripts subdirectory and
launch it from script menu.

If you don't see it as script inside directory after having pressed
update scripts directory,
you need update you'r copy of openpnp, there was recently (10 days
approx) a patch for that.

dzach, 2017-04-14:
Very nice! Is there a way to make them toggle buttons? Maybe by storing their state to a global variable? OR, if this cannot be done, then turn them off by clicking the right mouse button, so that there is no need to go to the keyboard for `shift`?

cri-s, 2017-04-14:
It was made for light groups where one actuator , as example TopDirect75 set the light to 75% and the off commands set it to 25% for not having too mutch actuator buttons on the UI and every other actuator on
the same group override the settings.
I have simplified it for removing that and updated the code. Just download it again.

dzach, 2017-04-14:
@cri-s this is perfect, thank you very much!

If one added a 'settings' frame for setting double values for the , then this would be a complete implementation for this Feature Request. I believe it would make it dead easy for many people to set up their machines and do daily work.

The issue of long button labels due to long actuator names that @vonnieda  mentioned above can be possibly solved by giving a separate parameter, e.g. `title` or `label' to each actuator, in addition to its `name'. One could also add an optional icon, and an option to use text, text and icon, or icon only, to save space.

What a difference would that make!

dzach, 2017-04-14:
My  comment above is wrong. I didn't realize `name` was actually user input and proposed `label` to correct the issue of long names.

cri-s, 2017-04-14:
For you'r convenience i have updated it to be integrated into panel.
Do whatever you want with the actors, and add this to the startup script.
good coding time.

[act1.bsh.txt](https://github.com/openpnp/openpnp/files/922576/act1.bsh.txt)

dzach, 2017-04-14:
Thank you @cri-s !

What do you mean by "startup script"? _openpnp.sh_ ?

cri-s, 2017-04-14:
> ... or, do you mean the Startup event? That will probably do it.

If you place the script Startup.bsh into scripts\Events directory that
get executed at startup.
You can copy the code, or include it, as example

cd(scripting.getScriptsDirectory().getAbsolutePath());
source("act1.bsh"); // actuator boolean ui

for include it.
eventually use
cd(getConfigurationDirectory().getAbsolutePath());
for the configuration dir.
for both case you can do
cd("..") or cd("script")
and then source("file.bsh");

dzach, 2017-04-15:
Thanks again @cri-s .
I enjoy very much the new toggle actuator window.
If you don't mind, I'd like to include some of these in the Scripting page of the wiki.