openpnp/openpnp #514: Driver Improvement Idea

trampas, 2017-04-13:
Currently the G-code driver system in OpenPnP uses a main driver and subdrivers.  Would it be possible instead to have the driver system operate on an Action and Request system.  The basic idea is that OpenPnP has actions it needs to perform like 'Move' but rather than directly tying the Move to a driver leave up to the controller server. 

A simple implementation example:
So the idea is you specify the the "controllers" for your machine these are like sub-drivers currently and for this example lets assume all are serial controllers.  The controllers are setup with serial port baud etc. such that my machine setup has ctrl1, ctrl2, ctrl3, etc.  

Now in the XML file what I can do is have the Action tied to the correct controller(s) commands.  For example a Move action might be:

Move(X,Y)
   ctrl1.send("G0 X%X Y%Y F1000)  //sends command waits for the regular expression response 
   ctrl1.send("M400")

I could change it so that it is:
Move(X,Y)
   ctrl1.send("G0 X%X F1000)  //sends command waits for the regular expression response 
   ctrl2.send("G0 Y%Y F1000)  //sends command waits for the regular expression response 
   ctrl1.send("M400")
   ctrl2.send("M400")

This way main OpenPnP application does not need to know low level details of the motion controller hardware. 

A More advance system would be a Client/Server:
OpenPnP creates Actions and Requests.  The implementation of  the actions and requests is up to the controller server.   That is the controller server runs as it's own process and OpenPnP makes Action and Requests to the the controller server.  Here the controller server is basically it's own process.  There are some nice advantages here, for example if the controller's job is to know the head location (ie current location is a request from OpenPnP). Then OpenPnP can be stopped and restarted without loosing positional control of the machine.

vonnieda, 2017-04-13:
I think this is a good approach for a new, universal driver. It's similar in concept to the HAL that LinuxCNC uses. I don't think it makes sense to try to change the GcodeDriver to work like this, since that driver already works fine for the vast majority of users. If this is something you want to add, I'd be happy to take a pull request for it.

Something to keep in mind is that we don't just have X, Y, Z, C. We have each of those axes for potentially unlimited number of tools. So there will need to be a mapping between tool axis and controller axis.

trampas, 2017-04-13:
I agree that breaking existing systems is not a good idea, so keeping the
existing driver system and just making a universal driver seems like the
best approach.

My thoughts are that any "tools" that need movement will be based upon an
Action request from OpenPnP.

For example if you have N heads then you would have an action to rotate
heads and pass in the head id/name, it is up to the controller server to
know how to handle the action for each of the heads.  This is a bit
different and can create issues, for example is it OpenPnP's job to know
how much to lower nozzle to get item from a reel feeder or do you abstract
that to controller?  Maybe it would be best to have three levels:

1. Main PnP system and GUI
2. Machine translation
3. Controller server

Then PnP system would say "pick part" the Machine layer would then
translate this to positions and actions for controller?   However this
might be too big of change right off and just the universal
driver/controller would be good first step.

I would love to work on this feature and do a PR,  but I still have not
figured out the code structure of OpenPnP to point I feel I could do this.
Would it be possible to have a youtube video of a code walk through? I am
thinking of a video where you just open your editor and walk through the
code structure/files and discuss the model/structure of the code.

Again I am not a Java programmer and have done little with Object Oriented
programming. I personally like the low level bit banging of bare metal
programming. So a code walk through would be very helpful to get a mental
model of what is going on.

vonnieda, 2017-04-13:
Have a look at https://github.com/openpnp/openpnp/blob/develop/src/main/java/org/openpnp/machine/reference/ReferenceDriver.java. That is the entire interface between OpenPnP and the hardware, and it includes detailed documentation about how it works.

If you have specific questions about the codebase, I'm happy to answer them, but I don't think a walkthrough is a good use of my time. There are hundreds of classes in the codebase and it will take hours to go through them all. It's better to focus on one specific piece of functionality, learn it and ask questions as needed. There is also a pretty in depth overview of the software architecture at https://github.com/openpnp/openpnp/wiki/Developers-Guide#system-overview and this covers all the important subsystems and packages.

BendRocks, 2017-04-16:
Having just went through the process of writing a new driver for a new machine (TVM920 binary interface over ethernet) I'll say that it was pretty easy.  There were a few things 'extra' I had to do, like noting if a status report fetched over ethernet was dirty or not because sometimes the driver got asked the same/similar question over and over a lot of times in a row and each ask took about 1-2 mS round trip over ethernet. And I suspect anything connecting over a cable is similar. 

I also did some profiling where I bypassed delays in nulldriver just to learn the upper bound of placement speed on OpenPNP. On my desktop it was about 250K parts per hour. On my machine running the TVM it's probably 50K parts per hour (both figures without any vision). The upshot is that the design of OpenPNP won't be a limiting factor at all for running fast.

With everything learned to date, I'm background-building an 8 axis USB controlled FPGA motion controller. I've built several of these in the past and used them on different machines, so it's a path I'm familiar with. But this one will takes my learning on openpnp into account. It's a binary interface, register-based and there's the ability to queue commands in the FGPA. So, as soon as you finish getting to your pick location, the controller starts to drop the nozzle autonomously, and as soon as the nozzle is dropped, the vacuum turns on, then a delay starts etc. In other words, the first few commands issued to the driver by openPNP will return immediately as they are queued in the FPGA. And then when PC interaction is needed (eg vision check), the driver will wait until the queue is drained. 

So, the above might be an important thing to think about for a few driver: async operation--should it 100% in the driver, or does it make sense for the higher levels of OpenPNP to understand that the next 5 commands (move to XY, lower head, turn on vac, wait 100mS, raise head, move to new XY...) can all be a queued in the controller that requires zero interaction from the PC. 

Of course, software stop recovery gets harder. 

But the TVM gets its speed from having some of the these ops canned in the controller. You can see the difference in the way the machines runs. The TVM920 never pauses running it's native software. It's just bang-bang-bang. On OpenPNP, there is a small delay--maybe 10-30 mS--between each segment of a placement as the PC and controller figure out what to do next. The TVM running native is really impressive. The TVM running openPNP isn't there yet. 

And as you cross the 3600 cph threshold, 5 or 10 of these 20 mS stutters really add up. Which is why I think the queue in the controller is really important

trampas, 2017-04-16:
From what you have described the problem sounds like a single thread
problem. That is OpenPnP requests an operation from the controller driver
and waits for response, it does not do further work until the driver
responds. So OpenPnP does sends driver commands and then it has to process
data, while processing data the hardware is idle.

Additionally based on testing and numbers if OpenPnP can do 250k parts per
hour with no hardware delay it means it takes 14ms to process each part.
This again makes sense with your number if OpenPnP is running with single
thread design. That is after hardware has placed part, OpenPnP has ~14ms of
processing to before placing next part, which is in line with your 10-20ms
between segments.

This delay can be improved by running the controller driver in a different
thread/process on a multicore PC. This is like off loading to controller
hardware but without making new controller hardware. This way OpenPnP sends
the "pick command" and the the driver thread/process sends the move XY,
lower head, turn on vac, wait 100ms, raise head ...) while the driver does
this OpenPnP is free to go grab the next pick location and process and send
to the driver thread/process which queues them up and runs them.  This
again assumes the delay is not in the communications with the controller
hardware which I don't think it is.

The advantage of this system is that everyone gets a speed improvement
assuming they have multicore/multithreaded machine.

Trampas

BendRocks, 2017-04-16:
>> From what you have described the problem sounds like a single thread
problem.

Yes, if every driver call queued a command info a fifo and another thread worked to drain the fifo command agree it could be a big help as machine speeds climb. I'm not there yet, but hopefully in 6 months or so I will have some learning to report.

DouglasPearless, 2017-05-03:
@BendRocks I am very interested in your work on FPGA and motion control as I am looking at that from the point of view of the Smoothie project (not to mention some robotics work I am undertaking too) - I am trying to teach my self FGPA (it has been too many years since I worked at the gate level :-) ).

Have you published any designs for your previous work? 

Cheers
Douglas

markmaker, 2021-04-25:
This is more or less exactly what the MotionPlanner inside Advanced Motion Control now does (among other things). 

https://github.com/openpnp/openpnp/wiki/Motion-Planner

> Something to keep in mind is that we don't just have X, Y, Z, C. We have each of those axes for potentially unlimited number of tools. So there will need to be a mapping between tool axis and controller axis.

Covered by Axis Letter configuration and Driver mapping on the GUI configurable ReferenceControllerAxis.

https://github.com/openpnp/openpnp/wiki/Machine-Axes#referencecontrolleraxis

> From what you have described the problem sounds like a single thread problem. That is OpenPnP requests an operation from the controller driver and waits for response, it does not do further work until the driver responds. ... This delay can be improved by running the controller driver in a different thread/process ... This way OpenPnP sends the "pick command" and the driver thread/process sends the move XY, lower head, turn on vac, wait 100ms, raise head ...) while the driver does this OpenPnP is free to go grab the next pick location and process and send to the driver thread/process which queues them up and runs them. 

Addressed by GcodeAsyncDriver's writer thread, asynchronous operation and flow control, coordinated (and interlocked as needed) by the above-mentioned MotionPlanner.

https://github.com/openpnp/openpnp/wiki/GcodeAsyncDriver

closed 2021-04-25 by vonnieda