openpnp/openpnp #515: Jython (Python wrapper) adds a 2nd newline to the print-statement in the Log-TAB.

ldpgh, 2017-04-14:
Jython (Python wrapper) adds a 2nd newline to the print-statement in the Log-TAB. 
- Use of sys.stdout.write() is not a real option. It adds effort to convert everything to a string.
- In case of a Javax/ScriptEngine topic ... close this issue.
- Btw. using `from __future__ import print_function` seems to hide the output completely. But it smells like a Jython issue.

```
print "line 1"
print "line 2"
print "line 3"
print 1,
print 2,
print 3
```
will be in the  Log-TAB
```
2017-04-14 09:08:17 SystemLogger INFO: line 1

2017-04-14 09:08:17 SystemLogger INFO: line 2

2017-04-14 09:08:17 SystemLogger INFO: line 3

2017-04-14 09:08:17 SystemLogger INFO: 1
2017-04-14 09:08:17 SystemLogger INFO:  2
2017-04-14 09:08:17 SystemLogger INFO:  3
```

-------------------------
Desired result in the Log-Tab
```
2017-04-14 09:08:17 SystemLogger INFO: line 1
2017-04-14 09:08:17 SystemLogger INFO: line 2
2017-04-14 09:08:17 SystemLogger INFO: line 3
2017-04-14 09:08:17 SystemLogger INFO: 123
```
The script wrapper should not issue a newline.

vonnieda, 2017-04-14:
I believe this is actually a bug in the logging code if someone wants to look into it: #288

pfried, 2017-04-17:
I pretty much figured it out, and the necessary changes will be a bit ugly. The core problem is that the PrintStream class has private methods which i cannot override which add characters like the newline to the output. To avoid this i will have to "reimplement" / override a lot of its methods. Overriding only the `write()` method is not a good solution as it results in the issue described above.

@ldpgh Which OS are you running?

cri-s, 2017-04-17:
I have a mod that send output data to separate tab as is without adding anything if script panel is present and script for adding the tab. Would this resolve you issue?

ldpgh, 2017-04-17:
OS = Win7

pfried, 2017-06-04:
> Desired result in the Log-Tab
>     2017-04-14 09:08:17 SystemLogger INFO: line 1
>     2017-04-14 09:08:17 SystemLogger INFO: line 2
>     2017-04-14 09:08:17 SystemLogger INFO: line 3
>     2017-04-14 09:08:17 SystemLogger INFO: 123

I added a pull-request to address this issue, but i wanted to note here that the output will be `1 2 3` as the ScriptEngine somehow adds a space character in between the prints.

cri-s, 2017-06-04:
I have presumed stdout/stderr output was broken and don't work on logging panel.
Probable this is engine and maybe OS depend.

pfried, 2017-06-05:
yes, the problem was the `SystemLogger` which is hopefully fixed now

vonnieda, 2017-06-14:
Fixed in #579

closed 2017-06-14 by vonnieda