openpnp/openpnp #516: Deactivate controls during jobs

cri-s, 2017-04-14:
# Feature Request
## Describe the Feature
What do you want it to do? Be as descriptive as possible.
When machine is not started, disable all TabPane inside jogControlPanel.
When job is runnung (not paused) again disable all TabPane inside jogControlPanel.
Further disable Camera moves (maybe using glass panel) from camera panel because it interferences with
job processing. 
Further disable Menu, or part of it, Scripts, Edit, File, View .
## Why Do We Need It?
1. What problem does it solve?
It solve the problem of wrong move/actuator/script launching during assemble or that user added controls trought scripts don't get disabled.
2. Is it useful for everyone, or does it only solve a problem for your machine?
It is currently a UI problem for all.

cri-s, 2017-04-17:
If you want close this.

vonnieda, 2017-04-22:
Note: When this gets implemented we can remove the checkJobStopped() stuff that was added in #532.

closed 2018-05-01 by cri-s