openpnp/openpnp #517: modifing job and saving it as ...

cri-s, 2017-04-14:
## Describe the Issue
If loading a Job, modify placements and then save it with different name, as example for Variant assembly or to exclude something, normally the user expects that the job get saved correctly and not change the previous loaded job.
## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1) load a demo job
2) remove 50% of placement
3) store it as demo_reduced
4) load demo job
5) parts that was deleted for demo_reduced is now missing inside demo job.
## Expected Result
Expected result is that on modified board placement data if job is saved as different name , the orginal
job don't get modified.
## Actual Result
Orginal demo jog get  corrupted.

vonnieda, 2017-04-14:
This is by design. Jobs and boards are stored separately. When removing placements you are modifying the board, not the job.

closed 2017-04-14 by vonnieda

cri-s, 2017-04-14:
I know that this are different, and no it's not imposed by design.
There are already flags that tells the program, board is modified.
When saving job as , and board is modified, board need to be saved on
same directory but with different filename, adding as example a number
until File.exists(file) return false.
Is the the project still active or just necessary maintaince ?

vonnieda, 2017-04-14:
I disagree. Boards are fundamentally unique. In fact, I think we've had this discussion before. A board represents a revision of a particular physical PCB, so it makes sense that changes to it are reflected in other jobs that use that same board. If you want different versions of the same board, create those files separately and add them to a job.

The project is still active.

cri-s, 2017-04-14:
We don't had this discussion before, maybe you remember different
discussion that on my branch that remember placed parts and allow
placements in several steps i must copy the boards to separate files
if used multiple times on the same job.

vonnieda, 2017-04-14:
Yes, I understand. But that is not the correct way to do it. You should store job status in the job file, not in the board file. The board file is meant to reflect a single physical PCB. This is why it is shared in the job. Status of individual placements should be stored in the job file, and will be when #280 is finished.

cri-s, 2017-04-14:
You probably have right. I have this modifications from long time.
It was implemented by annother user, and i have extended it for
partial assembly.
Initial i used it because it updated the placed position during runtime.
Later it gives the possiblility to do partial assembly where more tray
feeders are required and
split assembly is necessary. I know it could be done creating
different jobs for different steps,
but depending on already setup feeders having it on single setup and
allowing with minimal clicks to start assemble of a batch of file, and
complete it later with different feeder mounted until it's complete is
really different as maintaing the different job state synchronized
with
the actual available feeders. Further as it stores the placement
inside file, it allows in case
of power outage having ups or laptop saving the actual placement and
restart later.
Updating position on gui is recently introduced, #280 store it in
file, but i use this for years, and as example, i can start assembling
during finishing assigning parts and the progressive
assembly is great, especially for prototype where there exists trays
of resistor decades and caps that need to be placed separatly on
machine, because it don't allow for more then 3 trays at same time.
For boards with many components, this is different as using ignore and
place.