openpnp/openpnp #522: Movement beyond rails

trampas, 2017-04-15:
Using the Jog controls OpenPnP allows movement in the negative direction after homing.  

It would be nice if OpenPnP when attempting to move out of the machine range (after homing) would limit operation and pop up a dialog indicating that the move was limited due to machine range.  Additionally it would be nice to have a check box on the dialog page to enable the limit or disable the limit.

vonnieda, 2017-04-15:
Duplicate of #308

closed 2017-04-15 by vonnieda

KarlZeilhofer, 2017-10-08:
If someone is interested in the configuration of a TinyG:  
TinyG has the ability to enable softlimits. Here is the HOME_COMMAND for our LitePlacer  

```
$xtn=-1
$xtm=565
$ytn=-1
$ytm=378
$ztn=-42.7
$ztm=1
G28.2 X0 Y0 Z0
G92 X0 Y0 Z0 C0
$sl=1
```
G-Codes outside this limits are not accepted then, and the TinyG goes into a soft-alarm state. To reset this, 
we have added a machine actuator called ClearTinyG, which sends a ```$clear``` to the TinyG. 

This also related to #621