openpnp/openpnp #523: Execute a script from inside a script

dzach, 2017-04-15:
# Feature Request
## Describe the Feature
Run scripts (any accepted language) from inside another script.

## How Will It Look?
NA, but it shouldn't give this error:
[image]

## Provide Examples
Should be able to run:
```
scripting.execute(new File(scripting.getScriptsDirectory().getAbsolutePath()+"/tests/hi.js"));
```
or similar, from inside another script, in any of the existing engines.

## Details
Is there a reason why `scripting.execute`, i.e.:
```
private void execute(File script)
```
is `private` instead of `public`?

Note: `scripting.on()` is `public` and allows the execution of scripts in the Events subdir. 

I have lightly tested it and it doesn't seem to cause any problem. I can create a pull request with this change, if that's ok.

vonnieda, 2017-04-15:
I'm fine with the change from private to public. Go ahead and change both execute() methods.

vonnieda, 2017-04-16:
Fixed in #524

closed 2017-04-16 by vonnieda