openpnp/openpnp #525: DRO Z display not tracking selected nozzle in drop down

BendRocks, 2017-04-15:
# Problem Report
## Describe the Issue

Changing nozzle in drop down in machine controls only shows Z for the first nozzle, not other nozzles

This occurs in both current develop branch and master branch.

## Steps to Reproduce
1. Start with clean XML
2. Add a second nozzle to the machine (named N2)
3. Drop the N1 down to -1mm
4. Switch to N2. DRO indicates -1mm

## Expected Result
Expected DRO to indicate Z setting of selected nozzle

## Actual Result
DRO only indicates Z setting of first nozzle. What is inconsistent is the XY values change based on selected nozzle (assuming non-zero offsets)

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
Not sure if this is desired behavior or not. But it's confusing to me as-is

vonnieda, 2017-04-15:
This is likely misconfiguration. Please attach machine.xml

vonnieda, 2017-04-16:
Ping @BendRocks, in case that notification didn't come through. Please send the machine.xml this is happening on.

BendRocks, 2017-04-16:
xml is attached. I repro'd this again on Dec 19 2016 (build 5a946e)--a full install downloaded from the OpenPNP site that also exists on my machine.

As with last time, I deleted all xml, ran the program, added the second nozzle, lowered the first nozzle, switched to the second nozzle, and it still showed the Z of N1. 

[machine.zip](https://github.com/openpnp/openpnp/files/924391/machine.zip)

vonnieda, 2017-04-16:
Hi @BendRocks. This is just a limitation of the NullDriver, not of OpenPnP itself. NullDriver doesn't track independent Z axes. If you test this same operation with GcodeDriver, for instance, with a proper axis setup it works fine. I verified it last night since I suspected this was what you were seeing.

I'm closing this as "wont fix", since it's basically by design. If you feel it's important that NullDriver support independent Z axes please file a feature request.

closed 2017-04-16 by vonnieda