openpnp/openpnp #526: Test and verify nozzle tip changer code

vonnieda, 2017-04-16:
I recently made a change (https://github.com/openpnp/openpnp/commit/cfde21257b1b47dbb3978dafd7802459b68ff483) that allowed for loading and unloading nozzle tips when the nozzle changer is not enabled so that users who change nozzle tips manually can do that via the UI, but I have some concerns that this might have broken nozzle tip changing during job runs in some way. I'm not sure how, yet. Just a hunch. So I need to go back to this code and consider it from the job run aspect.

vonnieda, 2017-04-16:
Okay, now that I am more awake:
The change skips all the nozzle change motions if the changer is not enabled and just sets the current nozzle tip. Previously this would have thrown an exception if the changer was not enabled.

So, this means that in the case of a user that is manually changing nozzle tips (with no changer enabled) they will not get an exception when they try to change a nozzle tip. This also means that a job will try to change the tip if needed and not throw an exception, but just change the tip. This effectively breaks manual nozzle tip changes for jobs.

So, a different solution is needed. Probably we either need to pass a "force" flag, or maybe just have setNozzleTip() which is only called by the GUI and not the job.

cri-s, 2017-04-16:
I have used a script to store nozzleChangerEnabled state, 
enable it, do nozzle change, restore nozzleChangerEnabled if user click on the dialog message OK,
otherwise remount old nozzle. That can be done from a java function too.
For manual without script, one have to enable the nozzleChangerEnabled and then disable it after nozzle change.

vonnieda, 2017-11-21:
Also see #118

commit c415b7d9 (2017-11-21): Fixes #526
closed 2017-11-21 by vonnieda