openpnp/openpnp #527: com.sun.javafx errors ?

dzach, 2017-04-16:
# Problem Report
## Describe the Issue
Until very recently I used the following code, found [here](https://blogs.oracle.com/nashorn/entry/setinterval_and_settimeout_javascript_functions), to implement the `setInterval` and `setTimeout` javascript functions. Recent changes make this code throw an exception.

## Steps to Reproduce
1. Copy the javascript below to a file, e.g. _test.js_, and put it inside the _scripts_ configuration subdirectory:
```
// Timer functions
var Platform = Java.type("javafx.application.Platform");
var Timer    = Java.type("java.util.Timer");

function setInterval(func, milliseconds) {
	// New timer, run as daemon so the application can quit
	var timer = new Timer("setInterval", true);
	timer.schedule(function() Platform.runLater(func), milliseconds, milliseconds);	
	return timer;
}

function clearInterval(timer) {
	timer.cancel();
}

function setTimeout(func, milliseconds) {
	// New timer, run as daemon so the application can quit
	var timer = new Timer("setTimeout", true);
	timer.schedule(function() Platform.runLater(func), milliseconds);	
	return timer;
}

function clearTimeout(timer) {
	timer.cancel();
}

function fun() {
  print("ok");
}
setTimeout(fun,1000);
```
2. Run openpnp.sh from a command line.
3. From openpnp menu bar run Scripts->test.js
4. Observe errors in the console

## Expected Result
No error should happen. Output should be:
```
ok
```

## Actual Result
Errors happen:
```
Exception in thread "setTimeout" java.lang.IllegalStateException: Toolkit not initialized
        at com.sun.javafx.application.PlatformImpl.runLater(PlatformImpl.java:273)
        at com.sun.javafx.application.PlatformImpl.runLater(PlatformImpl.java:268)
        at javafx.application.Platform.runLater(Platform.java:83)
        at jdk.nashorn.internal.scripts.Script$Recompilation$134$796$Utility.setTimeout#L:27(/home/dz/.openpnp/scripts/utils/Utility.js:27)
        at jdk.nashorn.javaadapters.java.util.TimerTask.run(Unknown Source)
        at java.util.TimerThread.mainLoop(Timer.java:555)
        at java.util.TimerThread.run(Timer.java:505)
```
## Notes
Recent changes removed the dependency on `javafx`. However, the code above uses 
```
var Platform = Java.type("javafx.application.Platform");
```
Could the error be due to some kind of mismatch?

vonnieda, 2017-04-16:
See http://stackoverflow.com/questions/11273773/javafx-2-1-toolkit-not-initialized. The bottom line is that since we don't initialize JavaFX anymore (by creating a JFXPanel) then JavaFX is not initialized. You can either create a JFXPanel or a JavaFX Application to initialize it, but I would recommend not using it at all, instead. There's no value in running your code on the JavaFX application thread. Just create a new Thread, or use ScheduledExecutor, or SwingUtilities.invokeLater. There's lots of ways to run a function on another thread without requiring JavaFX.

Closing as this is a user script error, not an OpenPnP issue.

closed 2017-04-16 by vonnieda

dzach, 2017-04-16:
Thanks for the reply @vonnieda .
I found [this library](https://github.com/nikku/nashorn-async) that implements `setTimeout` and `setInterval` and does not depend on javafx. Seems to work fine.