openpnp/openpnp #528: question: jobprocessor

cri-s, 2017-04-18:
Can you please explain the canSkip and skip interface and the future of TestStatusListener in relation to the Bus interface? It is correct that there is no Run interface, just Step (next)  and that between next() placement data can be changed, or is supposed to be not the same ?l

public interface JobProcessor {
    public interface TextStatusListener {
        public void textStatus(String text);
    }

    void initialize(Job job) throws Exception;
    public boolean next() throws Exception;
    public void abort() throws Exception;
    public void skip() throws Exception;

    public boolean canSkip();

    public void addTextStatusListener(TextStatusListener listener);
    public void removeTextStatusListener(TextStatusListener listener);
}

vonnieda, 2017-04-18:
Hi @cri-s, yes, there is no run, just next(). It's expected that a caller just calls next() over and over. And yes, whenever next() is not being called, the machine is "paused" and you can make changes.

I don't plan to switch the TextStatusListener over to the Bus interface at the moment, but it probably will happen sometime in the future.

These interfaces are going to be gone in the next few weeks, so I don't think there is a lot of value in discussing them too much. It was a bad design to start and I'm getting rid of it.

Here is the new set of interfaces I've been working on. These aren't finalized yet, and may change. I'm hoping to have a test version of this up in the next week or so.

```
package org.openpnp.spi;

import org.openpnp.model.Job;

public interface JobProcessor extends PropertySheetHolder, WizardConfigurable {
    void initialize(Job job) throws Exception;

    /**
     * Shortcut for next(null).
     * @return
     * @throws Exception: Note that a JobProcessorException can be thrown instead, which can
     * include Commands to help resolve the problem.
     */
    boolean next() throws Exception;
    
    /**
     * Performs the given command. If the command is null the processor will perform
     * the next logical command given the state of the processor.
     * @param command
     * @return
     * @throws Exception: Note that a JobProcessorException can be thrown instead, which can
     * include Commands to help resolve the problem.
     */
    boolean next(JobProcessorCommand command) throws Exception;
    
    void abort() throws Exception;
    
    public void addTextStatusListener(TextStatusListener listener);

    public void removeTextStatusListener(TextStatusListener listener);
    
    public interface JobProcessorCommand {
        String getName(); // Skip Placement, Skip Board
        String getDescription(); // Skip the current placement, Skip all placements on the current board.
        void execute() throws Exception;
    }
    
    public class JobProcessorException extends Exception {
        protected final JobProcessorCommand[] commands;
        
        public JobProcessorException(Exception cause, JobProcessorCommand... commands) {
            super(cause);
            this.commands = commands;
        }
        
        public JobProcessorCommand[] getCommands() {
            return commands;
        }

        @Override
        public String getMessage() {
            return getCause().getMessage();
        }
    }
    
    public interface TextStatusListener {
        public void textStatus(String text);
    }
}
```

The basic idea here is that you call next() and it fails it throws an Exception. If the Exception is a JobProcessorException it may include a list of Command that you can call to fix the problem. Examples would be Skip Placement, Skip Part, etc.

I'm closing this since it's not an issue, but we can still discuss here if you like. It won't stop comments from being posted. In the future, development discussions would be better on the mailing list.

closed 2017-04-18 by vonnieda

cri-s, 2017-04-18:
I see, it goes to a direction that is the opposite direction of my requirement.
Why do you do that. I don't will you convince in the other direction,
just talking about that.
I suppose you maybe future code holds the assembly state and is able
to stop from assembly and start to finish it. This because you have
made some statements again that in the past. Having that, why you need
sofisticated skip codes ?

My understanding is, if job is started, it could be paused for
feeder/... issues, but job is not modificable after start. If
modification of job itself is requested, it must be stopped and
restarted again.

Cris

vonnieda, 2017-04-18:
Reasoning: https://groups.google.com/d/msgid/openpnp/68a93f18-1be8-41d8-abd6-d1e4e1a76c20%40googlegroups.com?utm_medium=email&utm_source=footer

Most people wanted additional options.

Yes, the goal here is that Job will hold the state of the job run, and it will be possible to restart the job to finish it if placements are skipped.

You will be able to set an option on a Placement to determine how to handle an error: Skip or Raise. If set to Skip, the job will just skip the placement, store the error and keep going. Raise will offer the options listed in the Commands.

So, there will be two ways that people can work:
1. Mark all placements error recovery to Skip.
2. Run job. Some placements may have errors.
3. When job finishes, look at errors, fix any you want and then run job again.

This will be better for production work.

or

1. Mark some placements error recovery to Raise.
2. Run job. If a placement has an error the user will get options to continue.
3. Fix problems as you go, or choose to skip as needed.
4. Re-run job if you need to.

This will be better for prototype work.

Currently there is nothing that stops you from modifying from job during a run, but you are right, it would probably be better if we locked that stuff down since some changes might not be picked up due to internal state.

If these options don't work for you, let me know why and we can try to figure it out.

cri-s, 2017-04-18:
Example of code that i have  written lately. assembly(....) have a
boolean flag named calc.
If that is set to true, it don't move, return only the total distance
it need  to travel including
all optimisations that it made after assignment of placemnt to nozzles
and at the same time
it numerate the placements order that it get assembled.
The real reason why placements could not be modified is when assembly
have several steps,
example resistor placed near ethernet connector with integrated magnetics.
The eth connector cannot be placed first, it must be placed after the
resistor, and if the placement of resistor is skipped for some reason,
the placing must be interrupted in order to correct the situation,
because after placing the eth connector the resistors cannot be
placed.
This is not the only situation where this occur, Other situation is
when placing parts higher as nozzle height clearance/2 - 0.3mm , ie
where it is not possible to drive over already placed high components
with high component on nozzle. If basically this two situation occur,
after a certain point changing assembly sequence is not possible. The
initial optimisation can be only initial, but after some code, the
placement get divided into two lists, placeable and not placeable
parts with the current mounted nozzleTips. If placement data can be
changed a lot of code need to be called, for each placement solution,
how to break down the solution to single pick/place, i actually don't
have idea. If user drop part and change nozzle, this means more code
need to be called, more then i have thinked first and additional logic
is required to consider the already picked parts. Doable in part, the
question is why. The error rise or delayed raise for me don't have to
do with different interface, or i miss something.
If part cannot is silently skipped and error is thrown later or at
error rising time, it's just a
boolean flag and the list of errors that store the exceptions that are
reissued later at necessary  time point.

vonnieda, 2017-04-18:
@cri-s This is very confusing. I'm not sure if you are saying there is an issue or if you are disagreeing. Given what I outlined above, will this work for you? And if it won't, can you say why in simple terms?

cri-s, 2017-04-18:
Beside i optimize placement a bit , there are 3 classes or steps of placements,
in reality is a bit more complicated but it can be simplified to 3
cases for understanding:
1) parts that can always be placed, generally parts under 3mm height.
2) parts that need to be placed before other parts
3) parts that need to be placed in sequence

case 2 include components placed inside other components, as example
resistors/caps placed between pins of ic, components placed under
modules, example switching power modules from TI, components placed
inside carriers.

Take a look at this pcb:
https://wiki.openwrt.org/_media/media/tplink/tl-wr703n/tl-wr703n_pcb_top.jpg
R104, C113 maybe D1 need to be placed before the USB connector.
The resistor near the led on the eth connector need to be placed
before eth connector.
R104 C113 D1 could be placed at the same manner near eth, connector,
sometimes pcb space requirement mandate that.
Annother example with gerber file, pnp centeroid data
http://www.cypress.com/documentation/reference-designs/ez-usb-gx3-usb-type-gigabit-ethernet-reference-design-kit
R10, C15 L... must all  be placed before eth connector.

cas3 is for high parts, as example this:
http://www.link-pp.net/sdp/1132867/4/cp-7448953/0/LPJU4130AONL_RJ45_Connector_with_10_100Base-T_Inte.html
That must be placed in sequence from camera , far placement first,
near last generally with
manhattan style move for the last 30mm or less.

vonnieda, 2017-04-18:
@cri-s I'm not sure what all this has to do with the changes I'm proposing. I realize you have very specific requirements for your setup, but I don't think that is on topic for the questions you asked above.

The purpose of the changes is to improve retry and error handling. This is documented here: https://github.com/openpnp/openpnp/issues/280

Right now my goal is just to fix the poor error handling, the badly designed FSM and the inability to support retry on pick failures.

cri-s, 2017-04-18:
Modifying sequence having not only case1 is problematic to not
possible after stage1 of assembly have completed.

2017-04-18 19:00 GMT+02:00, Cri S <phone.cri@gmail.com>:

vonnieda, 2017-04-18:
Oh, are you referring to changing job data while the job is running? If so, yes, I agreed above that we probably need to restrict that.  You also filed a ticket for that: https://github.com/openpnp/openpnp/issues/516, which I agree, we should fix.

cri-s, 2017-04-18:
Maybe language problems. Back to the interface.
I have asked about interface to jobprocessor especially about two
points, where the point of
canSkip and Skip was not answered.

On the old interface, if a feeder is not available, and skip is
pressed, it can be assumed to
skip all parts from that feeder.
If the error is from elsewhere, bottomvision, vacuum, .... only the
part have to be skipped.
For me the skip/canSkip interface is not full clear, but as new
interface is coming, it may be
irrelevant.

On new interface, as i have understand there is skipBoard,
skipPlacement, skipPart ...
and possibly others with more complicated interface where probably the
user tells what to skip.

Further there is not really clear what skip/canSkip mean, requeue for
assembling after all other assembling, then canSkip make sense, or
skip for the assembly job.

vonnieda, 2017-04-18:
Old Interface:
* skip: Discard part on nozzle, remove planned placement, mark placement skipped.
* canSkip: Whether Skip option should be shown to user, and if skip() can be called.

New Interface:
* skip and canSkip are removed.
* No hard coded skip option, but skip can be included in options to the user. Errors that are related to a placement can include skip placement or skip part. Skip placement skips just that one, skip part skips all placements for that part.
* Calling next(null) will default to whatever option lets the job continue to the next step. In most cases this will be "skip placement".

Like I said, new interface is not finished yet. I am working through all the use cases and may make changes before it's released. I'll be posting a test version when I think it's ready for someone to try it.

In both cases reassembly is not covered. That will be handled on it's own in a later update.