openpnp/openpnp #529: Squareness Compensation is ignored when coordinate doesn't change

vonnieda, 2017-04-19:
Due to how GcodeDriver skips axes that aren't involved in a move, if a movement happens in the axis that does not affect non-squareness compensation, it's possible the compensation will not be applied in repeated moves to only one axis.

vonnieda, 2017-04-19:
This should fix it. Need to test: 
[image]

vonnieda, 2017-04-19:
Note: This probably breaks backlash compensation, too.

vonnieda, 2017-04-19:
Another note: This likely means that the position report code is broken, since it doesn't subtract the non-squareness value. Probably need to think about refactoring all of this to maintain controller state and internal state coordinates.

vonnieda, 2017-04-21:
@sadman suggested the following fix: https://github.com/sadman/openpnp/commit/f8d1a9df8ecc39ec69dc60f5f90c707ef6f52b6e

aauer1, 2017-04-27:
I also submitted a pull request for this issue #533 . It solves the same issue. Sorry for the duplicate.

vonnieda, 2017-04-30:
This is now fixed. I used the basic idea in #533 but refactored the code to be easier to understand and also handle an edge case that was not handled in the original PR. Please test and report back.

commit 400e21fb (2017-04-30): Merge branch 'aauer1-develop' into develop

* aauer1-develop:
  Fixes #529: When a move happens in Y, if non-squareness compensation is enabled, a X move will be forced. Also refactored the decision tree for which axes to include to make it easier to understand and expand in the future.
  Fixed an issue with the non-squareness factor: If there was a movement only in y-direction the non-squareness factor wasn't considered. Fixed it in the GCode driver.
commit b166c63a (2017-04-30): Fixes #529: When a move happens in Y, if non-squareness compensation is enabled, a X move will be forced. Also refactored the decision tree for which axes to include to make it easier to understand and expand in the future.
closed 2017-04-30 by vonnieda