openpnp/openpnp #540: Does the ScriptRun pipeline stage work with BeanShell scripts?

dzach, 2017-05-07:
# Problem Report
## Describe the Issue
I'm trying to run a simple script written in BeanShell java, but I am getting an error:
```
java.lang.Exception: Unable to find scriping engine for .openpnp/scripts/pipeline-test.bsh
```
It seems that openpnp does not recognize the `.bsh` file extension as a possible ScriptRun script, while it does recognize it as a script. 

## Steps to Reproduce
1. Place any file with a `.bsh` extension into the scripts directory.
2. start OpenPnP or the CvPipelineEditor and start editing a pipeline.
3. Add a new ScriptRun pipeline stage to the pipeline being edited.
4. Set the scipt file to be used by the stage
5. Observe the error in the lower right pane of the pipeline editor.

## Expected behavior
BeanShell scripts should work with the SriptRun pipeline stage, since they do work from the Scripts menu.
 
## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
Scripts work fine otherwise.

vonnieda, 2017-05-07:
@dzach Are you running the most recent code? There was a BSH scripting problem about 3-4 weeks ago that has been resolved.

cri-s, 2017-05-07:
Have you tested it on openpnp, not standalone editor, as this are different and advanced algorithm or loose feeder pipeline don't work on standalone editor.

cri-s, 2017-05-07:
The bsh problem was a win only problem, on Linux it worked to my knowledge.

dzach, 2017-05-07:
I get the same error in both the stand alone pipeline editor and the OpenPnP pipeline editor. But `.bsh` scripts work fine from the menu.

vonnieda, 2017-05-07:
Okay, I see what the issue is. We don't use the same scripting engine instance for the ScriptRun stage as OpenPnP uses so that it can be used outside of OpenPnP. This means that the bug fix for the BSH problem didn't make it here. We just need to add the BSH hack in Scripting.java to ScriptRun.java.

vonnieda, 2017-05-07:
Okay, pushed a fix. Let me know if it works for you please @dzach

closed 2017-05-07 by vonnieda

dzach, 2017-05-07:
Yes, I don't get the error any more and I am able to run scripts.
Thanks @vonnieda !

commit 80eaf9eb (2017-05-07): Fix #540 by adding the BSH hack from #462 to the ScriptRun stage.