openpnp/openpnp #543: Can we detect part orientation with the pipeline stages as they stand today?

dzach, 2017-05-08:
# Feature Request
## Describe the Feature
Descriptions of methods and OpenCV pipelines that achieve part orientation using either the top or the bottom camera.

## Why Do We Need It?
One of the problems with bottom vision and feeders like the LoosePartFeeder is that "polarized" parts may not/do not have a known orientation, e.g. loose parts, or parts that for some reason moved while at the nozzle, et.c. There seems not to be any well known method (to my knowledge) to detect the orientation of such parts automatically. E.g. pin1 of an IC, the cathode of a diode or the + or an tantalum capacitor et.c. It would be helpful to create a space to collect user experience regarding this issue that can lead eventually to more or less successful "prescriptions" for OpenCV pipelines for orienting certain types of parts. These "prescriptions" can then be moved to a more visible place, possibly to some page on the Wiki.

## Provide Examples
By using certain pipeline stages appropriately it may be possible to achieve results like this:

[image]

The above is work in progress, but is indicative of the desired result. 

Is there some community knowledge that can be shared here?

cri-s, 2017-05-08:
you need lower illumination intensity or add more indirect light. The atmel sign and component id is
damaged at 50%, the production date can be read. This using logaritmic intensity with add -27  gain 14.
Maybe spots on you'r lightning setup ? Annother possibility is adding annother colored darkfield light
for loose feeder.  As the marking can be read to over 50%, finding pin1 hole is no problem.
My 2C, problem is not pipeline but you'r lightning setup.

vonnieda, 2017-05-08:
I've given this quite a bit of thought in the context of loose part feeding. I've mentioned this elsewhere, but for posterity's sake, here is my suggestion for loose part feeding, i.e. top recognition:

First, the part has to be trained:
1. Place a single part in the view of the camera, on a solid background. Make sure pin 1 is in a common place, e.g. top left, top right, etc.
2. Run a pipeline similar to bottom vision to get the contours and minAreaRect.
3. Using the minAreaRect, copy the part of the image surrounded by the rect, this becomes the part template.
4. Rotate the template by the minAreaRect's rotation to remove any minor rotation from the captured image.
5. Store the template.

Now we can recognize parts:
1. Use a pipeline similar to current loose part feeder, which finds min area rects for all visible parts in the camera view.
2. Filter the rects by ones that are too large or too small to fit the template with some tolerance.
3. For reach one found, rotate the template by the angle of the rect.
4. Mask out everything but the part in question.
5. Run a template match four times, rotating the template 90 degrees each time.
6. Whichever template match of the four is the best tells us the orientation of the part.
7. Using the template match angle and the rotated rect, we can derive the position and orientation of the part with respect to the training image.

I've performed some simple tests with this algorithm a year or so back and it seemed like it would work well.

Bottom vision can be done the exact same way, but I think that is less valuable. As long as we pick the part at < 45 degree error, bottom vision doesn't need to know pin 1.

dzach, 2017-05-08:
Thank you both @cri-s and @vonnieda . It'll take me a while to grok the details, and I hope to finish my example above and be able to compare results.

vonnieda, 2017-05-08:
@dzach There was one other thing I wanted to mention. This is just my $0.02, but I think trying to approach this by building up features (like circles at a certain position) is the wrong way to go about it. It will require a custom pipeline for every part and will be an intensive development process to identify features. Using matchTemplate is generic and should work for all parts. Another option is SIFT / SURF, which solves the same problem but in a different way.

dzach, 2017-05-08:
Yes, it probably is, but I have failed miserably with matchTemplate and wanted to create some stage infrastructure before I retry. The above example is just a place holder, which happened to give a positive result. SIFT / SURF are not present in OpenCV 3, they have moved to some third party repository which I think I have not included in my present binary. I haven't yet tried the 2.4 version that's included in OpenPnP.

cri-s, 2017-05-09:
user Orb/FLANN .... instead of SIFT/SURF if you want experiment with that.
If you correct you'r lightning you have better results.

dzach, 2017-05-09:
I've got some more questions. Hopefully, these can be constructive for the issue at hand.

I assume the pick process expects an ArrayList of RotatedRects from the LoosePartFeeder pipeline, so it can proceed with choosing the closest one and then picking up the part and placing it, or do whatever it is requested to do.

Questions:
- Is openpnp using a 360° (or +-180deg) orientation in the results of the LooseFeeder pipeline? Because for polarized parts this should clearly be the case.
- If it is so, then what is the "formal" orientation for each part and where does this appear? E.g. for ICs, will that be the lower left corner of the chip? 
- What about other polarized parts? 
- Is there a document that describes such details, or, if not, what kind of document would that be?

So, say that my demo pipeline above manages to derive the true orientation for a part. If that ends up in a `rotatedRect.model.angle`, will that result in a correct placement of the part on the board?

vonnieda, 2017-05-09:
The code for ReferenceLoosePartFeeder is here: https://github.com/openpnp/openpnp/blob/develop/src/main/java/org/openpnp/machine/reference/feeder/ReferenceLoosePartFeeder.java#L70

It's very simple, and easy to follow, I think. Line 70 is the feed method, which is called for feed and then getPickLocation() is called to pick.

It currently picks at the angle returned by the RotatedRect. In OpenPnP there is no particular standard for angles. It is expected that the user picks a standard that works for their workflow and sticks to it. This is usually driven by your CAD software. The bottom line is that once proper angle support is added, we'd probably want to add an angle offset field to the feeder so that the user can set the angle that they expect a part to be at after pick.

As far as I know there is no standard for this stuff.

And finally, yes, if your rotatedRect.model.angle is correct, placement will be correct assuming that you use the same relative angles between feeders and boards.

cri-s, 2017-05-09:
The looseFeeder expect a list of roatedRects.
Orientation, that depends on nozzle configuration. Further where pin1
is, lower left corner,
upper left corner, that depends on IPC guidelines used and own
definitions, as not all
package are covered with Jedec specifications. Orientation is always
recalculated based
on settings and imported data, -10 is equivalent to 350 deg and is
equivalent to 710 degr.
Extract from actual IPC specifications: Angle is always CCW

1) Chip Capacitors, Resistors and Inductors (RES, CAP and IND) – Pin 1
(Positive) on Left
2) Molded Inductors (INDM), Resistors (RESM), Molded Polarized
Capacitors (CAPMP) – Pin 1 (Positive) on Left
3) Precision Wire-wound Inductors – Pin 1 (Positive) on Left
4) MELF Diode – Pin 1 (Cathode) on Left
5) SOD Diodes – Pin 1 (Cathode) on Left
6) Aluminum Electrolytic Capacitors – Pin 1 (Positive) on Left
7) SOT Devices (SOT23, SOT23-5, SOT223, SOT89, SOT143, etc.) – Pin 1 Upper Left
8) TO252 & TO263 (DPAK Type) Devices – Pin 1 Upper Left
9) Small Outline Gullwing ICs (SOIC, SOP, TSOP, SSOP, TSSOP) – Pin 1 Upper Left
10) Ceramic Flat Packs (CFP) – Pin 1 Upper Left
11) Small Outline J Lead ICs (SOJ) – Pin 1 Upper Left
12) Quad Flat Pack ICs (PQFP, SQFP) – Pin 1 Upper Left
13) Ceramic Quad Flat Packs (CQFP) – Pin 1 Upper Left
14) Bumper and Plastic Quad Flat Pack ICs (BQFPC, PQFPC Pin 1 Center)
– Pin 1 Top Center
15) Plastic Leaded Chip Carriers (PLCC) – Pin 1 Top Center
16) Leadless Chip Carriers (LCC) – Pin 1 Top Center
17) Leadless Chip Carriers (LCCS Pin 1 on Side) – Pin 1 Upper Left
18) Quad Flat No-Lead ICs (QFN) QFNS & QFNRV, QFNRH – Pin 1 Upper Left
19) Ball Grid Arrays (BGA) – Pin A1 Upper Left

Check out https://blogs.mentor.com/tom-hausherr/blog/tag/ipc-standards/ ,

Furter some notes on unmodified LooseFeeder
        for (int i = 0; i < 3; i++) {
            pickLocation = getPickLocation(camera);
            camera.moveTo(pickLocation);
      }
This make no sense. One can move camera to getPickLocation() and if
nothing is found
retry at location, doing 3 times the same vision recognition and then
return always the
same part make no sense.
Further the returned part should be choosen randomly in order that if
feeder need to retry it don't retry on the same part that may be
inclined and because this don't pick up.

dzach, 2017-05-09:
Thanks very much @vonnieda !
 I see the OpenPnP demo board v2 has a number of polarized parts on it, and there is a TQFP32 there, too. And, at least the F.Fab layer in KiCAD indicates clearly their orientation. Looking good so far.

dzach, 2017-05-09:
@cri-s that's a load of info, thank you! It gives an answer not only to my question, but also says how to orient the parts when creating a package.

dzach, 2017-05-09:
Regarding the LoosePartFeeder, one of the problems I face is that sometimes the camera picks some noise outside the tray, and the feeder spends some of its 3 tries looking for a part at a place there will never be one. Maybe the area the camera is allowed to move should be confined by some setting, e.g. corners of a rectangle or similar.

cri-s, 2017-05-09:
Normally loose feeder have certain background color and vision mask out
everything that is not surrounded with that color/gray value. Eventually
disable that loose feeder can move camera for searching components. For
checking orientation it should move if necessary.

dzach, 2017-05-10:
Nice trick, I can do this now with `MaskModel`.  How would you do it without it, with the stock stages?

cri-s, 2017-05-10:
Extract background color with maskhsv
Find contours
Set color 255 white
Draw contours black -1 filled
Draw contours white 255
This is now the mask having only valid components in black and every else
is white.
Find contours
Image recall maybe bw
Draw contours black
Result is image with component and background or everything else set to 0.

dzach, 2017-05-10:
Ah, yes, you already suggested that.

dzach, 2017-05-11:
@cri-s 
Could you please describe the stages involved in this result from #539 ?

[image]

cri-s, 2017-05-11:
Standard vision processing using basic morphological operators.

Rectangle ( filter out led + ... , standard stage)
Rotate image based on biggest rotatedRect, red Line show the rotatedRect.
Pixel erode-dilate to group the two single reflexion of single pin.
DrawBoundingBox
For visual presenting only add .-128 and draw pin contours with -1
size in white.
Later used, not shown, pins ( draw 11 errors in this case and return
error object (String)
in order to differentiate from "no component found" )

cri-s, 2017-05-11:
As noted previously the pin error is probably caused by off the axis
optical lens mounting or insufficient lightning. The optical lens axis
error can be corrected either by adjusting the error
or by modify lightning patttern/angle.

2017-05-11 19:46 GMT+02:00, Cri S <phone.cri@gmail.com>:
> Standard vision processing using basic morphological operators.
>
> Rectangle ( filter out led + ... , standard stage)
> Rotate image based on biggest rotatedRect, red Line show the rotatedRect.
> Pixel [...]

dzach, 2017-05-11:
Thank you @cri-s . 
The IC in the picture is one that I took off a broken board, and there are still be traces of solder on the pins. I used just for testing an TQFP32 package.

Lighting is far from optimum, but has improved quite a bit with this new positioning of the top camera and the 4 LEDs farther apart. The problem is that the top camera, a stripped off Philips SPZ5000 USB webcam, is in the middle of the two nozzles with very tight spacing and whatever I put there apparently gets in the bottom vision field of view. I've tried a diffuser, but the mere fact the it was white made things more difficult.

With the bottom camera, which is the same type as the top one, lighting is a lot easier to correct. At present there is a 60mm LED ring at about the level of the PCB bracket and the top of the feeders. That's about -26.5mm from safeZ. I have found a spot for the nozzle where bottom lighting seems optimum, but of course all LEDs on the same plane facing upwards. This may change in the future by installing a "light well" of some kind, maybe as a first job for this pnp machine.

cri-s, 2017-05-11:
60mm led ring is too wide. 
Install  4 additional leds as small as possible near the lense.
It don't need to be high power leds, normal leds are fine.

cri-s, 2017-05-11:
you can use reflecting cardboard to test reflections, eventually mount
a alu foil on cardboard.

dzach, 2017-05-12:
I'll try the reflective cardboard idea and see how it goes, before I start designing a new fixture.

On the subject of part orientation, it seems that there is a need to know where is a mark expected to be, e.g. what quadrant of the part body it should be located. In @vonnieda's procedure above, this might not be needed for the pipeline, but something similar might be needed to relate the orientation of the part on the nozzle with the part on the board and the part on the feeder. The part package builder might be the most appropriate place to enter such info, although at present it does not have an entry for the position of the mark. If this info were to be stored together with the rest of the package dimensions, it could be easily retrieved by any part of the system.

But then, there are also the template images that need to be stored somewhere and retrieved when needed.

cri-s, 2017-05-12:
Feeder data have the angle. Loose part feeder need to be setup with part
and during setup it have that info if you rotate the image. I remember you
don't do that.

cri-s, 2017-05-12:
What motion controller you have?

dzach, 2017-05-12:
The motion controller is Ramps1.4  with Marlin.
Feeder rotation says how the part should be rotated to correspond to the part on the pcb but it does not directly relate to the mark on the chip for zero-orientation.

With the loose feeder, this info is not relevant, I think. Unless someone puts all loose parts in some order.

cri-s, 2017-05-12:
No , Feeder Rotation says how the part is rotated relative to the
rotation convention.
Placement rotation says  how the part should be rotated to correspond
to the part on the pcb.
For loose feeder it says how the part that is used to setup loose
feeder  are oriented respect to the rotation convention. And as
consequence the rotation of  the templates stored inside the feeder.
The returned location is:
  loc = findPart(cam,nozzle,part);
  pickLocation = location.addWithRotation(loc).subtract(location);

The location for the setup parts is first normalized, ie roation is
always in multiple of 90 degree mostly, luxeon star led as example
uses multiple of 60 degree.

dzach, 2017-05-12:
What do you mean `how the part that is used to setup loose feeder`? Do you refer to @vonnieda 's description above or to something else? I was under the impression that in the loose part feeder the parts are just ... loose, and there is no `setup` involved. Any link to educate myself?

So for the loose part feeder, say we find the part and the mark on it says the part is at 60°. Assume the rotation convention for that chip is 270°. Then the part will be rotated 270-60=210° (or 360-210 = 150°) and then, to place it, it will add the placement rotation.

Is this correct?

dzach, 2017-05-12:
If I got the above right, it seems a little  confusing to have the `rotational convention` (i.e. the zero orientation angle) for each package expressed as the rotation of the feeder. It seems to me that these are two distinct properties, for two distinct concepts. 

Maybe thats what @vonnieda means by 

> The bottom line is that once proper angle support is added, we'd probably 
> want to add an angle offset field to the feeder so that the user can set the 
> angle that they expect a part to be at after pick.

But I can't see why the angle offset should be added to the feeder and not the package, where it belongs. An maybe name it `zero-orientation` too.

cri-s, 2017-05-13:
Placing a part on rotation 0 must place the part in respect to the used rotational convention.
When using upper left part as pin1 reference, normally the camera is set with rotation 90 degree
if camera image height is smaller as camera image width. 
Feeder rotation is the rotation relative to rotational convention of the part.
If all is setup correct, you don't need additional rotation offset field,
Now, probably the part you miss.
For LooseFeeder you need reference image for templatematch, and contours of Pins + Outline.
How to get this data ? It is a Stage that return this data if data is present, otherwise it create the data..
That "reference" script, if args contains the img name it returns the data, otherwise it used the vars and
commands inside args to create the reference data.
Having that reference Image, you have a Reference rotation that you can change and adjust.
Now the LoosePart feeder is the same as every other feeder because you now have a immutable camera image where correct rotation is returned.
If you do circle lookup on reference image, it must be in the same quadrant as
the captured image, if not the angle need to be tweaked.
Bottom camera don't need to know rotation angle, it is enought to correct the +-22 degree of max allowed angle error. Yes, the rotational convention at the end is setup on feeders, but have nothing to
do with part data, and the same part from the same producer is available on tape having different quadrants for pin1 because there exists two standards.It's possible you have two reels on
two adjacent feeders that have different rotation because pin1 is orientated differently inside tape carrier.

dzach, 2017-05-13:
@cri-s , here is how I understand what you describe above:

> Placing a part on rotation 0 must place the part in respect to the used rotational convention.

Yes. This is the principle that governs the rest with regard to the orientation of the part. I prefer the term "zero orientation", according to the documents you provided above, to the term  "conventional rotation", because I understand it as saying clearly: "when you place the part with a _zero orientation angle_, it should look like this.". 

> It's possible you have two reels on
> two adjacent feeders that have different rotation because pin1 is orientated differently inside tape carrier.

This is understandably set up in the _feeder_ and not in the _package_ that describes the part.

> the same part from the same producer is available on tape having different quadrants for pin1 because there exists two standards

Quadrants refer to the part's  _package_ if I understand you correctly. Then, if there exist two different _standards_,  there ought to exist two different _package_ specifications.

> Feeder rotation is the rotation relative to rotational convention of the part.

Now, I understand that practically the whole thing _works_ by tweaking the feeder's rotation. This doesn't change the formal specification of a _package_ and its "zero orientation", it merely hides this information in the feeder rotation setting.

Feeder rotation is the rotation relative to the machine's coordinate space, and may practically include a (conceptual) offset for the orientation of the particular part in the loaded tape. The feeder holds a part which is the one that has a "conventional rotation" characteristic, or a "zero orientation", and delivers that part in a final rotation that takes both into consideration. This "zero orientation" of the part is a characteristic described in its _package_ specification and does not change with the feeder. The feeder rotation (actually the offset for the particular part loaded)  may change with the package it holds to deliver the part in its "zero configuration". The two entities are described independently. The information that apparently is missing today from OpenPnP is each package's "zero orientation" and the conceptual offset of the part inside the tape/feeder.

Do we really need the "zero orientation" of a package, if we can practically get away without it? I'd say yes, because it is a piece of standard information that can be found in a part's specification, and is easy to find. It make things clear and formal, and removes confusion caused by hiding information inside ad-hoc settings. I'd prefer to have this information available as well as the offset of the particular part in the tape, because then all entities are described clearly, and make sense. If someone does not want to go into the trouble to enter this data, then one can always set the offset to compensate for it.

Just my 2c on a subject I'm in no way any kind of an expert.

cri-s, 2017-05-13:
you have not understand the principal point, and further you don't understand the "rotation convention",
What if you have to assembly layout made 2008 or 2005 that uses totally different zero point component
rotation convention. Reconfigure everything, change every data line manually ? read all datasheets ? No.
If you import that data to be set inside openpnp, you need to really do it and rotation is not anymore easy to setup.

dzach, 2017-05-13:
@cri-s It may be so. Here is a work flow for a job in OpenPnP, as i envisage it.

Ideally one should be able to select a suitable standard package for a part and get both the footprint/pads and the zero orientation data automatically into OpenPnP. The user then sees the resulting outline of the part and compares it with an actual image coming from a camera above the pick position of a feeder.  If the part proves to be at a different angle, then he sets this different angle in the proper place and proceeds.

In essence all I'm saying is that if standards exist, then let's use them. They usually come with data prepared by others painstakingly, for a reason.

vonnieda, 2017-05-13:
@dzach I think having information in the part definition that gives the user a bit more information about the orientation of a part when rendering it is a good idea, but I am not sure it can be used for loose part vision. The purpose of loose part vision is to find the orientation of the part using markings and the markings follow no standard at all that I know of.

It's all well and good to know that "pin 1 is in the upper left" but if you don't know what the orientation of the part is that information doesn't help you. You need to look at the markings for that information. 

This is why I think template match or a more sophisticated feature matcher is the only real solution. The markings follow no standard, and they are extremely varied. It's likely every part will need to be trained.

One benefit of using a strong feature detector like SIFT is that we could build up a shared database over time so users could avoid having to train common parts themselves. I don't think that would work as well with template matching.

cri-s, 2017-05-13:
@jason, do you had success with sift/surf/orb ?
I has to less keypoint in order brute match or flann worked.
OK I was not using 4mpixel camera.
Further from lightning I have seen this is really structured with intensity
pattern that on same pins different keypoint at present. Just 1mm offset
and key points are different.
Yes I could train pins but the really different lightning inhibit the
possibility to share pipelines or pattern coupled with different macro
factor and as result different parameters.
Matching the round hough circle inside the 2 or four quadrants works, same
as doing template match for the logo or marking and compare this with the
result from template.
Having position of pin1 inside pad data, yes its possible, but really,
where is using that pad and footprint. Further the footprint is not updated
on footprint reticle, no one wants that because unused, but it is a
eyecatcher, could be useful, difficult to maintain.

dzach, 2017-05-13:
@vonnieda , I think I've taken the opposite approach. I am _assuming_ the feeder/vision system has a way to find the marking and I'm trying to see how this correlates _automatically_ with the specification of the package. One cannot simply say, "ok, here is pin 1, turn that to the lower left and we are done!". This may be totally wrong. Instead, either before or after having found the marking, there should be a lookup somewhere that will give a result like: "Fine. Pin 1 should be in the middle of a side, looking at 12 o'clock for zero orientation. Turn the part so that it faces that direction and we are done". So, the way I understand this is that the standards will help to automatically retrieve such information for standardized packages, without regard to the feeder. The feeder, any feeder, even the loose feeder, helps answer the question "where is pin 1?", but the package  says "it should be facing north for zero orientation". At present we just substitute this automatic process with our manual intervention to set the feeder rotation to what "works" in practice. But I think that such a manual process is hard to turn to automatic.

vonnieda, 2017-05-14:
@dzach It's not clear to me what the benefit of storing and managing that information is, though. All that matters is that if a user expects a part to be oriented a certain way for zero angle, that we are able to determine that orientation. It doesn't really have anything to do with pin 1 or the standards since, as I said, the standards to do not define anything about *finding* pin 1. 

As an example, let's say that the standard says that on a TQFP32, pin 1 is the leftmost pin in the north row of pins when oriented at 0 degrees. Pin 1 looks like every other pin. If we want to use the loose part feeder to pick this part and orient it correctly, we have to look at the markings. The markings are not standardized (as far as I know). So since we need to look at the markings, and the markings tell us the orientation, what value does knowing and storing pin 1 have?

dzach, 2017-05-14:
@vonnieda I see that I need to express myself better. This is how I think about this:

If you store a photo of the part the at zero position, and then you match this photo to get the "zero orientation", then there is no value in knowing what the standard says, because by putting the part _at the correct position_ and declaring it "zero orientation" you become the standard yourself. Still, that _correct position_ has to also be known to the CAD package that lays out the PCB and exports the file with the positional data for each part. Or this has to be done by trial and error, the other way around.

This agreement is usually provided by the "standard", whatever that is, even a self made convention, if need arises. An automated method could take the data that the CAD program has supposedly stored in a database of some kind, about the footprint of a part and hopefully location of pin1, and convert it to e.g. a template for "zero orientation" at the P&P level. Since there exist thousands of parts in such databases, many of them open source and free, an automated process could import this data to OpenPnP and create the appropriate templates. OpenPnP pad dimensions and this "zero orientation" we are talking about come to mind. That's why I started trying to detect pin1 using geometrical characteristics, because all this data existing in open databases could potentially be used. I haven't done much in that direction yet.

In your description, this portion of the job is done by the person that places the part to the _correct position_ after having read the standard, or by trial and error. There is no automation involved: each part has to be manually located at _the correct position_ and a photo be taken and stored.

However, there are many component packages that have an distinct outline, or even  standard markings, e.g. a discernible dot. E.g. these SOT packages, coming from the sources @cri-s quoted above:

[image]

Due to their asymmetry, they seem easier to detect using some of already available stages. @cri-s mentioned that somewhere above, but these comments are now gone.

OpenPnP does not make much use of such standards, and maybe for a good reason: conformity and maintenance of standard data does come with a cost, at least in time spent to keep them updated and useful. OpenPnP users have to enter their own data for every footprint, body size and pads, and the "zero orientation" location, quadrant, sextant  or whatever, is not yet considered relevant. Sharing of such data is not common practice, as far as I can tell. 

By limiting use of standards, one limits the interoperability of systems, etc. I think the rest is known to most of us. That's why I first considered the possibility to use standards, as much as possible.

I hope the above make sense.

vonnieda, 2017-05-15:
I'm closing this ticket since it's not actually an issue that needs to be resolved, but is a discussion. The discussion can certainly continue, closing it won't affect that. In the future, I'd suggest using the mailing list for discussions like this as it's more likely to get input from other people, and also avoids the issue of comments being deleted.

closed 2017-05-15 by vonnieda

dzach, 2017-05-15:
The mailing list seems to work differently, being a lot more user oriented than developer oriented.
In any case, this may end up a `notes to myself` kind of an entry, which might be indeed better if I continued it on my own fork's space. I hope to have some hard code in the end, as a proof of concept or a more workable solution.

@cri-s 's comments have been of great value to me, as they have always provided new information, sometimes even code written on request, but being a newbie about Java and PnP, I have to decrypt them slowly. I greatly appreciate the help he is offering.

I gave MatchTemplate another try and got very good results this time. Will soon have a couple of simple stages that help produce template images en mass, by doing automatic rotation of detected models etc.

The problem I see with template matching, and I thing @cri-s also mentioned it, is that templates made by one user may be of no value to other users, or maybe even to their creator, after changes to the vision system are made, since they seem to be specific to the system that created them. To avoid that, it seems like some kind of standardization will again be necessary, e.g. images scaled to the dimensions of a standard template, color normalization or gray scale images, etc.

dzach, 2017-05-15:
Advice needed:

I have a new `Rotate` stage that takes a `RotatedRect` from anywhere in an image and delivers a new `RotatedRect` cropped to the boundaries of the initial `RotatedRect` and rotated by a new requested angle.

If one enters 0 for the requested angle (let's call this "rectification" of the image) and then follows this stage with an `ImageWrite` or `ImageWriteDebug`, the image can be saved as a template ready to be used together with a subsequent `MatchTemplate`. This takes care of all 5 steps of "part training" in @vonnieda 's _suggestion for loose part feeding, i.e. top recognition_.

By similarly detecting and then "rectifying" any single `RotatedRect` in an image and then rotating it 4 x 90deg while doing a `MatchTemplate` each time, one can have recognition of parts, as outlined in @vonnieda 's _suggestion for top recognition_.

Question:
1. What should this stage's name be? The name `RotateCropRotatedRect` sounds a little strange, but is accurate.
2. What should the next one, i.e. the complex one that will do a 4x90deg rotation and deliver the best match, be called? `MatchPart`? Such a generic name could be a wrapper for a number of different recognition methods which may be implemented at a later time, similar to the way `MaskModel` or my new `DrawModel` and `ColorToModel` (described elsewhere) work on a variety of different models. Or, if a simpler stage doing just one type of recognition is preferred, it could be named `MatchPartTemplate` hinting that the `MatchTemplate` stage is being used.

Advice needed!

vonnieda, 2017-05-16:
Hi @dzach, I'm having trouble visualizing this. Can you show a screenshot or video or something showing how it works? Or just commit to a branch and give me the link.

dzach, 2017-05-16:
@vonnieda Sure. Here is a Beanshel script that does what the stage does. I have renamed it `OrientCropRotatedRect` since its function is to _orient_ the rect _to_ the given angle and not _rotate_ it _by_ the given angle:

```
import org.opencv.core.Core;
import org.opencv.core.Mat;
import org.opencv.core.Rect;
import org.opencv.core.RotatedRect;
import org.opencv.imgproc.Imgproc;

import org.openpnp.vision.pipeline.CvStage.Result;

Mat mat = pipeline.getWorkingImage();

modelStageName = "..";
Result mStage = pipeline.getResult(modelStageName);
RotatedRect rect = null;
Mat image = new Mat();
double degrees = Double.parseDouble(args);
Result result;

if (mStage.model.get(0) instanceof RotatedRect) {

  rect = mStage.model.get(0).clone();
  Rect bbox = rect.boundingRect();
  Mat dst = new Mat();

  Mat mapMatrix = Imgproc.getRotationMatrix2D(rect.center, rect.angle - degrees, 1.0);
  Imgproc.warpAffine(mat, dst, mapMatrix, mat.size(), Imgproc.INTER_LINEAR);
  mat.release();
  // adjust rect to new angle
  rect.angle = degrees;
  // find bbox for new rect
  bbox = rect.boundingRect();
  // adjust rect to the new center
  rect.center.x = bbox.width/2.0;
  rect.center.y = bbox.height/2.0;
  // crop the image to the new bbox
  image = dst.submat((Rect)bbox);
}
new Result(image, rect);
```

It expects an image and a RotatedRect in `modelStageName`; The default name `..` should be changed accordingly. The orientation angle is set in `ScriptRun`s parameter `args`.

vonnieda, 2017-05-16:
Thanks @dzach. IMHO this should be broken up into smaller stages. As an example, I think you could accomplish the same thing with a Rotate stage that can rotate around a RotatedRect's center point and a Crop stage that crops to a RotatedRect. These two new, small stages would be generally useful for building more complex stages. I do realize we may have a difference of opinion there.

Anyway, for a name, I'd think somewhere along the line of what is actually happening. You are capturing a region of an image defined be a rotated rect and then potentially rotating that region. So maybe ExtractRegion, or something like that?

dzach, 2017-05-16:
Sure, I'll break it to simple stages, they may be useful to other situations, especially `crop`. I don't know about `rotate` though. Putting a rect to a given orientation is not the same as just rotating it. It is rotating it that much that will bring it to the desired orientation.

The script above needs some more consideration on this very subject, as I did all my tests with a limited number of rects. I'll update it soon.

dzach, 2017-05-16:
@vonnieda  Also, rotating/orienting a part _in place_ is kind of strange. I can't imagine why someone would like to do that.

cri-s, 2017-05-16:
@dzach,
you should not do   mat.release(); on getWorkingImage as the image
from previous stage
could be used on annother stage.
Normally you rotate to   -rect.angle   in order to adjust for rotation.

dzach, 2017-05-16:
Thanks @cri-s I copied that part blindly from `Rotate` stage, didn't thing about the complications.

dzach, 2017-05-17:
After more testing using the test images from #562, I now have two beanshell scripts, the first `CropRotatedRect` does what it says and the second `RectifyRotatedRect` orients a `RotatedRect` with its long side horizontal. Here are some tests:

[image]
[image]
[image]

[image]
[image]
[image]

[image]
[image]
[image]

[image]
[image]
[image]

[image]
[image]
[image]

[image]
[image]
[image]

The pipeline I used and the scripts are attached.
[pipeline.zip](https://github.com/openpnp/openpnp/files/1006398/pipeline.zip)

cri-s, 2017-05-17:
@dzach, as it seems you check only the ic body, where image inverting ,
thresholding and size filtering resulting Rotated rectangle is enought. Do
you have already tested resistors in combination with other components?

.

>

dzach, 2017-05-17:
Corrected the name of one the scripts in the zip file.
Progress so far demonstrates how to produce templates for parts on a loose part feeder. The scripts are just for testing, there is almost no error control.

Here are the results from a quick run though the bin of used parts that I use for testing:

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

The above original and template images were stored by using `ImageWriteDebug` after the relevant stages. No manual intervention whatsoever. The originals were resized to 596x480px for uploading here. It took me a hell of a time to sort out the originals with their corresponding templates. 

`ImageWriteDebug` ought to use an incremental counter together with the random number so that all images coming out of the same pipeline run would have the same count, e.g. orig_23_923478989.png templ_23_423424323.png.

dzach, 2017-05-17:
Hey @cri-s , no I rectify whatever the preceding masks render as a model. Out of hundred of images I only got one case where a SOT23 came out rectified in a diagonal position. The rest all came out horizontal.

Here are some results from a colored body mask:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

cri-s, 2017-05-17:
> ![tmpl_

I post the script that generate the reference image, probably today.
Need to remove some code first and remove the call to own pipeline processor.

dzach, 2017-05-17:
> @vonnieda 
> [image]

> With the offset lighting the markings are clearly visible. So, I think I need to improve my lighting before I can make useful images.

> What do you think?

This one is not detected by the other pipelines and, although it has far better contrast for the letters, it also has huge shadows that hide its outline. It obviously needs illumination from the other sides too, so that the effect of the shadows can be minimized. Markings are clearer but the dot is not. It is a very nice idea though and easily implementable too, since the a loose feeder has plenty of space and a convenient shape to install small LEDs on its sides.

[image]

vonnieda, 2017-05-17:
I think the same effect can be achieved without the shadows just by using a
ring light with LEDs mounted at 45 degrees. I've also seen box lights with
LEDs angled that seem to work.

cri-s, 2017-05-18:
Could you test the stage:

Because of gui bug, it uses temporary files.
the pipeline is setup using debug, remove dbg=1 and it's on working state.
if you set reset to 1, it delete the associated file, if you remove
just the args or id,
it recreate a new file not deleting the old.
I'm unsure if it's wise to add file.deleteAtExit() or not, what you
think about that?

Unfortunatly  i was not able to recreate the "wrong rectangle" on
sot23 in order to make
special case for it.

cri-s, 2017-05-18:
For any question or feedback, just write.

2017-05-18 21:21 GMT+02:00, Cri S <phone.cri@gmail.com>:
> Could you test the stage:
>
> Because of gui bug, it uses temporary files.
> the pipeline is setup using debug, remove dbg=1 and it's on working state.
> if you set reset to 1, it delete the associated [...]

dzach, 2017-05-18:
Question:
Where can I find the pipeline?

cri-s, 2017-05-18:
sorry, i have send the wrong files, probably bsh too.
It is not setup to detect parts, only for reference.
It expect the part at center of camera, otherwise exception is generated.

dzach, 2017-05-18:
@cri-s I havn't seen the files yet.

[49 more replies]