openpnp/openpnp #548: SimpleBlobDetector : java.lang.Error: Configuration instance not yet initialized

dzach, 2017-05-09:
# Problem Report
## Describe the Issue
As soon as a `SimpleBlobDetector` is selected, the following error occures:
```
Exception in thread "AWT-EventQueue-0" java.lang.Error: Configuration instance not yet initialized.
        at org.openpnp.model.Configuration.get(Configuration.java:82)
        at org.openpnp.vision.pipeline.stages.SimpleBlobDetector.process(SimpleBlobDetector.java:244)
        at org.openpnp.vision.pipeline.CvPipeline.process(CvPipeline.java:192)
        at org.openpnp.vision.pipeline.ui.CvPipelineEditor.process(CvPipelineEditor.java:152)
        at org.openpnp.vision.pipeline.ui.PipelinePanel$2.tableChanged(PipelinePanel.java:146)
        at javax.swing.table.AbstractTableModel.fireTableChanged(AbstractTableModel.java:296)
        at javax.swing.table.AbstractTableModel.fireTableDataChanged(AbstractTableModel.java:198)
        at org.openpnp.vision.pipeline.ui.StagesTableModel.refresh(StagesTableModel.java:25)
        at org.openpnp.vision.pipeline.ui.PipelinePanel$5.actionPerformed(PipelinePanel.java:225)
        at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
        at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
        at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
        at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
        at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
        at java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:289)
        at java.awt.Component.processMouseEvent(Component.java:6533)
        at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
        at java.awt.Component.processEvent(Component.java:6298)
        at java.awt.Container.processEvent(Container.java:2236)
        at java.awt.Component.dispatchEventImpl(Component.java:4889)
        at java.awt.Container.dispatchEventImpl(Container.java:2294)
        at java.awt.Component.dispatchEvent(Component.java:4711)
        at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888)
        at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4525)
        at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466)
        at java.awt.Container.dispatchEventImpl(Container.java:2280)
        at java.awt.Window.dispatchEventImpl(Window.java:2746)
        at java.awt.Component.dispatchEvent(Component.java:4711)
        at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
        at java.awt.EventQueue.access$500(EventQueue.java:97)
        at java.awt.EventQueue$3.run(EventQueue.java:709)
        at java.awt.EventQueue$3.run(EventQueue.java:703)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
        at java.awt.EventQueue$4.run(EventQueue.java:731)
        at java.awt.EventQueue$4.run(EventQueue.java:729)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
        at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
        at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
        at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```
## Steps to Reproduce
1. Start the standalone CvPipelineEditor
2. Before selecting an image, select and accept a `SimpleBlobDetector` stage. 
3. The stage **does not load** and throws the above error.

## Expected Result
The stage should load.

## Actual Result
The stage does not load. If an image is selected prior to selecting the stage, then the stage loads.

## Environment
Running on a Linux 64bit machine, Kubuntu 16.04

cri-s, 2017-05-09:
This is inside Standalone processor, several stages don't work, as example the stages that access the Part dimension from nozzle and use that size to mask out things (background on uplooking, parts on downlooking) because Standalone processor don't load the configuration and in case processor need to 
have access to configuratin, as example only to store and load template images from parts, it don't work without configuration. I have several stages that don't work on standalone processor, eventually @vonnieda should
clearify if the standalone visual tool should load configuration, if this stages should adapted to remove
configuration dependency, here is possible, and for the stages where this is not possible.

dzach, 2017-05-09:
No, unfortunately the same happens inside OpenPnP too:

[image]

See how the pipeline table is messed up. The selected stage in the screenshot is SimpleBlobDetector, yet it doesn't display the list of its properties. Nothing works after I load this stage. After it's removed, everything goes back to normal

vonnieda, 2017-05-09:
I'm looking at this. There are several problems with the stage that make it unsuitable for using in the pipeline. I'm fixing them now and will have it pushed very soon.

vonnieda, 2017-05-09:
Fixed in https://github.com/openpnp/openpnp/commit/4b1d7520c59564aa92f580df2e905840705d3ca2. Here is an example pipeline that works:

```
<cv-pipeline>
   <stages>
      <cv-stage class="org.openpnp.vision.pipeline.stages.ImageRead" name="0" enabled="true" file="/Users/jason/Projects/openpnp/openpnp-test-images/Parts/Loose/1/Fiducial Cam_2016-06-26_11.08.33.468.png"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.SimpleBlobDetector" name="1" enabled="true" threshold-step="10.0" threshold-min="50.0" threshold-max="220.0" repeatability="2" dist-between-blobs="10.0" color="true" color-value="0.0" area="false" area-min="25.0" area-max="5000.0" circularity="false" circularity-min="0.800000011920929" circularity-max="-1.0" inertia="true" inertia-ratio-min="0.10000000149011612" inertia-ratio-max="-1.0" convexity="true" convexity-min="0.949999988079071" convexity-max="-1.0"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.DrawKeyPoints" name="2" enabled="true" key-points-stage-name="1"/>
   </stages>
</cv-pipeline>
```

closed 2017-05-09 by vonnieda

dzach, 2017-05-09:
```
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007f9329dc7f4a, pid=6333, tid=0x00007f93641b6700
#
# JRE version: OpenJDK Runtime Environment (8.0_121-b13) (build 1.8.0_121-8u121-b13-0ubuntu1.16.04.2-b13)
# Java VM: OpenJDK 64-Bit Server VM (25.121-b13 mixed mode linux-amd64 compressed oops)
# Problematic frame:
# C  [libopencv_java2411.so+0x6a8f4a]  cv::SimpleBlobDetector::findBlobs(cv::Mat const&, cv::Mat const&, std::vector<cv::SimpleBlobDetector::Center, std::allocator<cv::SimpleBlobDetector::Center> >&) const+0x46a
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /home/dz/develop/openpnp/hs_err_pid6333.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#
```

vonnieda, 2017-05-09:
What was the pipeline?

dzach, 2017-05-09:
The one you provided above.

dzach, 2017-05-09:
Java eventually exits, after 1 min,

vonnieda, 2017-05-09:
Can you experiment with it some and see if you can find a cause? Maybe try adding SimpleBlobDetector into a pipeline you know works?

dzach, 2017-05-09:
It doesn't seem to work. 

- The standalone editor shows the stage's parameters after I load an image, not before, but OpenPnP does not, in any case.
- In the standalone editor I still get the error `Exception in thread "AWT-EventQueue-0" java.lang.Error: Configuration instance not yet initialized.`
- When the parameters do show correctly in the pane, even then I cannot get it to work, no matter what the values are. I have entered the values off the pipeline above, which are the same as the defaults, but it doesn't work.
- The current image is not displayed, after this stage.
- No image can be displayed after this stage. I tried an ImageRead, but nothing is shown.

vonnieda, 2017-05-09:
If you are still getting the configuration error you did not update your
code properly. The configuration call has been removal completely. Check
your got status and if you are building from code do a clean rebuild.

cri-s, 2017-05-09:
It's normal that this pipeline is not working correctly.
On drawKeypoints the colors must be defined,
otherwise this illegal code is called.
            Features2d.drawKeypoints(mat, matOfKeyPoints, mat, FluentCv.colorToScalar(color), 0);
This is (probably) valid code:
            Features2d.drawKeypoints(mat, matOfKeyPoints, mat, FluentCv.colorToScalar(color), 1);

dzach, 2017-05-09:
```
git pull upstream develop
From https://github.com/openpnp/openpnp
 * branch            develop    -> FETCH_HEAD
Already up-to-date.
```

dzach, 2017-05-09:
What do you mean by `a clean build`?

Clone another openpnp to a separate directory?

vonnieda, 2017-05-09:
@dzach Are you using Eclipse? Right click the project and select Refresh. You can also look at the file to see if the Configuration reference has been removed. If you are still seeing it then something else is wrong. Look at this commit and compare it to your file. See if the Configuration call is removed: https://github.com/openpnp/openpnp/commit/4b1d7520c59564aa92f580df2e905840705d3ca2

dzach, 2017-05-09:
Let me try again a : 
```
mvn package
```

vonnieda, 2017-05-09:
If you are using Eclipse, always do the following after a pull:
* Right click Project, select Refresh.
* Go to Project Menu, select Clean.

If you are building from the command line:
* mvn clean pakage

dzach, 2017-05-09:
No, I'm not using Eclipse, just plain command line.
The Configuration reference is removed, but
I didn't do a `mvn clean package`.

Doing it now

dzach, 2017-05-09:
Nope. Now the Standalone editor crashes immediately with this (same?) error.
```
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007fecf5a7ef4a, pid=7046, tid=0x00007fecf4671700
#
# JRE version: OpenJDK Runtime Environment (8.0_121-b13) (build 1.8.0_121-8u121-b13-0ubuntu1.16.04.2-b13)
# Java VM: OpenJDK 64-Bit Server VM (25.121-b13 mixed mode linux-amd64 compressed oops)
# Problematic frame:
# C  [libopencv_java2411.so+0x6a8f4a]  cv::SimpleBlobDetector::findBlobs(cv::Mat const&, cv::Mat const&, std::vector<cv::SimpleBlobDetector::Center, std::allocator<cv::SimpleBlobDetector::Center> >&) const+0x46a
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /home/dz/develop/openpnp/hs_err_pid7046.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#
```

vonnieda, 2017-05-09:
Okay. Things I would check:
* Try it on a non-Linux OS if you have one available. 
* Try using the Oracle JDK instead of OpenJDK. 
* Try commenting out the blobDetector.read() call in SimpleBlobDetector.java to see if it's a parameter issue.
* Take a look at the log file it produced (  /home/dz/develop/openpnp/hs_err_pid7046.log ) and see if there are any clues there.

The call that is failing ( cv::SimpleBlobDetector::findBlobs ) is called internally be OpenCV during the blobDetector.detect() call, which is line 268 of SimpleBlobDetector.java. So you can ignore anything after that.

dzach, 2017-05-09:
I meant, "Now the Standalone editor crashes immediately when I paste the pipeline from above, with this (same?) error."

vonnieda, 2017-05-09:
Yea, I understand. It seems like the stage is crashing on your machine. The crash is happening inside SimpleBlobDetector. I'm not sure why, as it works fine on my machine. The things I suggested above are ideas on how to debug the stage. I can't do it, since it's working correctly here.

dzach, 2017-05-09:
Hm, I have completely abandoned Windows since more than 10 years ago. I still have a Toshiba TECRA, circa '98, with Windows 95 on it, but I don't have an PCMCIA network adapter to work with it, since the one I have has a cable that died about a decade ago. I had a WinXP VM somewhere, but it needs frequent updates, as its "free" license expires every month or so.

Just (half) kidding.

The thing I could try though is run it with Oracle Java, but I have to install it first.
In any case, as it stands now this stage is unusable for me, and maybe for other people too. Is it used in any of the standard pipelines in OpenPnP? Because if it is, then it will break OpenPnP all together.

cri-s, 2017-05-09:
change the return in the SimpleBlobDetector to this:
        return new Result(null, keypoints);

It works then on standalone editor.

vonnieda, 2017-05-09:
@cri-s Are you seeing the same error in Linux? FWIW, I'm testing in StandaloneEditor and it's working fine on Mac.

vonnieda, 2017-05-09:
@dzach No, the stage is not used in OpenPnP currently and I don't know of anyone using it. The stage, as it was before, did not do much and was basically useless, so there was not a great reason to use it. I'm sure we can solve this issue, but I need to rely on you guys to help, since I cannot reproduce it.

cri-s, 2017-05-09:
I'm running it on 32bit system and not getting the error/coredump but no image.
Changing that code it works. but just tested, not always.
 For me the error is still the drawKeypoints as mat is always
the same and drawKeypoints kill that mat allocating a new mat on the
first loop, then even ImageRead don't display the mat again,
displaying only the red crosses.

If i add color to drawKeypoint and change image, it works on all images.

<cv-pipeline>
   <stages>
      <cv-stage class="org.openpnp.vision.pipeline.stages.ImageRead"
name="0" enabled="true"
file="\Users\jason\Projects\openpnp\openpnp-test-images\Parts\Loose\1\Fiducial
Cam_2016-06-26_11.08.33.468.png"/>
      <cv-stage
class="org.openpnp.vision.pipeline.stages.SimpleBlobDetector" name="1"
enabled="true" threshold-step="10.0" threshold-min="50.0"
threshold-max="220.0" repeatability="2" dist-between-blobs="10.0"
color="true" color-value="0.0" area="false" area-min="25.0"
area-max="5000.0" circularity="false"
circularity-min="0.800000011920929" circularity-max="-1.0"
inertia="true" inertia-ratio-min="0.10000000149011612"
inertia-ratio-max="-1.0" convexity="true"
convexity-min="0.949999988079071" convexity-max="-1.0"/>
      <cv-stage
class="org.openpnp.vision.pipeline.stages.DrawKeyPoints" name="2"
enabled="true" key-points-stage-name="1">
         <color r="255" g="51" b="51" a="255"/>
      </cv-stage>
   </stages>
</cv-pipeline>

vonnieda, 2017-05-09:
@cri-s Okay, that is a different problem then. @dzach's error is a core dump in the blob detector, not in drawing the keypoints. drawKeypoints without a color is supposed to use random colors, which is what it does for me.

vonnieda, 2017-05-09:
@cri-s I will change it to return null. The effect should be the same, but if it helps, that's fine.

cri-s, 2017-05-09:
@dzach, have you tested the pipiline i have posted that include the color ?

vonnieda, 2017-05-09:
Okay, changed to return null for image to see if that helps: https://github.com/openpnp/openpnp/commit/234808b5fa69e3584296da5fa7b7c8874c9982ca

dzach, 2017-05-09:
Retested with standalone editor, same build as before, just in case, still nothing.
Tested the same previous build with OpenPnP and it works! No error in the console whatsoever, not sluggish behaviour, I can change parameters, I see blobs detected.

Now, just retested with the standalone editor, and the damn thing works, without an error! That's strange. I have not updated the code, nor have I rebuilt the code, but it now works, after running OpenPnP. Something seems to have happened after I ran OpenPnP.

Do you want me to look for anything, before I pull the new fix?

vonnieda, 2017-05-09:
@dzach Nothing I can think of. If it happens again, we can investigate further.

dzach, 2017-05-09:
Well, @cri-s 's pipeline from above:
```
<cv-pipeline> 
  <stages> 
  <cv-stage class="org.openpnp.vision.pipeline.stages.ImageRead" name="0" enabled="true" file="\Users\jason\Projects\openpnp\openpnp-test-images\Parts\Loose\1\Fiducial Cam_2016-06-26_11.08.33.468.png"/> 
  <cv-stage class="org.openpnp.vision.pipeline.stages.SimpleBlobDetector" name="1" enabled="true" threshold-step="10.0" threshold-min="50.0" threshold-max="220.0" repeatability="2" dist-between-blobs="10.0" color="true" color-value="0.0" area="false" area-min="25.0" area-max="5000.0" circularity="false" circularity-min="0.800000011920929" circularity-max="-1.0" inertia="true" inertia-ratio-min="0.10000000149011612" inertia-ratio-max="-1.0" convexity="true" convexity-min="0.949999988079071" convexity-max="-1.0"/> 
  <cv-stage class="org.openpnp.vision.pipeline.stages.DrawKeyPoints" name="2" enabled="true" key-points-stage-name="1"> <color r="255" g="51" b="51" a="255"/> 
</cv-stage> 
</stages> 
</cv-pipeline> 
```

still crashes both OpenPnP and standalone editor, as soon as I load an image.

Log attached.
[hs_err_pid8084.log.zip](https://github.com/openpnp/openpnp/files/987086/hs_err_pid8084.log.zip)

vonnieda, 2017-05-09:
Thanks for the log. Still the same problem. It's crashing inside a method
in OpenCV, which makes it difficult to debug.

Jason

commit 4b1d7520 (2017-05-09): Fixes #548:
* Configuration is no longer needed, temp file is created in system temp directory.
* Stage now returns List<KeyPoint> as expected by other keypoint stages.
* Stage now frees the temporary keypoints Mat.
* Stage returns a valid mat in image so that pipeline can display it.