openpnp/openpnp #559: Casting error in SimpleBlobDetector

dzach, 2017-05-14:
I'm trying to cast the model out of a `SimpleBlobDetector` as a `java.util.ArrayList`, but it proves to be a `java.util.Arrays` construct and not a `java.util.ArrayList`, like most other model collections returned in a `result.model`, to my knowledge. So I'm getting this error:

```
java.lang.ClassCastException: java.util.Arrays$ArrayList cannot be cast to java.util.ArrayList
```

It seems that the issue is in `FeatureDetector`. 

To turn the return construct to a `java.util.ArrayList` and avoid the error, I changed:

```
        List<KeyPoint> keypoints = new ArrayList<KeyPoint>(kpMat.toList());
```
to:
```
        List<KeyPoint> keypoints = kpMat.toList();
```
taking advice from http://stackoverflow.com/questions/28851652/java-lang-classcastexception-java-util-arraysarraylist-cannot-be-cast-to-java.

Patch attached.
[SimpleBlobDetector.diff.zip](https://github.com/openpnp/openpnp/files/999359/SimpleBlobDetector.diff.zip)

vonnieda, 2017-05-15:
You should not try to cast to a concrete implementation, since there is no guarantee that it will be an ArrayList. You should cast to List<?> or List<KeyPoint> which is the generic interface. The reason for this is that a person implementing a stage may choose to store the KeyPoints in any type of class that implements List, for reasons of performance, ease of use or anything else. In general, unless an API specifically says it will return a concrete class (like ArrayList) you should always use the generic interface (like List).

closed 2017-05-15 by vonnieda

dzach, 2017-05-15:
Ok. That makes then SimpleBlobDetector a special case that has to be dealt with separately from the others.

vonnieda, 2017-05-15:
@dzach No, it shouldn't be. You should *never* try to cast to ArrayList unless an API specifically says it returns that. In other words:

```
public ArrayList<KeyPoint> someFunction() { } // okay to cast return to ArrayList
public List<KeyPoint> someFunction() // not okay to cast to ArrayList, no matter if it sometimes works or not.
```

If there is anywhere you are casting model results to ArrayList, that is a bug, and should be changed to List. If the reasoning for this isn't clear, let me know and I'll be happy to explain further.

vonnieda, 2017-05-15:
Okay, took a look through the code and I see where you've used this type of cast. These will need to be changed, as they will break if the incoming List is not an ArrayList.

Problems are in:
* DilateModel:87
* MaskModel:103

The change is easy. Just need to do List l = (List) result.model; instead of (ArrayList) result.model. The API is the same, so no change is needed in the rest of the code. It's just that you have to use the interface rather than the concrete class.

dzach, 2017-05-15:
Thank you @vonnieda , I changed my code by casting to List<?> and now it works. This mistake made me think it was a special case. It's clear now.

Also, I mistakenly copied the code I changed in my first post the other way around. It is correct in the diff file though.