openpnp/openpnp #560: User friendliness of pipeline stages

dzach, 2017-05-14:
# Problem Report
## Describe the Issue
Some pipeline stages require more user involvement than necessay, e.g. by requiring user to enter values that are not directly related to his actual goal, while these could be set automatically in code.

## Example 1
The `MinAreaRect` stage: its goal is to deliver a model for a minimum area rectangle that includes all pixels in an image that pass a set threshold range. The deliverable is a RotatedRect model. The stage however requires the user to convert the image from color to gray, if the image is not already in gray. The question is why the user should bother with the color conversion, which he may not need otherwise. 

## Example 2
The "Draw..." stages. They all take a model as input and draw a shape according to the apearence settings, but instead of having just one to do the drawing, there is one for each type of model. This forces the user to think about the model in the pipeline each and every time and decide about the appropriate stage to use to just simply draw the model. The user e.g. has to know that the SimpleBlobDetector outputs a KeyPoints list which must be drawn with the `DrawKeyPoints` stage. Why? All necessary data for an automatic decision are already available internally to the pipeline. The user could just set the parameters for the desired appearance of the model, or accept the defaults.
 
# Another way of looking to stages
Writing code for stages while having in mind a normal, non expert user, can make pipelines easier to work with. This may imply a move away from the established KISS (Keep It Simple Stupid) doctrine. Easier to use stages may become more complex and more capable, they may look less like the assembly language of OpenCV and more like an interpreted higher level language, with lots of options; they may become more intuitive, more user friendly and more flexible.

The latter is the intention of this proposal.

# Practical outcome.
I have a small number of not-so-simple stages that demonstrate the above idea. Namely:

1. `DrawModel`. Takes as input a model, by default a result from the previous stage or from a named one, finds the model in it, if any, and draws it according to the line/fill parameters entered by the user. It covers all existing models in one stage, i.e. Contours, Circle, Rect, RotatedRect, <List>Circle, <List>RotatedRect, KeyPoints and TemplateMatches.  
2. `MaskModel`. Already committed to OpenPnP. It takes a model for input, by default  from the previous stage or a named one, and filters or masks the working image with its shape(s). It should probably include DrawPolygon, with  a separate parameter `shapes`.
3. `ColorsToModel`. This stage performs an inRange filtering/masking on the working image to filter/mask out unnecessary image content and then tries to detect the model requested by the user. It outputs a result with both an image and a model. For the output image, user options include the unchanged original image, or a black and white threshold, a gray, or a color filtered/masked image.  For the output model user options include either none, or one of Contours, BoundingRect, Circle, MinAreaRect, RotatedRects, Circles.
4. Default use of `..` in a requiredStageName takes as imput the results of the previous stage. This patch has already been submitted for inclusion in OpenPnP.

These stages aim to reduce the amount of work a user has to do himself. The idea behind them is that repetitive sequences of simple stages that comprise the majority of common pipelines can be substituted by fewer stages doing more work. With fewer and more goal oriented stages the pipeline editor becomes uncluttered and is easier to work with. Advance users will still be able to use the simple stages to perform the same variety of tasks as before.

vonnieda, 2017-05-15:
In general, I am open to the idea of smarter stages, but there is a fine line. For example, you mention the user needing to convert to gray. This can definitely be done automatically in many cases, but it would need to be done in such a way that the user can still provide their own grayscale image. So something like:
```
if (!imageIsGreyscale(mat)) {
    convertToGrayscale(mat);
}
```

And the imageIsGreyscale and convertToGrayscale methods are more complex than it first seems. The input image may be in HSV, HSV_Full, BGR, RGB, etc. So the format will need to be detected and accounted for.

Your example of drawing the model, I see no reason that we can't do it as you described. It would certainly make it easier for users.

So, yes, I am definitely open to smarter stages, but I don't want to sacrifice flexibility. My driving goal with this library has been to make it easy to piece together simple pieces to form complex behavior and I don't want to give that up for a little bit of convenience. If we can do both, that's perfect.

It's also important to consider "hidden functionality" and the loss of feedback to the user. By having a pipeline explicitly define each step, it's easy to step through the pipeline and see how the image is transformed each step of the way. This is one of the most powerful and useful features of the pipeline editor and part of the reason I wrote it. I was having trouble seeing everything that was happening in more complex algorithms and wanted to be able to see the steps. By doing multiple transformations inside the stage we lose that important feedback.

It may be worth looking through some of the documentation for RoboRealm ( http://www.roborealm.com/help/index.php ) as that library was a large part of the inspiration for this one.

closed 2017-05-15 by vonnieda

dzach, 2017-05-15:
@vonnieda Thanks for the reply and the link! That's a great resource, and its completeness level looks amazing.

The openpnp CvPipeline is indeed very didactic, like an assembly language for OpenCV. For this reason and as it stands today, I believe it carries along with it all the difficulties of an assembly language to an end user: its is extremely analytic, to a degree of becoming tedious when a non-developer has to repeat the same steps over and over. Assembly language has macros to help alleviate that problem. 

Complex stages cannot do without simpler ones. Even when simple stages are not performing the main task they are the glue for tying together larger blocks. So the idea is not to eliminate simple stages but to provide more complex stages that can perform specific tasks in an easier way.

What seems to be necessary to get to the next level, is a change in the editor, so that stage Categories will become functional, and even better, will be user editable, or allow for custom made ones. Then custom grouping of stages will be possible and can make working with the pipeline a lot easier, like the RoboRealm interface seems to be.

I know, ideas are cheap. I hope the code I provide is more help than nuisance and takes some of the work that's needed to make ideas happen off other people's back.