openpnp/openpnp #561: pipeline editor not update description

cri-s, 2017-05-15:
# Problem Report
## Describe the Issue
on pipeline editor description is not updated but remain cached.
Example of code used:

	public String getDescription() { return getDescription(null); }

    public String getDescription(String id) {
		if(mode==Mode._____)  		{
			if(id==null) 				return "Select Mode";
		}
		if(mode==Mode.Gammal) 	{
			if(id==null) 	return "Gamma adjust";
			if(id.equals("param1")) 	return "Gamma value";

## Steps to Reproduce
Jason surely know how to fix it without needing to reproduce iut.

## Expected Result
When the mode is changed the description should get updated.
Even enabling/disabling the description using the description help button don't change anything.

## Actual Result
result is cached

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
[image]
[image]

vonnieda, 2017-05-16:
Fixed.

closed 2017-05-16 by vonnieda

cri-s, 2017-05-16:
Thanks.
I have annother strange UI problem,
When i run this stage, the ui blocks some seconds.
I test it now cloning the mat, if it works, but maybe it's totally different reason and you know 
immediatly why it's blocking. It is the orginal stage, just the enum is changed/increased.

package org.openpnp.vision.pipeline.stages;

import org.opencv.core.Mat;
import org.opencv.imgproc.Imgproc;
import org.openpnp.vision.FluentCv;
import org.openpnp.vision.pipeline.CvPipeline;
import org.openpnp.vision.pipeline.CvStage;
import org.simpleframework.xml.Attribute;

/**
 * Converts the color of the current working image to the specified conversion.
 */
public class ConvertColor extends CvStage {

    public enum ColorCode {
BGR2BGR555(Imgproc.COLOR_BGR2BGR555),
BGR2BGR565(Imgproc.COLOR_BGR2BGR565),
BGR2BGRA(Imgproc.COLOR_BGR2BGRA),
BGR2GRAY(Imgproc.COLOR_BGR2GRAY),
BGR2HLS(Imgproc.COLOR_BGR2HLS),
BGR2HLS_FULL(Imgproc.COLOR_BGR2HLS_FULL),
BGR2HSV(Imgproc.COLOR_BGR2HSV),
BGR2HSV_FULL(Imgproc.COLOR_BGR2HSV_FULL),
BGR2Lab(Imgproc.COLOR_BGR2Lab),
BGR2Luv(Imgproc.COLOR_BGR2Luv),
BGR2RGB(Imgproc.COLOR_BGR2RGB),
BGR2RGBA(Imgproc.COLOR_BGR2RGBA),
BGR2XYZ(Imgproc.COLOR_BGR2XYZ),
BGR2YCrCb(Imgproc.COLOR_BGR2YCrCb),
BGR2YUV(Imgproc.COLOR_BGR2YUV),
BGR2YUV_I420(Imgproc.COLOR_BGR2YUV_I420),
BGR2YUV_IYUV(Imgproc.COLOR_BGR2YUV_IYUV),
BGR2YUV_YV12(Imgproc.COLOR_BGR2YUV_YV12),
BGR5552BGR(Imgproc.COLOR_BGR5552BGR),
BGR5552BGRA(Imgproc.COLOR_BGR5552BGRA),
BGR5552GRAY(Imgproc.COLOR_BGR5552GRAY),
BGR5552RGB(Imgproc.COLOR_BGR5552RGB),
BGR5552RGBA(Imgproc.COLOR_BGR5552RGBA),
BGR5652BGR(Imgproc.COLOR_BGR5652BGR),
BGR5652BGRA(Imgproc.COLOR_BGR5652BGRA),
BGR5652GRAY(Imgproc.COLOR_BGR5652GRAY),
BGR5652RGB(Imgproc.COLOR_BGR5652RGB),
BGR5652RGBA(Imgproc.COLOR_BGR5652RGBA),
BGRA2BGR(Imgproc.COLOR_BGRA2BGR),
BGRA2BGR555(Imgproc.COLOR_BGRA2BGR555),
BGRA2BGR565(Imgproc.COLOR_BGRA2BGR565),
BGRA2GRAY(Imgproc.COLOR_BGRA2GRAY),
BGRA2RGB(Imgproc.COLOR_BGRA2RGB),
BGRA2RGBA(Imgproc.COLOR_BGRA2RGBA),
BGRA2YUV_I420(Imgproc.COLOR_BGRA2YUV_I420),
BGRA2YUV_IYUV(Imgproc.COLOR_BGRA2YUV_IYUV),
BGRA2YUV_YV12(Imgproc.COLOR_BGRA2YUV_YV12),
BayerBG2BGR(Imgproc.COLOR_BayerBG2BGR),
BayerBG2BGR_VNG(Imgproc.COLOR_BayerBG2BGR_VNG),
BayerBG2GRAY(Imgproc.COLOR_BayerBG2GRAY),
BayerBG2RGB(Imgproc.COLOR_BayerBG2RGB),
BayerBG2RGB_VNG(Imgproc.COLOR_BayerBG2RGB_VNG),
BayerGB2BGR(Imgproc.COLOR_BayerGB2BGR),
BayerGB2BGR_VNG(Imgproc.COLOR_BayerGB2BGR_VNG),
BayerGB2GRAY(Imgproc.COLOR_BayerGB2GRAY),
BayerGB2RGB(Imgproc.COLOR_BayerGB2RGB),
BayerGB2RGB_VNG(Imgproc.COLOR_BayerGB2RGB_VNG),
BayerGR2BGR(Imgproc.COLOR_BayerGR2BGR),
BayerGR2BGR_VNG(Imgproc.COLOR_BayerGR2BGR_VNG),
[... 146 lines cut ...]
        }

        public int getCode() {
            return code;
        }
    }

    @Attribute
    private ColorCode conversion = ColorCode.BGR2GRAY;

    public ColorCode getConversion() {
        return conversion;
    }

    public void setConversion(ColorCode conversion) {
        this.conversion = conversion;
    }

    @Override
    public Result process(CvPipeline pipeline) throws Exception {
        Mat mat = pipeline.getWorkingImage();
        Imgproc.cvtColor(mat, mat, conversion.getCode());
        return null;
    }
}

vonnieda, 2017-05-16:
Sorry, I don't know why that would happen. Try decreasing the enum to half
and see if it halves the time. If so, it could be that the property editor
component has poor support for large enums.

commit ef03036c (2017-05-16): Fixes #561: Description now refreshes when a property changes.