openpnp/openpnp #562: loose feeder

cri-s, 2017-05-15:
Jason, it is possible that you give reference image for loose feeder and examples from mixed loose feeder. Reference image need have the component at center and Other components should not touch it. Example images are then just images without the reference. Further  the pipeline to separate components from background, that background  part is black. I provide then reference stage and modified loose feeder.

cri-s, 2017-05-15:
I close this as the intent was to provide image and discussion for wiki.

closed 2017-05-15 by cri-s

vonnieda, 2017-05-15:
Hi @cri-s, there are some test images at https://github.com/openpnp/openpnp-test-images/tree/master/Parts/Loose but these don't contain the reference type images you mentioned. Maybe you or @dzach could make the references? I'm short on time and the loose part feeder is low on my priority list right now.

cri-s, 2017-05-15:
The point is that my lightning is far better as the average openpnp
user and as such not
replicable or usable as reference.
If it's low on you'r priority list, forget it.

vonnieda, 2017-05-16:
@cri-s Okay, I see. I'll put a few up tonight. I don't have many parts on hand right now, but I should be able to put up at least 4-5.

vonnieda, 2017-05-16:
@cri-s Okay, I tried taking some sample pictures, but it looks like my lighting is basically useless for this. Here are two examples:
[image]
[image]

Here is the first part, but with the head moved so the lighting is offset:
[image]

And here is the second part lit by a flashlight from about a 45 degree angle:
[image]

With the offset lighting the markings are clearly visible. So, I think I need to improve my lighting before I can make useful images. 

What do you think?

vonnieda, 2017-05-16:
Looping in @dzach since they may find this interesting.

cri-s, 2017-05-16:
Can you please check the soic chip at the approx same offset position
as the capacitor.
On capacitor the marking can be read on that position, probably on the
soic chip too.
It is possible to make a picture too with both components, just to
have annother picture with component rotated, eventually more then 90
deg.

vonnieda, 2017-05-16:
Sure, here they are:
[image]
[image]

And here are some more of all the parts I have handy:

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]