openpnp/openpnp #569: CvPipelineEditor improvements

dzach, 2017-05-21:
# Feature Request
## Describe the Feature
There is a number of little or bigger annoyances encountered by a user when doing work with the CvPipelineEditor. This request aims to describe functionality that could make the editor much easier and user friendly to work with:
* Entering values into parameter fields in the pipeline editor can be a time consuming work for a user.  So:
  * Reduce the number of keystrokes that are necessary to activate an entry field for keyboard data entry. 
  * Provide a `slider` widget for coarse adjustment of a field.
  * Provide a `range slider` widget for ranges of values.
  * Provide `spinner` widgets for numeric fields to achieve mouse-only entry.
* Setting/getting coordinates of features/shapes from the image display is hard. So:
  * Instead of having the user hover the mouse over the image and hold it there steady just to see x/y pixel coordinates and the pixel's colors, have a mouse click  freeze the values so they can be easily copied.
  * Provide copyable labels for copying/pasting display data into fields.
  * Allow for definition of shapes with mouse clicks, e.g. for defining custom masks on unwanted permanent features.
  * Get a color range from the image by drawing a rectangle or a polygon of the desired ROI.
  * Enter color ranges directly from the image into the property fields, e.g. for the extremely useful `MaskHsv` stage, by drawing a ROI rect or shape on top of the image.
  * Alternatively, provide for keyboard entry of color ranges, all into one field, and
  * provide for direct entry of color ranges into one field by defining a shape on the image

* Miscellaneous
  * Re-process only the stage that is being edited, when setting a value, for responsive live adjustments.
  * Pin the image that is displayed to a user definable stage, instead of only the current one. This way, a user may immediately see the effect of his changes to the result of another stage or to the final result
  * Change the color information from RGB to BGR, since this is what is mostly used with OpenCV and this is what `MaskHSV` understands for colors.
  * Display copyable RGB (or BGR) ranges along with the coordinates in the results panel.

## How Will It Look?
The last one described above could look like this:
[image]

## Why Do We Need It?
1. Helps the GUI become more user friendly, by shortening the time and effort to create opencv pipelines with good results. For everyone.

## Provide Examples
Is this something that exists in other pick and place software? Provide examples and references if possible.

I'm not aware of such but the above are common place nowadays.

vonnieda, 2017-05-21:
All good suggestions. One thing I'll note is that you can do the "pin the image that is displayed" one now. If you select a stage and then use the forward/back buttons in the upper right you can select a different output image while adjusting the current stage. A more intuitive way to do this, though, would be nice.

dzach, 2017-05-22:
Thank you for the "pin the image" tip! I totally missed it, if it is documented. 
I can write some of the code for the above features.

cri-s, 2017-05-22:
The Gui code for the pipeline editor is actually really unstable, longer string or bigger enums freeze the gui
for several seconds. For slider, some notes. short sliders are not usable because of missing resolution, update need to be fast, and resizing image give false impression or hide some pixels that can be important, slider need to know the start and end value, and this depend on color space, if limit is 1 or 180 or 255.

dzach, 2017-05-22:
The pipeline editor could also be implemented as a text editor with an optional popup menu to help choose parameters from, instead of memorizing and typing in their names. This would make some of the above "automations" a lot easier to implement. A console underneath the editor box would show messages from the parser, e.g. warnings etc. Something like the following, taken from the pipeline in the image above:

```
_____________________________________________________________________
0 ImageRead file ~/.openpnp/vision/test.png
1 BlurGaussian kernelSize 3
2 MaskCircle radius 200 >
3 ConvertColor conversion BGR2GRAY
# just mask a range of reds and pass everything else through
10 maskhsv huemax 126 huemin 36 redmax 45
S7 maskhsv huerange 36-116 saturationrange 0-200 valuerange 0 - 255 *
...
_____________________________________________________________________
> Warning: "radius" is not a formal parameter of stage "MaskCircle".  
  Default values  were used instead:
-> MaskCircle diameter 100
> Error: "redmax" cannot be mixed with "huemax" in stage MaskHsv.
_____________________________________________________________________
```
* Stage text could be case insensitive to ease keyboard input.
* Non existent parameters would be ignored with a warning.
* Wrong parameters would create errors.
* Partial but unambiguous parameter names would be accepted.
* Resulting stage images could be (optionally) written to disk or to a pipe, with a `> filename`
* The current view could be pinned to one particular stage with a `*`

If a parser for text commands, like the above, existed, then the editor could be implemented externally to the CvPipelineEditor, in any language, even as a remote editor written in HTML in a browser. Then sliders, buttons, etc could be implemented a lot easier by any developer having e.g. knowledge of HTML and javascript. Stage images written to disk could be utilized for fast browsing of stage results. Tools for GUIs from other languages/graphic environments exist, and some are vastly superior to those of Java.

dzach, 2017-05-22:
@cri-s 
I have your code for actuator buttons in daily use, auto-loading at startup, since you posted it, and it is a breeze of comfort. Something similar with the editor would be easy to test and verify without (yet) touching the editor's code.

How difficult would be to add, at first as a script, a rect drawing capability to get color ranges out of a ROI? That would be a really great improvement that could make defining a mask 10 times faster. Even just looking to the numbers instead of guessing would be a huge benefit.

dzach, 2017-05-22:
Snapshot of control panel created with @cri-s code.

[image]

dzach, 2017-05-22:
Since ideas come cheap:

The next incarnation of OpenPnP could be a hard core of functionality and logic, and a mechanism for executing commands. A fully documented API would be all that's required to allow developers create addons, or GUIs. These GUIs could be anything capable of sending OpenPnP commands and displaying their results. The existing GUI would still be needed for users to be able to run OpenPnP, but could be just one to of many to chose from.

GRBL has gone this way, devoting most energy to the core functionality as a controller with a minimal text interface, and others have created GUIs that take advantage of the core functionality.

dzach, 2017-05-29:
@cri-s , now that I'm doing some work with the CvPipeline editor, do you have any tip, or even better, any code snippet, on how to proceed with drawing a polygon on the editor's image to get the color ranges from that ROI instead of guessing and setting them every time by hand?

That will be a great addition to the editor. I'd prefer not to do this myself, but since I have suggested it and there is this haunting `Help wanted` tag, I thought I'd better ask for personal help myself.

vonnieda, 2017-05-29:
@dzach Unfortunately there are no shortcuts that makes this easy. The basic gist is:

1. Listen for mouse button down events.
2. Record the position where the button went down (start), and a flag indicating that you are selecting.
3. Paint a rectangle from the start to the current mouse position.
4. Repaint every time the mouse moves, i.e. listen for mouse movement or mouse drag events.
5. Listen for mouse button up events to end the selection and use the start and end position as your corners.

A more complex version of this exists in CameraView.java, if you'd like to see how it's done. That one also handles animating a dashed outline instead of just a line, and it allows you resize the selection area after the selection ends.

I just noticed you said polygon, so if you indeed mean an arbitrary polygon, you can change the above to just recording clicks (mouse listener) and draw lines between the clicks each time a new click happens. Somewhat easier, although you'll need logic for determining when the polygon becomes closed.

dzach, 2017-05-30:
Thank you @vonnieda . I'll go through CameraView.java  to acquaint myself with Graphics2D and OpenPnP, although it looks rather involved. The polygon I mentioned will always be closed by default, i.e. last point auto joined with first point.

vonnieda, 2017-05-30:
Okay, that does make it quite a bit easier. A rough outline of what to do would be:

1. Add a MouseListener for click events in MatView.java.
2. Each time there is a click, add it's location to a list and call repaint().
3. In paintComponent(), use the list of points to draw your poly (g.drawLine). 
4. Add some way to clear the list - maybe a right click.

The more interesting challenge, I think, is what you will do with the resulting poly. Unless you have extremely uniform lighting and a fairly solid color there are going to be lots and lots of variations in the pixels under the poly, so creating a range for that is going to be pretty tough, I'd think.

dzach, 2017-05-30:
We need three ranges, one for each B, G, R color, so I'd use the polygon to mask all outside pixels of the image black, and then find for each of B, G, R channels the extreme values (maybe remove outliers, do some binning, et.c). These will be the color ranges for the polygon area. At first I'd just display these ranges in the editor's panel for manual entry in `MaskHsv`. That's the idea, any way.

cri-s, 2017-05-30:
I suggest you to use external tools, python, gimp, ....
The scaling of image and the small possible sliders are not optimal if
embedded into gui.
gimp as example use 360 for max Hue, openCv 180 or 255 (Full)

dzach, 2017-05-30:
@cri-s The point is to get the range easily. If I understand it correctly `MaskHsv` can also be used for BGR masking so I was hoping to at least get an idea of the ranges in a polygon area and use it as a best guess before manually tuning `MakdHsv` parameters. Sure an external utility can be used, but unless it is called automatically, it will take more time to do that than to just take a wild guess and start fiddling with the parameter values right away.

I was hoping you might have some code ready for drawing polygons on an image in you magic box of tricks. @vonnieda 's tip from above seems straight forward and I'll give it a try.

dzach, 2017-05-30:
@cri-s Yes, sliders fit in that little space will not have enough resolution to be accurate, but still, they are a lot easier to use to get an initial range set, before manual tuning.

cri-s, 2017-05-30:
Do a blur or increrease the already used blur, and with the mouse get
the color that is displayed if you want it accurate and fast, just
check the HSB value, first value.
HSB is same as HSV_FULL . Ways better then what you'r propose.
Further you need setup it only once.

dzach, 2017-05-30:
@cri-s , that sounds very conservative. What you describe is _exactly_ how I'm doing it right now, and, I bet, how everybody else is doing it. If you run the OpenCV C++ demo for the `inRange` function, live with the sliders, you'll see how it feels to be able to adjust the ranges quickly and easily.

cri-s, 2017-05-30:
GaussianBlur at 333 or higher.
The problem is that without external tools, it is hard to do, and
implementing it is a lot of works
and a lot of users know too little of vision processing in order to
use it even if you do the lot of
works implementing a lot of tools.
As example for DetectCircle you need to see the canny results and
adjust it with sliders,
This when  you want setup something thats works reliable, not just
that works on this sample..
Further the resizing operation of image is evil, as it hides pixels or lines.
The standalone editor is a external tool itself having big problems as
some stage don't work
outside openpnp and as example selecting ROI for polarized parts
detection is only possible
inside openpnp and not inside standalone editor.

dzach, 2017-05-30:
Ok, the CvPipelineEditor is what it is, and has served OpenPnP well so far, but say now we want to make it better. What do you think it would it take to have a better external editor? How would you do it?

dzach, 2017-05-30:
Yes, that would be a lot of work, that's why I think that doing little improvements may be easier and little by little a pool of tools could be created that could be very useful in an future code overhaul.

dzach, 2017-05-31:
With the help of @vonnieda 's educational tips, drawing the polygon and scaling it was rather easy, about 15 lines of new code and some `infinite monkey (TM)` copying of existing code. Here it is, scaling with the panel:

[image]

Now to the essence.

vonnieda, 2017-05-31:
Nicely done!

cri-s, 2017-05-31:
If you run the Editor or standaloneEditor() from openpnp instead from
command line using bsh, you have access from bsh to all the images and
current selected class and can call external
python tools with just a menu entry.
Alternativly you can install a hook or event into wizard that stores
pipelineEditor inside propety or just pass the editor variable to the
script and use it from bsh.

Example for run standalone editor from bsh script:

	setAccessibility(true);
	se = new StandaloneEditor();
	se.setDefaultCloseOperation(JFrame.DISPOSE_ON_CLOSE);
	editor=se.editor;
	machine.setProperty("pipelineEditor",editor);
	machine.setProperty("pipeline",editor.pipeline);
	machine.setProperty("pipelineStage",editor.resultsPanel.selectedStage);
	editor.pipeline.setNozzle(gui.getMachineControls().getSelectedNozzle());

Running it from bsh gives you the possibility to run as example menu commands or
scripts that store the image inside config directory and call python, example:
  I assume you have loaded image result from pipelineStage into img.
 cmd is hsv.py in this case and passed as arg or hardcoded

Example of call
 Highgui.imwrite(arg=config.getconfigurationsDirectory().getAbsolutePath()+"/img.pnm",img);
 txt = ExecHelper.execUsingShell(cmd+" "+arg).getOutput();

As you have totally different idea of how image processing should be
done, i don't think
sharing the external tools make any sense and hsv/canny/houghCircles
tools with sliders
are already included into opencv samples.

dzach, 2017-05-31:
Thank you for the code @cri-s . At present I resist the urge to use external tools and I'm trying to make good with the tools available with OpenPnP.

I don't have a totally different idea on how image processing should be done, I just think of myself as being more or less a dump user. I do take advantage of the open source nature of this project and try to make things easier for myself, hoping that making it easier may help others too. In this respect @vonnieda 's KISS principle for pipeline stages really makes a lot of sense, as most of the code in this project is easily reusable. What a great think this is! 

So my view here is: "Let users explore the pipeline. Make it easier for them, even if they don't have a clue about computer vision". Selecting color ranges with a few clicks on the image is easy. Gimp has that "Select by color" tool that does something similar. That is a good feature, that's possible to implement in this editor too, e.g. select background color range with one click + a threshold color difference, and remove the range of colors by masking that color range. These two, the polygon selection and the one-click color range selection are very similar in nature and can probably be done in one go. One click select color seems even easier than the polygon and I may try it first.

So, just a single click on the image can provide a lot of "easiness" to a user: an xy coordinate with the color of the clicked pixel frozen and ready to copy, a color range selection within some tolerance zone. Add a few more clicks to define a polygon and you can be in control of the exact zone to extract the color ranges from.
 
But before all these, there needs to be a place to put these goodies to be copyable. That's the next thing to do.

dzach, 2017-05-31:
I very much appreciate your code snippets, and I copy them shamelessly when the need comes.

dzach, 2017-06-01:
Some more progress:

[image]

Coordinates and color of clicked pixel, frozen and copyable.
Ranges of colors can also be created easily from of a set of clicked pixels.

dzach, 2017-06-02:
I have now implemented:

* Setting/getting coordinates of features/shapes from the image display is hard. So:
  - [x] Instead of having the user hover the mouse over the image and hold it there steady just to see x/y pixel coordinates and the pixel's colors, have a mouse click  freeze the values so they can be easily copied.
  - [x] Provide copyable labels for copying/pasting display data into fields.
  - [x] Allow for definition of shapes with mouse clicks, e.g. for defining custom masks on unwanted permanent features.
  - [x] Get a color range from the image by drawing a rectangle or a polygon of the desired ROI.
  * Enter color ranges directly from the image into the property fields, e.g. for the extremely useful `MaskHsv` stage, by drawing a ROI rect or shape on top of the image.
  - [x] Alternatively, provide for keyboard entry of color ranges, all into one field, and
  * provide for direct entry of color ranges into one field by defining a shape on the image

* Miscellaneous
  * Re-process only the stage that is being edited, when setting a value, for responsive live adjustments.
  * Pin the image that is displayed to a user definable stage, instead of only the current one. This way, a user may immediately see the effect of his changes to the result of another stage or to the final result
  - [x] Change the color information from RGB to BGR, since this is what is mostly used with OpenCV and this is what `MaskHSV` understands for colors.
  - [x] Display copyable RGB (or BGR) ranges along with the coordinates in the results panel.

Here is an animated gif with the color range selection additions:

[image]

Click the image to view it in 720p.

vonnieda, 2017-06-02:
This looks really good @dzach, nice work!

cri-s, 2017-06-02:
what advantage you have to use such brute force segmentation ?
The normal segmentation see you here. I have used only official stages with no scripting.
The resulting rotatedRect get aligned where the wider axis are the X axis and then on the pins ,
here drawed in yellow, it is checked if the upper/lower rigth/left side contains more or less pixels and
if so the image/angle is rotated 180 degree. you have alread see that code.
On this type of components, no reference image is necessary. 
[image]

dzach, 2017-06-02:
Yes @vonnieda, it's really easy to get the first set of values and then with a few more clicks in areas that resist, expand it to include those too.

I like your solutions @cri-s , they are like tricks from a magician. Few people can do this. The demo here is not the "brute force segmentation" but the drawing tools. These drawing tools are intended for use by the average user; get them to experiment more and reach the end result quicker. My view was outlined a few posts earlier.

I've rearranged the toolbar to be at the top, and added a couple of icons for color ranges. Now that the basic drawing functionality is present, I might also add some more tools for drawing mask shapes. Key to all these is the ability to enter the shape coords directly to a stage's parameter or, at least, copy and paste them to a field in one move.

I can't create a new PR for these because they will be appended to the one that's pending. I might push these additions to a new branch in my repository and post a link so that you can test them.

vonnieda, 2017-06-02:
@dzach When submitting a PR, it should be submitted from a branch anyway. That keeps the changes separate. I typically create a "feature" branch for every feature or PR. I'll name it like feature/toolbars. Then when I submit a PR it only includes the changes to that branch, and you can submit other PRs from other branches.

dzach, 2017-06-02:
Ah, good to know it. Thanks!

dzach, 2017-06-03:
Pushed the feature to the [feature-CvEditor](https://github.com/dzach/openpnp/tree/feature-CvEditor) branch of my openpnp fork repository. Give it a try.

Instructions:
* Circles take the pixel color of their center. Many circles can be created. The color ranges will be the BGR extremes of these pixels.
* Polygons are created closed. To start a new polygon, hold the shift key down while left clicking. For color ranges, the extreme inner polygon pixels are used.
* To clear the drawing and the color ranges, right click on the image.
* Circles and polygons can coexist. Color ranges are the extremes of their union.

dzach, 2017-06-24:
As tools for the pipeline editor develop, there is a need for stage parameters to be settable by the GUI. At the same time there are some stages that need a different kind of parameter input, like the `/tmp/{part.id}.png` thing which came up in the LoosePartFeeder pipeline discussion. There is also a need for a generic `allParameters` field that would be parsed and would allow bulk setting of parameters either manually or programmatically. I've already used such a parameter in `MatchPolygon` where it is called `shapes`. The same need has arisen in my experimental ColorsToModel, where the color ranges are passed en masse to the stage by the GUI. There it is called `ranges`. I am working right now on passing shape data to MatchModel to be used as a mask, the same way it is used in MatchPolygon`'s `shape`. It is also possible that such a generic parameter might be used in other places where complex input might be required.

I need some advice with this matter:
It seems to me that a generic parameter named `allParameters`, `oneLiner`, or just `parameters` or even `data`, of type String would be flexible enough to be used in all the above cases and by all stages that need it. Its value will be parsed with a regular expression and could be, for example:
* a totally new structure that needs to be designed ahead of time
* an ad hoc key:value or dictionary construct. E.g. I was thinking of having such a parameter take values like `BGR:123:456,234:012,111:222; XY: 123,456` as ColorsToModel does right now when parsing the human readable text field, or `CIRCLE:123,456:122; RECT:123,456:789,555;RRECT:12,34:56,78:910,112:000,333;PLINE:123,456:124.235:...` etc for all kinds of shapes, that would also be human readable.
* a JSON object
* anything else

I have a practical need for such a stage parameter for at least `MaskModel / MaskPolygon`, and `ColorsToModel` colors and shapes, and having such a name ahead of time would be ideal.

So, what would be a good name for such a generic stage parameter?

vonnieda, 2017-06-26:
Hi @dzach,

I don't quite understand what you are proposing. Would `allParameters` (or whatever it is named) live on CvPipeline? Or are you proposing a generic name for a parameter that would be applied to stages that get their parameters from elsewhere?

For example, are you suggesting you'd replace MaskPolygon.shape with MaskPolygon.allParameters?

I think I'm probably misunderstanding, so if you could give a simple example I think that would help.

Thanks,
Jason

vonnieda, 2017-06-26:
@dzach Now might be a good time to mention how I see the future of CvPipeline models, properties and parameters working. This is just something I've been thinking about and it's not all nailed down yet.

1. We need a better way to add properties to the pipeline for use by stages. For example, we've recently added CvPipeline.nozzle and CvPipeline.feeder, but adding things like this to the pipeline is not a good long term solution. I intend to replace this with a general CvPipeline.setProperty(String name, Object value) in the near future. We'll call these "pipeline properties".
2. We need a way to specify substitution variables in parameter values. As you mentioned, this would be like `/tmp/${part.id}.png`. This would use standard Java property naming and would be able to reference pipeline properties as described above (`${pipeline.nozzle.id}`) and values from other stages such as their parameters, result images and result models. 

So, with these changes you would be able to do things like:
* `${pipeline.nozzle.id}`
* `${pipeline.nozzle.part.id}`
* `${pipeline.results.somePreviousStageName.image}`
* `${pipeline.results.somePreviousStageName.model}`
* `${pipeline.results.somePreviousStageName.model.someModelProperty}`

The benefit here is by doing the work to integrate the ability to set and read properties we open up a world of functionality that can grow in the future without requiring changes to the pipeline code every time someone wants to add a variable. In addition, it removes the need to tie the pipeline editor closely to OpenPnP. When we added Feeder and Nozzle we explicitly locked the CvPipeline code to OpenPnP, but it's always been my intention that it could be used outside of OpenPnP. by removing this tight coupling we get that back.

I think, and hope, that this might be a solution to what you are suggesting above, as well. If the GUI needed to pass something to the pipeline it can just call pipeline.setProperty("colors", someObject) and then stages could reference this using the variable substitution.

dzach, 2017-06-26:
Hi @vonnieda 

Indeed, this seems to be a very elegant solution.
> Or are you proposing a generic name for a parameter that would be applied to stages that get their parameters from elsewhere?

This. A generic stage name for serialized parameter input.  With the copyable / editable text pane that I've added to my experimental editor, I've realized that besides helping display lengthy values, like polygon coordinates or color ranges, it is also a quick and easy way to set parameters in bulk by hand, like a console for parameter editing. A proper name for such a generic parameter would make things more obvious.  I don't think that this is a concern to most users, though, nor is it an absolute necessity for the 2-3  experimental stages that can make use of it. 

Maybe we can wait for the new functionality you mentioned above and see what needs to be added then.

dzach, 2017-06-26:
Here is what I mean about the editable text pane:

[image]

Ideally it should be placed under the stage parameter list, and the parameter description should be moved to the split pane above the parameter lists, or become a popup tooltip.

cri-s, 2017-06-26:
Sorry for introducing in this discussion.
Preface:
 I have send reference.bsh for testing, it was finished and ok, before
making wiki,
but instead of testing  @dazach have copied part and modified it
resulting in having at
the end less functionality , but ok.

Before making such changes that can have sense or not  @dazach should
test his stage
on real usage, where black body touch together, pins are not so
cleanly separated as
seen under testing videos.

Further this is a special case because of colored background and the
attached picture show
the result of threshold operation of converted bgr2hsv , there is no
need to select any color in this case and .

Question: What real value have this UI drawing beside a programming
excercise for @dazach

dzach, 2017-06-26:
OpenPnP could make use of a generic drawing widget, both in the CvPipelineEditor and in the package footprint editor, and possibly in other places too. Once a basic drawing capability exists, its quite easy to produce shapes for all kinds of jobs. 

I'm finishing the functionality for producing masks with the mouse using the pipeline editor, and was thinking of replacing the `MaskPolygon` stage with a more capable `MaskModel`, that will also take model input from a new `ShapesToModel` stage. The trilogy will include `DrawModel`. Together with `ColorsToModel` they can form a group with all necessary functionality for producing and displaying models. 

These models will include:
- Point
- Polyline (possibly a new Result.Polyline)
- Circle (Result.Circle)
- Rect
- Ellipse (possibly a new Result.Ellipse)
- RotatedRect
- Polygon (possibly a new Result.Polygon)
- Contour
- KeyPont
- TemplateMatch
- whatever new comes in the future

vonnieda, 2017-06-27:
Hi @dzach, something to think about for some of the use cases you are describing are bean property editors. See https://docs.oracle.com/javase/tutorial/javabeans/advanced/customization.html. These allow you to create a UI that pops up when editing a specific type of property. This is how, for instance, the color selector pops up when editing a color property. This all happens automatically an behind the scenes.

I just thought of / remembered this earlier today, so sorry for not mentioning it earlier. This would be the best way to add the ability to edit properties in complex ways without having to add each new type of "thing" to the pipeline editor itself.

The value here is that it keeps the pipeline editor code simple while allowing for any level of complexity in individual property editors.

dzach, 2017-06-27:
Thanks for the tip @vonnieda !
This could also help with the problem of finding where to put the extra pane needed for editing long string parameters. A popup editor will be ideal, making an editor appear only when the user actually intends to use it.
Will have a better look at this in a couple of days.