openpnp/openpnp #573: Improvements in the Loose Part Feeder pipeline

dzach, 2017-05-27:
# Feature Request
## Describe the Feature
This feature is an outcome of the discussion in #543. Work has been done on @vonnieda 's outline of an improved loose part feeder pipeline that included a template creation method and a part-matching-template method. This proposal is a description and an implementation of those two methods. Code is provided herein for testing the proposed methods and commenting on further development and final implementation.
 
## How Will It Look?
There is no change in the GUI with this feature. However, to get an idea of "How It Will Work", take a look at these demo YouTube videos: [Loose part feeder, OpenPnP](https://www.youtube.com/watch?v=FbnqZSvf9pM) and [Loose part feeder and OpenPnP: Back into the strip tape.](https://www.youtube.com/watch?v=1hLC32XwLsI)

## Why Do We Need It?

Users of OpenPnP are already able to gather loose parts in a tray, and use the LoosePart feeder type to pick and place them on a board. This works fine with non polarized, passive SMT parts, like ceramic chip capacitors and resistors et.c. The way it works is by having the top vision system detect the outline of a loose part, filter it, produce a rotated rectangular model of it and deliver this model to the machine for picking and placing. 

This procedure works well for most rectangular parts, but it cannot distinguish between the corners of square parts, and it can not detect features on polarized parts, like diodes, transistors, ICs, electrolytic and tantalum capacitors and other active components. As a result, the exact orientation of these parts cannot be determined and there is a significant risk that they will be placed with the wrong orientation on the PCB. Determining the correct orientation of parts in a Loose Part Feeder is what the present feature aims to describe and provide code for.

## How it works
The work is done by two pipelines: 
* the `Crete Template pipeline`: creates template images of the desired parts 
* the `Match Oriented Template pipeline`: matches parts to templates created ahead of time and delivers oriented rotated rects to the machine. 

For the purposes of this work a conventional, or reference, "zero orientation" is defined, i.e. a well known and easy to remember angular position for the templates. The reference "zero orientation" used here is the same with that of a compass, i.e. "0 degrees up", or "North up", and angles increasing clockwise.

A second convention used in this work is that a model is considered "zero oriented" in its "upright" position if it is oblong (non-square), or, for square models, when an identifying mark, e.g. pin1, appears in the upper left corner, or near the upper edge.

### The "Create Template" pipeline

Create a new ReferenceLoosePartFeeder and edit its pipeline.

Steps:
1. Take a snapshot of the part in the newly created feeder.
2. Detect the part and create its model, a RotatedRect that describes it.
3. Isolate the part by masking out the background. For best results use the contour outline of the part as a mask.
4. Enter the name of the model stage in the `CreateTemplate.bsh` ScriptRun stage. This will make the sides of the model vertical and horizontal, by rotating it in the opposite direction of its present rotation, by the same angle, taking into consideration the conventions outlined above. This stage uses an `warpAffine` OpenCV transformation which also crops the image to the size of the rotated model. If the image appears upside down, or otherwise rotated (for square parts), set the `degrees` parameter to the appropriate (e.g. -90, 90, 180) angle to get it in its upright position. 
5. Save the image. This is the template.

Showing these steps with pictures:

1. Take a snapshot of the part

[image]

2. Detect the part and create its model, a RotatedRect that describes it.

[image]

3. Isolate the part. Here the contour of the part is used as a mask, so that background pixels around the part are masked out.

[image]

4. Use the `CreateTemplate.bsh` `ScriptRun` stage to transform and crop the image to the upright position

[image]

This is the template.

Code for this pipeline is provided in this zip file  
[CreateTemplate.zip](https://github.com/openpnp/openpnp/files/1033800/CreateTemplate.zip). The Beanshell script file can be saved in (a folder in) the `Scripts` directory for easy access. The code is set to work as a ReferenceLoosePartFeeder pipeline, but can also work with the StandaloneEditor for CvPipelines.

For testing, it is suggested to use the StandaloneEditor and a stock of test images from [here](https://github.com/dzach/openpnp-pipeline-tests). Just enable the `ImageRead` stage at the top of the pipeline. To run the StandalonEditor in Linux, enter:
```
java -cp target/openpnp-gui-0.0.1-alpha-SNAPSHOT.jar org/openpnp/vision/pipeline/ui/StandaloneEditor &
```
at a command line prompt and in the editor window that opens choose the directory with the test images, with the button at the upper left.

Coming up next: The "Match Oriented Template" pipeline.

dzach, 2017-05-28:
Apologies for not posting yet the code for the "Match Oriented Template" pipeline, but that this addition will make working with the Loose Part Feeder even easier.

Added a new ScriptRun stage, `PartTemplateWrite`. This is the new zip file of a working demo: 
[new-CreateTemplate-pipeline.zip](https://github.com/openpnp/openpnp/files/1034232/new-CreateTemplate-pipeline.zip) . Using this stage instead of the `ImageWrite` can automatically save a properly named template of the part detected by `CreateTemplate` as follows:

When run from inside OpenPnP, e.g. either by using `Edit Pipeline`, or by feeding a part in a LoosePartFeeder that includes `CreateTemplate` and `PartTemplateWrite`, the stage will now do the following:
1. Check if the `templateFile` name has a `.png` extension. If so, then it will proceed with saving the specified image file.
2. If `templateFile` name does not end with  `.png`, then it is assumed to be a directory holding part templates. The stage then tries to get the id of the part loaded in the LoosePartFeeder that the pipeline may belong to. If a feeder exists and has a part loaded, then the part's `id` will be used with a `.png` extension as the `templateFile` name. The template image will be saved in the directory specified with `templateFile`. The directory will be created, if it does not exist.

With this addition, part template creation can be fully automated for each part loaded in a properly set LoosePartFeeder. The following images show how an example of such a setup.

[image]

Notice the `ReferenceLoosePartFeeder` named  _Create Template_ that is highlighted, **but not enabled**. Editing its pipeline, or just feeding one part by pressing the [image] icon, will create a new template of a part detected in the feeder, in this case a `TO-252-2Lead-LM317AMDT` from a `.pos` file created with KiCad. However, since the feeder is not enabled, it will not take part in an automatic place operation, if the part that happens to be loaded also exists in the current job.

The pipeline is a generic one (i.e. it does not refer to a specific part), and looks like this:

[image]
Notice that `templateFile` is set to a directory name instead of a `.png` file.

To make a template for a new part, just replace the part in the feeder and either feed one part by pressing the [image] icon, or edit the feeder's pipeline.

Now, the `MatchOrientedTemplate` pipeline too will load the template that corresponds to the part loaded in the Loose Part Feeder automatically, if the template image already exists.

vonnieda, 2017-05-28:
This looks really great @dzach! The video putting the parts back into the tape is really impressive!

How would you like myself and others to respond to this? Are you looking for more testing? Do you need help or guidance turning it all into a usable feeder implementation? How can I help?

This is probably a good opportunity to clarify my feelings on simple vs. complex stages. I was thinking a lot about this recently in the context of our discussions on some of your other stages. I think that I can generalize my thoughts on this with the following:

1. If a stage is likely to be generally useful to many users and many pipelines, try to make it it's own small, self contained stage. Examples are things like Crop, ConvertColor, Rotate, Scale, etc. These are primitives that can be used in many ways.
2. If a stage implements significant logic and complexity for a specific task, name it according to that task, rather than in generic terms. Examples could be things like BottomVisionSURF, FiducialTrainingNN, LoosePartTemplateDetection, etc. 

For example, I think it would be fine to take the latter parts of this template stage, along with the code that is in the script, and turn that into a "LoosePartTemplateStage" or something like that. It indicates what the thing does and also indicates that it might not be something you want to mix in to some other random pipeline. Rather than trying to come up with an all encompassing name that describes what it does like "RotateAlignOrientLongSideTemplate" or something like that. 

I hope that's helpful and not just more confusing. In any case, I've given it a lot of thought and I will be less stringent on what gets in. I think this will all be eased considerably when we add the category tree to the list, anyway.

dzach, 2017-05-28:
@vonnieda I'm glad you like the way this came out. Credit goes to you and @cri-s for the help you provide. 

I totally agree with your view for the pipeline stages, having them simple or complex . In the process of writing the code for this feature I have created both simple and complex stages, trying to see which ones will work better in this context. This is not an easy decision, and it will always be an issue for discussion. I don't mind that at all, to the contrary, it helps me get a better understanding of the issues. I plan to post a list of such simple stages that have been produced in this process, to see which might be eligible for inclusion in OpenPnP.

This last thing that I added really makes it easy to automate as much as possible working with the loose part feeder . Which in turn may make using such a feeder more appealing to users.

I still have to publish the code for the rest. The code is ready, but writing the "manual" is not as easy as writing Beanshell scripts, and I don't want to just throw in some plain files.

How to proceed? What I was thinking is have those few who watch these pages test the code and get more results, both successful and failed, and see what can be improved. The code is surprisingly simple, thanks to the provisions you already made ahead of time in the structure of this project. Ideas, help and guidance are more than welcome, and I look forward to them. This is a collaborative effort with all the joy of open source present.

If not much needs to be changed/added any more, then announce this on the mailing list and get as much feedback as possible, for a week or so. Then proceed and make the scripts regular java stages.

We'll need to discuss again the "simple or complex stages" issue, after you guys try the code. In some cases there is hardly any new stuff going on, just glue code joining existing stages.

Thinking about it, I better post the code now and come back and add the "manual", than keep it for when it  will all be ready. Life can get in the way, when you least expect it. So here it comes: 

[PartTemplateMatch.zip](https://github.com/openpnp/openpnp/files/1034322/PartTemplateMatch.zip)

I have already renamed the stages, as it was difficult even for me to make out which one is which among the simple ones and the glue ones. When all this naming thing settles down, I may edit the previously posted names for clarity.

dzach, 2017-05-28:
Here is a problem I have not been able to pinpoint yet. When I use a rotation value in the feeder, it doesn't change anything in the placement angle, no matter what. I can probably try to read this value from the feeder and add it to the result of the pipeline, but is this the right way to do it? Isn't it supposed to be added by the feeder code? Placing at different _part_ angles shows no problem, only the feeder's rotation is ignored.

dzach, 2017-05-28:
### Match Oriented Template pipeline

This pipeline makes use of multiple `MatchTemplate` stages to get the best matching between the template and the part.

To do that, the template is rotated so that it matches the rotation of the part and then rotated 3 more times in sequence, each by 90 degrees, to create all possible matching positions. These 4 rotated template images are then matched with 4 corresponding `MatchTemplate` stages and their results are compared to get the best matching.

The steps:

1. Get an image from the loose part feeder this pipeline belongs to.
2. Convert color, mask, process the image to get the rotated rects of the parts in the feeder.
3. Read in the template to be used by the feeder to match the part loaded. For that, either:
  * set the name of the image file for the template, or 
  * set the directory such templates can be found in. In this case the id of the part loaded in the feeder will be used to form the filename for the template. 
  * Input is expected to contain more than one part. Loop through their detected rotated rects to find the one closest to the center of the camera. While looping, filter out the ones that do not fit the size of the template +/- a tolerance
  * If a closest RotatedRect fitting the template's size is found, its rotation angle is transfered to a `rotate` stage **downstream** in the pipeline.
  * This step is implemented by the `PartTemplateRead` ScriptRun stage.
4. Process the template so that it is suitable for use with the image in step 1.
5. Rotate the template 4 times starting from the angle of the detected RotatedRect.
6. Use one `MatchTemplate` stage to match each of the rotated templates.
7. Compare the results of all 4 `MatchTemplate` stages and get the winning one. The angle of the winning template is the true orientation of the model and is set as the result of the pipeline.
  * This step is implemented by the `PartTemplateMatch` ScriptRun stage.

The pipeline is called multiple times by the feeder, each time picking the closest one to the center of the camera, until the parts are exhausted and no part is detected any more.

Here are the above steps illustrated with screenshots of the stages at work:

[image]

Stage "10" outputs the `RotatedRect` of the part, as seen in the results box. Notice that the `RotatedRect` has its `width` larger than its `height`. This is opposite to the convention used in this proposal, so the rect will be converted internally to one that conforms :

[image]

Stage "tmpl1" is the template rotated by the amount of the rotation of the `RotatedRect` above, _after internally correcting width, height and rotation angle, as shown in the parameter degrees=-33.231712341308594_:

[image]

The other template stages are rotations by 90deg of the previous one:

[image]

[image]

[image]

The rotated templates are matched with the masked part:

[image]

and the winning stage, "match3" with "tmpl3", is used to get the orientation of the part.

[image]

A line is drawn showing the result.

[image]

dzach, 2017-05-29:
Here is a rundown on the pipeline, and my thoughts about how it could be changed to be simpler and maybe better:

[image]

* The stages from 00 to 10 are preparatory and can be anything necessary to deliver the `RotatedRects` of interest to the "match templates" mechanism.
* The stages _tmpl1_ to _tmpl4_ are just rotations of the same template image. The stages _match1_ to _match4_ are identical, except for their template input, which is one of the _tmpl1-4_ above. They are preceded by identical `ImageRecall` to get the masked initial image from stage 17 they operate on. These repetitive identical stages do not offer any extra flexibility by been separate, they just increase the length of the pipeline and make it look complex
* The template has to be read from the file system and be processed. For that, either the template filename, or the directory it resides have to be set. However, stage `PartTemplateRead`, named like `ImageRead` with the intention to separate it from other kinds of templates, does a lot more than just reading in the template image; it sets the initial rotation in a stage **downstream** the pipeline, and filters the `RotatedRects` from stage 10 to fit the size of the template and eventually get to the one to focus. The latter is necessary because:
  * The template's size is the best representation of a rectangular filter with which the `RotatedRects` will have to be screened at some point, sooner or later. Sooner makes sense and could have been done earlier, but that would cost another loop delay and would need judgment about the size of the filter, while now it is done automatically. 
  * The feeder will pick only one out of a group of `RotatedRects` and will repeat the process by centering 2 more times before it picks one. It is practical to deliver just one oriented `RotatedRect` to the feeder, each time it calls the pipeline.
* The `PartTemplateMatch` script glues the logic together and does the necessary comparisons of the `MatchTemplate` stage results. This cannot be done in any other simple stage, unless flow control stages are introduced to the pipeline.
* Bringing the models to a conventional orientation, e.g. upright, is necessary to insure consistency of code and logic.
* Drawing of the orientation line, or some other mark should be done separately with a simple stage.
* It seems to me that the `PartTemplateRead` stage should be lighter, and most of its filtering and choosing logic done later.
* Not much is gained by having repetitive stages exposed in the pipeline. Rotating and `MatchTemplate`ing can be done internally in one "complex" stage. Intermediate results seem not to be of interest for further processing. Image processing stages however seems necessary to be separate, because the are dealing with the part that varies greatly with every installation of OpenPnP.

So it all comes down to the following:
* Read the template automatically, if possible (new simple stage).
* Process the template.
* Get an image of the parts in the feeder
* Process that image and produce RotatedRects
* Filter RotatedRects, choose closest to center, put it to conventional "upright" position, rotate template and MatchTemplates all done in one new complex stage. One set of `MatchTemplate` parameters, one model stage name input, one template stage name input and maybe some more optional parameters, like whether to use 2 comparisons instead of 4 for oblong rects. 

So. that means one simple stage like `PartTemplateRead` or 'LoosePartTemplateRead` and one complex stage with the glue code, like `PartTemplateMatch`, for the rest. Optionally, draw orientation with either a complex `DrawModel` (I already have one that draws most of the models produced) or with a new simple one `DrawOrientation` or `DrawAngle`.

vonnieda, 2017-05-29:
@dzach I haven't had time yet to go through these posts, but I will soon. I just want to reiterate, though, that if you want comments from anyone aside from myself and @cri-s you should post this to the mailing list. No one aside from a small handful of people read every issue posted.

dzach, 2017-05-29:
@vonnieda we will certainly post the feature to the mailing list, but I'd prefer to have the second pipeline in its final form before we do. The way it is today looks unnecessarily complex and has some hard-coded parts that make it very inflexible, e.g. tmpl1-4 and match1-4. I did it this way trying to make use of simple existing stages as much as possible, but I don't like the way it turned out very much. The pipeline was even lengthier some iterations back, with separate Crop, RotateModel and some other stages. 

So, I was looking for some comments on that matter, and I think the end result would be much simpler to understand and a lot easier to use than it is now.

dzach, 2017-06-05:
I've made good progress with the loose part pipeline. Processing time is now significantly less: even with all new stages written in Beanshell scripts the total processing time is well under 1000ms, and some parts like SOT23 are around 350ms, running the standalone CvPipelineEditor. Running from inside OpenPnP seem to take a few hundred ms longer.

The multiple `MatchTemplate` and `Rotate` stages are now included in one stage which I call `MatchPartTemplate` that does the 4 rotations and matches of the part in one go. Process time savings come mostly from cropping the part image so that it is just a little larger than the template image. Some ms were saved by doing 90deg flips instead of 90deg rotations. The stage that reads the template is now very simple. Filtering and choosing the closest rect is now a separate stage, since it may be useful in other cases too.

This is the solution I prefer. All new stages are simple, except for the last one that does the matching. The new pipeline+code will be posted tomorrow and I will start converting the beanshell script to java stages.

dzach, 2017-06-05:
Cleaning the code from my experimental stages proved to be time consuming. I've checked this  version with a vanilla `develop branch` installation.

I experimented a little with different camera resolutions. It may seem silly, but a number of templates made with the camera in 1280x1024 resolution work with the cameras in 640x480, although matching scores are a lot lower; I've set the threshold to 0.2 and it seems to work. I've included a scale factor to script stage `ClosestModel`, to scale down the filter in there. With the lower resolution, processing time of the whole pipeline is absolutely minimal. 

I have so far ignored warning about the loose part feeder being experimental. I would appreciate some more info on what this means and how to proceed. I have not been able to find an easy way to follow the ReferenceLoosePartFeeder code. The feeder rotation seems not to be included in the location calcultations, but it is needed to adjust the "conventional orientation" between the potentially different ideas OpenPnP and the CAD layout program have for each part.

In any case, here is the new code. 
[scripts.zip](https://github.com/openpnp/openpnp/files/1053357/scripts.zip)

dzach, 2017-06-05:
I've already transferred to Java the script stage `MatchPartTemplate`. Processing time is approximately a third to a half of the that of a script. I mentioned that because I was afraid that the time would be prohibitively high. The videos show a significant delay between detections. Not any more. I'll try to do one follow-up video with the new code.

dzach, 2017-06-06:
Added a new short video showing the improved pipeline: From strip tape to loose part feeder and back into the tape. https://www.youtube.com/watch?v=DaDuDr0hzRo

dzach, 2017-06-06:
Fixed a couple of bugs. Replaced [scripts.zip](https://github.com/openpnp/openpnp/files/1055183/scripts.zip) .

cri-s, 2017-06-06:
Without support from  pipeline, you need something like this:

	List<RotatedRect> pickLocations = null;
	int idx=-1;
	
    @Override
    public void feed(Nozzle nozzle) throws Exception {
        Camera camera = nozzle.getHead().getDefaultCamera();
        // Move to the feeder pick location
		pickLocation=checkPickLocations(camera);
		while(pickLocation==null) {
						{ pickLocation=location; }
			if(idx>=1) { if(idx==1&&pickLocation.equals(location1)) { idx++; }
						 pickLocation=location1; }
			if(idx>=2) { if(idx==2&&pickLocation.equals(location2)) { idx++; }
						 pickLocation=location2; }
			if(idx>=3) { if(idx==3&&pickLocation.equals(location3)) { idx++; }
						 pickLocation=location3; }
			if(idx>=4) { if(idx==4&&pickLocation.equals(location4)) { idx++; }
						 pickLocation=location4; }
			if(idx>=5) { if(idx==5&&pickLocation.equals(location5)) { idx++; }
						 pickLocation=location5; }
			if(idx>=6) { if(idx==6&&pickLocation.equals(location)) { idx++; }
						 pickLocation=location; }
			if(idx>6) { idx=-1;
				throw new Exception("Feeder " + getName() + ": No parts found.");
			}
			pickLocation=vision(camera,nozzle,pickLocation);
			if(pickLocation==null) { idx++; }
		}
    }

    private Location vision(Camera camera, Nozzle nozzle,Location
location) throws Exception {
        MovableUtils.moveToLocationAtSafeZ(camera, location);
        // Process the pipeline to extract RotatedRect pickLocations
        pipeline.setCamera(camera);
        pipeline.setNozzle(nozzle);
        pipeline.setFeeder(this);
        pipeline.process();
        // Grab the pickLocations
        pickLocations = (List<RotatedRect>)
pipeline.getResult("pickLocations").model;
        if (pickLocations.isEmpty()) {
			return null;
          // Find the closest result
        }
        MainFrame.get().getCameraViews().getCameraView(camera).showFilteredImage(OpenCvUtils.toBufferedImage((Mat)
pipeline.getResult("pickLocations").image), 1250);
		return checkPickLocations(camera);
    }

	private Location checkPickLocations(Camera camera) {
		if(pickLocations==null) { return null; }
		while(pickLocations.size()) {
			int i=Math.random()*(pickLocations.size()+1);
			RotatedRect  result = pickLocations.get(i);
			pickLocations.remove(result);
			Location loc = VisionUtils.getPixelLocation(camera,
result.center.x, result.center.y);
			MovableUtils.moveToLocationAtSafeZ(camera,location.derive(loc.getX(),loc.getY(),null,null));
			pipeline.process();
			rect=null;
			Result res = pipeline.getResult("result");
			if(res!=null&&res.model instanceof RotatedRect) {
rect=(RotatedRect)res.model; }
			if(res!=null&&res.model instanceof List) { List l=(List) res.model;
				if(l.size()>0&&l.get(0) instanceof RotatedRect) {
rect=(RotatedRect)l.get(0); }
			}
			if(rect==null) { continue; }
			loc = VisionUtils.getPixelLocation(camera, result.center.x, result.center.y);
			MainFrame.get().getCameraViews().getCameraView(camera).showFilteredImage(OpenCvUtils.toBufferedImage(pipeline.getWorkingImage()),
1250);
			return new Location(LengthUnit.Millimeters,loc.getX(),loc.getY(),location.getZ(),location.getRotation()+rect.angle);
		}
        return null;
	}

dzach, 2017-06-06:
Renamed the stages and replaced all zip files to this one, to avoid confusion. Please update any previously downloaded files.

[LoosePartFeederPipelines.zip](https://github.com/openpnp/openpnp/files/1055753/LoosePartFeederPipelines.zip)

dzach, 2017-06-06:
Hey @cri-s thanks! Can you be more verbose? Where does this code go?

dzach, 2017-06-06:
Posted [New LoosePartFeeder pipeline, detecting polarized parts](https://groups.google.com/forum/#!topic/openpnp/zqeeh6mGqtk) in the mailing list

vonnieda, 2017-06-07:
> I have so far ignored the warning about the loose part feeder being experimental. I would appreciate some more info on what this means and how to proceed. I have not been able to find an easy way to follow the ReferenceLoosePartFeeder code. The feeder rotation seems not to be included in the location calculations, but it is needed to adjust the "conventional orientation" between the potentially different ideas OpenPnP and the CAD layout program might have for each part.

It was marked experimental because at the time I released the code, I didn't really feel it was ready for release. Several people were asking for it to be in the codebase so they could experiment with it. The primary problem was the polarization issue, which is what you've been working to solve.

As for rotation, all you need to do is add getLocation().getRotation() to your result in getPickLocation(). The general process would look something like:
1. OpenPnP calls feed(), which moves to the general location of the parts. Currently it moves to the rotation angle, as well, but this is a no op with the camera, so it doesn't matter.
2. feed() calls getPickLocation() 3 times, finding the closest match each time, and ideally homing in on it.
3. feed() returns.
4. OpenPnP calls getPickLocation() to get the location that it should pick from. 

Right now, on line 104, we setup the location that will be returned as the location of the found part and the we include result.angle, so that we pick the part at "0" degrees. All you have to do is change that to result.angle + getLocation().getRotation(). This means that OpenPnP will pick the part at the vision angle plus the configured angle, and when it rotates back to 0 it will be at true 0.

For example:

1. Feeder's location is 10,10,0,90
2. We perform vision and find a part at 11,11,0,13
3. We return 11,11,0,103 (90 + 13)
4. OpenPnP moves the nozzle to 11,11,0,103
5. OpenPnP picks the part
6. OpenPnP rotates the nozzle back to 0, which removes the vision angle plus the specified angle.
7. The part is now oriented at the user's preferred angle (90) with regard to it's "correct" 0 angle based on vision.

That might all sounds a bit complex, but what it really comes down to is line 104 should change to
```
location = location.derive(null, null, null, result.angle + getLocation().getRotation());
```

And that should take care of it.

cri-s, 2017-06-07:
You need to return
pickLocation.derive(null,null,location.getZ(),location.getRotation()+pick
location.getRotation ());
instead of just pick location inside getPickLocation.
Further there is no reason to call 3 times getPicklocation and returning
the result nearest the nozzle is problematic.
Returning Math.random()*results.size() element is a lot better as this
avoid looking up feeder on retry.
You don't need only return rotation from location, Z is required too.

dzach, 2017-06-07:
@cri-s what you describe has to do with the whole logic for the ReferenceLoosePartFeeder, not just this pipeline. Randomizing the rect to pick sounds like an interesting idea, but the path I've taken for the pipeline is to deliver a single RotatedRect, in order to avoid running matchTemplate x 4 for each one of the detected rects in the image, which would be prohibitively slow.

So the only thing left to be done is to home in on this one rect, and get its position as accurately as possible, because it may be used directly to place the part, i.e. without a follow up of a bottom vision alignment.

I'm currently finishing transferring beanshell code to java and will have a test branch ready hopefully by tonight.

dzach, 2017-06-07:
@vonnieda thank you very much for the detailed explanation. I've applied the fix you provided and all looks good now! 

There might still be an issue with the convention I've used regarding the direction of rotation, e.g. whether +90 is clockwise or counterclockwise. I've just checked with KiCad which I have handy, and it seems it is the opposite of what I have used, .i.e. angles increase counterclockwise there while they increase clockwise in my code. This is not an issue with feeders where parts are in a fixed orientation, since the orientation of the feeder will correct them all, but might be a problem with a loose part feeder where the parts came in random orientations. To take care of any eventuality, there may need to be a parameter specifying the direction of rotation. 

I have now concluded the translation of the code to Java and will push to my depository a branch with all these stages.

dzach, 2017-06-07:
@cri-s my understanding of the internals of the ReferenceLoosePartFeeder comes solely from  @vonnieda's explanation above. I can see that .getZ() is taken into account in the next line of code (line  106) but cannot comment any further.

vonnieda, 2017-06-07:
Hi @dzach, yes, angles should increase counterclockwise. This is the standard that OpenPnP uses and it's used in many other systems. See https://github.com/openpnp/openpnp/wiki/User-Manual#coordinate-system for more info.

cri-s, 2017-06-07:
Yes, the getPickLocation function is inside ReferenceLoosePartFeeder,
Actually the ReferenceLooseFeeder calls the pipeline 3 times just for fun.
Returning just one location is dangerous, as if feeder need to repick
because of failed pick,
as example not detectiong correct vacuum level with the pipeline you
just return the same solution, and this solution could be not
pickable, just because the component is a bit tilted
or because its orientation is wrong and as result the height is wrong
that don't allow pickup the
component.
Returning just one solution don't allow to check on setup if as
example all resistors get
recognized or not.
As example i'm attachine picture and pipeline for detecting resistors or caps.
Doing the same for diodes or leds require 3-4 additional stages.

vonnieda, 2017-06-07:
@cri-s It's not "just for fun". The purpose is to eliminate camera scaling error and get closer to the exact position of the part. It's the same reason that we do three checks in the fiducial code.

It is assumed that the part will always get picked, so subsequent checks would not find the same part over and over. I do agree with the approach in your code though - if there are multiple good results, picking one at random rather than always picking the closest is a good idea.

dzach, 2017-06-07:
@vonnieda I've checked with an actual test pcb in KiCAD and it works fine, because I already negate the angle before delivering it. This presents a problem though with drawing the solution, as DrawRotatedRects draws these rects with the wrong angle.

vonnieda, 2017-06-07:
@dzach We probably need to invert the angle in the LoosePartFeeder code. We do this in the bottom vision code for the same reason: https://github.com/openpnp/openpnp/blob/develop/src/main/java/org/openpnp/machine/reference/vision/ReferenceBottomVision.java#L118

Just change line 103 to return the negative of the angle.

dzach, 2017-06-07:
Thanks @vonnieda

I did a little test and that last inversion in _ReferenceLoosePartFeeder.java_ seems to have done it. I have to set the rotation of the feeder to -90°, but that should be expected since KiCAD's 0° is -90° for my "compass" 0° convention. It doesn't really matter, but I might consider changing that in a next step, or add a choice parameter in `MatchPartTemplate`.

The pipeline with all the new stages has been pushed to https://github.com/dzach/openpnp/tree/develop-loosePartPipeline. 

Processing time for the whole demo pipeline _LoosePartFeederPipeline.xml_, which is included in the root directory of the repository, is approx 483ms, of which 410ms is the time needed to process `ImageCapture`, leaving approx 73ms for all of the rest. The `MatchPartTemplate` stage that does the rotation and matching is around 16ms in this machine. That is 2 orders of magnitude less that where I started. I do a 6 part pick and place in-place test and this is the result:
```
2017-06-08 01:20:11 ReferencePnpJobProcessor INFO: Job finished 6 parts in 31.8 sec. This is 678.4 pph
```
Please give it a try and suggest/do any corrections that seem necessary to make it better.

cri-s, 2017-06-07:
I think for loose feeder calibrated camera is needed.
Further if ReferenceStripFeeder need camera calibration and don't work
without or at least
wizard and autosetup don't work, i think for looseFeeder it should be
the same rules.
Further I'm against taking capture at arbitrary location without cropping.

For fiducials 3 attemps make sense for uncalibrated camera and the
penality for doing it with
calibrated camera is low.
For loose feeder, this is not the case.
If pick and place position is off by 0.2 mm it don't matter  for 0402
(imperial) size.
and if it's off 0.4mm it woks for 0603. Bigger components allows error
up to several mm.
Uplooking vision does fine align, nothing comparable with precision
needed on fiducials..

If prototype assembly with 25 loose feeders and 150 position is needed
in addition to some strip feeder with annother 32 positions for total
of 182 placements.
Assuming the average pipeline needs 0.8 second and machine runs at 360
cph, need 10 seconds
for one pickNplace.
150*2.4=360 Second for vision + 1500 Seconds + 320 seconds for strip
feeder is total time of
37 minutes for assembly with -17.5% for loose feeder vision penality.
Now on machine that can 1800 cph or that need 2 seconds for every component.
182*2=364 seconds for placing parts, and now it matters if vision need
120 or 360 seconds
as cph goes from 1800 to 1350 (-25%)  or 900 (-50%) cph.

If loose feeder have 25% penality, it is used, with 50% penality, the
usage is limited to rare cases.
You maybe saying, average pipeline time is more 0.6 sec and not 0.8 sec.
That is maybe true and the above statements remains still valid.

dzach, 2017-06-08:
Since this pipeline is delivering just one rotated rect to OpenPnP, for performance reasons, it is the pipeline that will have to adopt the strategy for choosing "which one" to deliver. 

Therefore, I might need to implement @cri-s suggestion, maybe in addition to the 'closest' one, and have the user choose. I'll have to rename the stage that does it from `ClosestModel` to something like `AutoChooseModel` or something more intuitive.

In fact, I just did. As it comes out, with 3 tries, each time a different one is picked. In the end, whichever happens to be the _chosen_ one the 3rd time around, that one is picked. 

Judging from the amount of movement the second and 3rd time, when homing in on the closest, my system seems to be  mis-calibrated. I suspect other users may too suffer form similar mis-calibrations, and there seems difficult to find one solution for all. It seems that 3 homing tries make it possible to pick and place directly from the loose part feeder, without any bottom vision alignment.

vonnieda, 2017-06-08:
I agree @dzach, I think leaving it to the pipeline to determine what to return is the way to go. This keeps it flexible.

vonnieda, 2017-06-08:
HI @dzach, I had a brief look at this tonight. I was not able to get the matching pipeline working yet, but I didn't have a lot of time to spend on it. I did get the template create pipeline working and that looks pretty solid. I'll spend more time on it later this week.

dzach, 2017-06-08:
I'm sorry to hear that @vonnieda . The script stages have plenty of debug `print` statements to figure out what happens in every step, but I've omitted them in the Java stages. I also omitted input checking Exceptions and now return `null` to let OpenPnP deal with no detection, with it's popup message. Maybe that's not the way to do it?

One critical stage to watch is `ClosestModel` where filtering happens. Doing this silently may not be a good idea. I'll push a new version with debug logging as soon as possible.

cri-s, 2017-06-08:
What are the speed diff of java and bsh?
Are you interested in speedup of bsh ?

dzach, 2017-06-08:
Hi @cri-s 
It's 2/1 to 3/1, Java/Beanshell. 

Sure I'm interested! I have to find a way, though, to mimic Java in returning results. Using nested `if`s isn't so friendly.

dzach, 2017-06-08:
Fixed a naming error in _LoosePartFeederPipeline.xml_ that was preventing `MaskModel` to mask the result.

dzach, 2017-06-08:
Pushed a number of fixes. The pipeline should work "out of the box" now (LOL!), after adjusting `MaskHsv` and other image processing parameters.

vonnieda, 2017-06-14:
Hi @dzach, First, thank you again for all the work you've put into this. The feature is looking great and lots of people are really excited to have this functionality in OpenPnP!

I'd like to start discussing what it will take to get this work merged and usable for the community. My goal here is to figure out how we simplify the use and installation of this so that it's as simple as creating a new feeder. Here is a rough review of what I think we'll need to do to get this ready:

# UI and Feeder Changes
* Modify the ReferenceLoosePartFeeder (RLPF) Wizard to add controls for managing a Train pipeline, similar to the existing controls for the main pipeline. So we'd end up with a Train pipeline and a Feed pipeline.
* Modify RLFP Wizard to include Train button which basically just fires off the Train pipeline.
* Move the CreateTemplatePipeline and LoosePartfeederPipeline into the default pipelines for the `feeder at src/main/resources/org/openpnp/machine/reference/feeder`, so that they are loaded into new instances of the feeder.

I realize that the UI is not your area of expertise, and I'd be happy with help with this part.

# Stage Changes
## CreateTemplateImage
* Needs a better name. Remember, these stages are global in nature and need to be generally useful. In this case I think a user's questions would be: Template for what? Create based on what?
* What about replacing this with OrientRotatedRect stage that lets you set Portrait or Landscape and then a Crop stage that crops to a RotatedRect input?

## ReadTemplateImage / WriteTemplateImage
* Again, there are naming issues here similar to CreateTemplateImage above.
* These should also use the functionality in Configuration.createResourceFile and Configuration.getResourceFile so that the template files are stored in the configuration directory. Having the user specify a directory for these should not be necessary. Note that this would tie it into only running inside OpenPnP, so that may be a concern.

## ComposeResult
* Remove the image portion of this and rename to ModelRecall. Can then be used in a pipeline as ImageRecall followed by ModelRecall instead of duplicating the functionality of ImageRecall.

## MatchPartTemplate
* The matMaxima code appears to be duplicated from MatchTemplate. Should be moved into a utility function and both should use it, rather than duplicating it.
* Needs a better name. I don't think there is anything here that is Part specific. So maybe just something like MatchTemplateQuadrants or something like that.

## ClosestModel
* Needs a better name. It only handles RotatedRect, as far as I can tell, so it should probably note that.
* Seems like it's really just a composition of several other simpler stages. Filter and find closest match. Maybe needs to be broken down to be more generally useful.

# Pipelines
Overall the pipelines make good sense, but I'm questioning the value of making the more complex portions into stages at all. With a lot of these new stages, I just don't see them being useful in a wide variety of use cases. They seem pretty specific to this loose part feeding algorithm. I could be wrong there, maybe you can convince me otherwise.

I'd like you to consider maybe moving a lot of the basic functionality into single stages, something like CreateLoosePartTemplate and FindLoosePart. I think that in the end, these stages would be a lot less complex because they wouldn't be trying to handle a variety of inputs. 

My concern here is that these pipelines are just too complex for a normal user to really understand or edit without messing it all up. I'd rather see a couple stages that just have a few intuitive settings that can be adjusted, rather than many stages that require a lot of logic to understand the flow of.

I'm offering this up for discussion. I'm not saying I won't accept it as it is now, but I think that it's worth thinking about the user experience it creates.

Finally, I want to reiterate that this is not a dictation, it's intended to be the start of a discussion. My goal is to keep the code complexity low, and the user experience high. I'm open to getting to that point from various directions.

robermeyer, 2017-06-14:
Might I suggest a default simple pipeline with an advanced mode to exploit a larger pipeline and complexities.

vonnieda, 2017-06-14:
@robermeyer I think that would end up duplicating a lot of code. Something I've learned about this project after working on it for so long is the vast majority of people just want it to work. I think we need to get this feature to where it pretty much just works, and allow for adjustments in the basics, like background color, light threshold, etc.

dzach, 2017-06-14:
@vonnieda thank you for looking into this work.

I don't really have any major objection to any of your suggestions. I tried to follow the general principles that apply to this project, but my end choices may have be influenced by my own convenience. I couldn't agree more with your comment above for stages that `just work`, from a user's point of view.  I'll try to elaborate on issues one by one, clarifying my opinion, but I can only do that in so many words.

## Default ReferenceLoosePartFeeder pipeline

One major concern in previous versions of the MatchPartTemplate stage, one that was left as a question in the code, was about what to return when there is no match. 
1. Return null and reject the part the usual way.
2. Return null and warn the user about the problem.
3. Return a rect (or rects) and warn the user, and prompt for his entry, maybe showing current solution and asking for confirmation etc.

At present 1) is the action taken, so the user has to intervene and correct things manually. However, now that the decision is for this to be the default pipeline, it should be able to take care of both polarized and non-polarized parts.

There are three ways that I can think about this issue:
1. Have two different pipelines for the RLPF, one for polarized parts and a second non polarized ones. Both will return regular RotatedRects. In fact, the non polarized one could comprise the first part of the polarized one, ending at the detection of MinAreaRectContours. This is in fact what is done in the proposed pipeline. The rest comprise the stages that decide about the orientation.
2. Have a kind of a signal in the returned model(s), that will enable proper interpretation of the result by OpenPnP. We could adopt the convention for representing non-oriented RotatedRects with a negative `rect.angle` (which I think is what happens now) and oriented RotatedRects with a positive angle (which is what MatchPartTemplate does now, but I'll have to check). If that's the case, it should be really simple to have the same pipeline for both cases. But still, it's the user that should decide what the mode is. So, there may be a need for one more checkbutton saying `polarized part` or something similar.
3. Have OpenPnP look for results in fixed named stages, e.g. `results` for non oriented RotatedRects or `oriented` for oriented ones. What to look for would again be a user's decision.

This is a major issue and needs a clear decision about how to proceed.

# UI and Feeder Changes
Yes, please go ahead and do as you feel appropriate. Indeed, the time I spend for just adding a button is very disproportional to the result in the end, also considering my need for help by others.

# Stage Changes
Processing time has been a concern with all new stages. Many decisions about them have been based on that concern, to the extend I understand what's going on.

The limitation of having a "soup" of stages all in one big list effectively limits the number and functionality of stages, and seem to favor simpler stages against more "user friendly" ones, at least in my view. Grouping stages according to Categories would allow for more stages, some more specialized than others, and would make working with the pipeline a lot easier. The anticipation of Category grouping working at some point in time has also influenced the decisions I've made here. I imagine these stages in a separate category for specialized tasks, that `just work`.

## CreateTemplateImage
* I've used `CreatePartImage`, `CreatePartTemplate` and other names locally trying to see what describes the functionality better. I ended up using the word "template" instead of "part" because of the possibility to not only use a parts image to create a template, but also use a package's shape to the same effect. In other words, use the package shape, create an b&w image of its filled shape and store that as a "template". Then, when using the LoosePartFeeder, create a b&w image from the selected part, retrieve the template of the part's package, and finally match the two b&w template images and get the orientation of _the package_. Any part could be oriented without the need for a specific part image, as long as a shape has been defined for it's package, meaning package body and pads. I believe something similar is done with optical homing. In such a case, "template" sounds  more suitable a word than "part". But I don't mind changing the name to anything sounding better.
* The reason I decided on a single stage doing both rotate and crop, is this comment I put in the code:
```
      /*
        warpAffine can also crop the image, after rotating it, to bbox.size(), 
        so there is no need for a following crop stage
      */
```
In all cases of rotating a model's pixels, a crop operation is necessary to isolate the section of the image that's going to be rotated from the rest of the image. In other words, not doing a rotate+crop requires more work+time than just doing both in one go. We can create a separate `CropModel` stage, if such is necessary at some point, with minimal code. CreateTemplateImage, from a user's point of view, `just works`.

## ReadTemplateImage / WriteTemplateImage
For their naming, see my view above. The idea of having them work off the configuration directory is great. Because this involved the automatic testing process when building OpenPnP, I didn't dare touching it, since I don't know the implications of changing things there. Having an optional parameter fo [... cut ...]
Ah, this one is my favorite! Brings me peace of mind, as an end user. Define nothing and you have a `nop` stage, both image and model pass through (using `getWorkingModel()`). Define one of image or model and you have either `ImageRecall` or `ModelRecall`, the other passes through. Define both and you have what you want in one piece and forget about anything else. The name says it all. No need to think about which one should come first, `ImageRecall` or `ModelRecall`. Code duplication might actually result to about 3 lines of code more. But, yes, no problem, we can have a `ModelRecall` instead. I'll just keep `ComposeResult` for local use.

More to come.

vonnieda, 2017-06-15:
Hi @dzach, going to respond partially for now, and more later. For the time being, please don't jump into making a bunch of changes until we've come to some conclusions. Obviously don't let me stop you from doing whatever you want to do, but please don't start implementing changes I've suggested until we both agree.

# Default Pipeline

For the first question, regarding what to return, I think we should keep this simple for now and consider expanding later once we see what works and what doesn't. To that end, I say let the user specify a quality of match and if nothing hits that quality then return null. This will get interpreted in the feeder as "No matching parts found" and the user will be alerted to that.

I'm not sure I see a need to differentiate between polarized and non-polarized. There's no harm in running the full pipeline, aside from taking a bit longer, is there? I'd say let's use the polarized pipeline as the default and if people feel strongly about wanting to save the extra milliseconds they can delete the extra stages. We'll include information in the documentation about how to do this.

# UI Changes

Sounds good. I will try to get these changes in place in the next few nights. My plan here will be to integrate everything you've written so far into the `test` branch, which is distributed on the website automatically at http://openpnp.org/test-downloads/. Then we can iterate on it till we're happy with it and in the meantime users can try it out without having to build it themselves.

# CreateTemplateImage

I'm okay with leaving rotate and crop in there, but I do feel strongly about the name. To me CreateTemplateImage doesn't tell me anything about how I would use this stage. Yes, I know it's going to create a template image, but I don't know what data it needs to do that, or what the purpose of said template will be. Put another way, the name is too generic for how specific the functionality is.

How about something like IsolateRotatedRect? I feel like that describes the functionality very well. Or even IsolateModel if you intend to eventually support other models?

# ReadTemplateImage, WriteTemplateImage

Agreed on the user specified directory. If blank, use the Configuration API, if specified, store in that directory. 

For the names, I feel that these must include Part since they specifically base their names off part IDs. 

To be honest, this really highlights a limitation of the CvPipeline that I had hoped to address eventually. I believe we need very simple variables and/or expressions in stages. I don't think it makes sense to keep adding getPart(), getNozzle(), getFeeder(), etc. to the pipeline each time we need to expose some piece of data from the caller. Instead, I think it be better if we just dumped in every value that might be useful and then allow stages to have strings such as `/tmp/{part.id}.png` in them. This is how RoboRealm works, which is one of the systems I used as inspiration for CvPipeline.

Until that happens, though, we have to be careful about names. I'd appreciate if you would change these to ReadPartTemplateImage and WritePartTemplateImage for now, and down the road we'll see if these can be refactored to be more generic. 

# ComposeResult

Okay, I'm convinced. Leave it as it is.

# Summary

So, in summary, here are my suggested changes against the code as it stands now:
1. Return null from the pipeline if no matching parts are found.
2. I will add the UI code and release on the `test` branch.
3. Change CreateTemplateImage to IsolateRotatedRect, or something similar.
4. Change ReadTemplateImage to ReadPartTemplateImage and use Configuration.getResourceFile() by default, and allow override with the user specified directory.
5. Change WriteTemplateImage to WritePartTemplateImage and use Configuration.createResourceFile() by default, and allow override with the user specified directory. 

What do you think?

Jason

dzach, 2017-06-15:
@vonnieda  I should have started replying from the last section of your comment. It may give answers to all other parts.
# Pipelines
The logic of the pipelines is indeed quite inelastic. "Playing" with this pipeline would only work for image processing, like changing MaskHsv color ranges, color conversions, decisions of what kind of a template image to save etc. I've tried to see how I could make the stages either as simple or as comprehensive as possible. The logic steps for the match template pipeline are:
1. Process the input image and produce RotatedRects of all parts present. This is done by the user using stock stages.
2. Read in the template image, after finding what file name to use, and create a rect from it to be used as the filter for the input RotatedRects. This is a new stage. 
3. Process the template image to make it compatible with the input image for template matching. This is done using stock stages and should be adapted to user's system.
4. Find closest model. Filtering comes as a bonus. Needs template size for the filter and a list of models to chose from. This is a new stage.
5. Process a fresh input image using the closest model as a mask to isolate it from the rest of the image. Make it compatible with the template image for template matching.
6. Match template quadrant and decide about the winning one. Calculate orientation. This is a new stage
7. Display results for user. This is a modified stock stage that draws the orientation of the resulting rect.

I cannot really see how these steps, which alternate between stock and new stages, can be changed.  All odd numbered steps are image processing steps, that are particular to each vision system. We cannot decide on behalf of the users what kind of image processing would be appropriate for their vision systems and what parameter values would be optimal. We cannot, for example, join steps 2 and 4 because we need to image process the template in between. We cannot join 4 and 6 because we need to image process and isolate the closest model in between. We could eliminate the freedom to decide what type of a template image to save (color, gray, normalized, equalized, etc) and therefore decide on the kind of fixed processing to use when reading the template, but that would severely limit the adaptability of the pipeline to varying user conditions. E.g. I initially saved processed gray template images only to find out later that I had lost the possibility of using the red channel of a color image to get better feature recognition for a specific part. Now I only store color images.

What do you think?

dzach, 2017-06-15:
Oops, took a long time to write this and you wrote another one, ending in the same phrase :-) Will read it and reply tomorrow.

dzach, 2017-06-15:
@vonnieda 

In general:

I don't believe there is a disagreement, but here are a few points I'd just like to stress:

I think of all those "complex" stages I tend to write the way an object oriented approach would call for: Have one `toString()` function for all kinds of objects; have one draw() function for all kinds of shapes; have one `closest` stage for all kinds of models; have one `mask` stage for all kinds of models; Have one overloaded `DrawModel` stage replace 6 different simple draw stages, and one `MaskModel` stage replace 4-5 `mask` ones. Free the user from the details that are not important to his goal. He doesn't really want to learn OpenCV, he just wants to place parts.

It all comes back to the same basic idea: I see the pipeline editor more as part of the application than as an assembly language for OpenCV primitives. The pipeline editor is a good application tool, but too limited to play the role of an assembly editor for OpenCV, as it stands today. For users, the editor is the application interface offering CV functionality for pick&place parts. That's why I don't see the "orthogonality" of pipeline stages as the governing principle for all stages. For instance, I see no point in having everything pass through the `ImageRead` stage when all this stage does is call the same basic functions as any other stage that needs to read an image directly. Slogan: Free the user from the burden.

To the specifics:

# Default pipeline
We need to differentiate between polarized and non-polarized parts because we use a template to match parts. If there is no template for a non-polarized part, the pipeline will fail to deliver a model before it reaches its final stages. In other words, for this pipeline to work, there needs to exist a kind of template image for every part, even for non-polarized ones. That's a reason to also use a part's package to derive a template, when possible. For polarized parts, having a template should be expected. Should we enforce the same for non-polarized parts? That's not necessary, and it's not how it works today, but it can be done if it helps. 

# Summary
1. Depends on how we solve the problem with polarized vs. non-polarized parts in the default pipeline.
2. Thanks!
3. How about `CropOrientRect`? Crop is the operation that has a major effect on the image size and characterizes this stage. `Crop`, and `Orient` as a verb, signify what we do with this stage. `Rotated` could be omitted to make the name shorter, since we actually crop to a bbox Rect and not a RoatatedRect. 
4. I've tested the idea of using a black and white template image of a part/package to filter and orient parts. Even without any change to the match pipeline settings, just reading in a b&w template image, produces these scores:
 ```
    MatchPartTemplate INFO: rotation1 score = 0.28731775283813477
    MatchPartTemplate INFO: rotation2 score = 0.38409703969955444
    MatchPartTemplate INFO: rotation3 score = 0.5455046892166138
    MatchPartTemplate INFO: rotation4 score = 0.3428635597229004
    MatchPartTemplate INFO: winning rotation = 3
``` 
Using a thresholded b&w input image and the b&w template image yields:
```
    MatchPartTemplate INFO: rotation1 score = 0.6351249814033508
    MatchPartTemplate INFO: rotation2 score = 0.6849485039710999
    MatchPartTemplate INFO: rotation3 score = 0.9003778696060181
    MatchPartTemplate INFO: rotation4 score = 0.6889756917953491
    MatchPartTemplate INFO: winning rotation = 3
```
Surprisingly, as it stands today the pipeline yields a lower score:
```
    MatchPartTemplate INFO: rotation1 score = 0.319961279630661
    MatchPartTemplate INFO: rotation2 score = 0.0
    MatchPartTemplate INFO: rotation3 score = 0.6488186717033386
    MatchPartTemplate INFO: rotation4 score = 0.0
    MatchPartTemplate INFO: winning rotation = 3
```
It seems we can do filtering and orientation of a part without much of user intervention and with much fewer templates, at least for asymmetrical packages. Non-polarized parts can get the benefit of automatic filtering from the size of their package shape. That is the reason I'd like this stage to be as generic as possible: the user wants to read a "template" and match a part with it, we'll do all the dirty work for him and see what template we can come out with: if a part template image exists , we'll use that (from part.id); if a package template image exists, we'll use that (from package.id); if a package shape has been defined (more on this elsewhere) then we'll produce a b&w filled shape on the fly out of that (from package.id) and use it as a template. So I'd rather rename  `ReadTemplateImage` to `ReadPartTemplate` or something similarly generic (`ReadPartDescriptors`, `ReadPartCharacteristics`?) instead of making its name too specific to today's functionality. We can think of it as an overloaded OO method. This stage could  be amended later on with anything needed to support different matching methods, e.g. SIFT, etc. The user will only need to know the name of one stage to read in some data for a part and do a `MatchPart` operation. These stages will belong to a category of stages that `just work`.

Will use Configuration.getResourceFile() and create a directory `templates` if it does not already exist.

5. This is more specialized than 4. I'd like to have a boolean parameter `usePackageName` or `asPackage` to indicate how to derive the source of the name and have the stage named `WritePartTemplateImage`. Or think a more generic name and rename it `WritePartTemplate` or `WritePartDescriptors` (can't think of something good). The need to have a generic name is not as strong as when reading in a template, because the method that produces them will be very different and can be chosen with a menu on the relevant panel, when creating the template.

dzach, 2017-06-15:
I don't want to be misunderstood:

The limitation of the CvPipelineEditor comes from the fact that it works as a pipeline, with limited control structures, no variables etc. It is application oriented, and its purpose is make it easy to the user to do CV, not just present each and every parameter existing for opencv primitives.

It's usefulness though seems to be inversely proportional to its "limitations": It delivers solid results with a limited number of pipeline stages.

Needless to say, it's the only OpenCV tool I personally use up to now.

dzach, 2017-06-15:
Pushed a modified `DrawRotatedRects` stage that draws an orientation line.

dzach, 2017-06-17:
Pushed added ability to use package ID for automatic name creation for a template.

https://github.com/dzach/openpnp/tree/develop-loosePartPipeline

Have given some more though to the default pipeline issue. I think it would be acceptable to have a template created for each and every part that is to be fed from a LoosePartFeeder. The time needed to produce a template image is minimal. For passive parts, it could be made event easier by using body dimensions from the shape of the package, so no creation of template image would be required, although color information from an image may add to template matching performance. I haven't tried this yet, but doesn't seem to be very hard to implement.

dzach, 2017-06-18:
Added ability to create a template for parts from package body, when no image template, either for part or for package, can be found. Useful for non-polarized rectangular parts, like SMT resistors and capacitors. For this to work, the width and height of the body need to have been set in Packages->Footprint. This usually comes from the name of such package e.g. 0602, 0806

@vonnieda 
It looks like package footprint pads are not good candidates for representing package pins. I see there is an Outline class and an outline field in the Package class, but I'm not sure what is its current role in OpenPnP; it returns a null shape when I call it from the pipeline. Could that be potentially used to hold a proper part outline? I'm not sure it's worth the effort though since creating an image template is so much easier.

dzach, 2017-06-20:
Added default file paths for writing and reading templates:
```
__configurationDirectory
  |__ templates
     |__ LM1117.png
     |__ SOT223.png
     |__ new
        |__ test-part.png
        |__ test-pkg.png
```
New templates are created by default inside the `templates/new` directory. Each time the pipeline is run, the file is overwritten, so it's best if this doesn't happen with already saved template images, hence the reason for the `new` directory. When the user decides an image is good he moves it to the parent directory, i.e. to `templates`. This can be overridden if the user sets the directory manually.

This is how the whole thing has worked best for me. I believe that making the user aware of these details helps avoid problems, and makes it easier to manage part templates, if a need arises. 

Code is here: https://github.com/dzach/openpnp/tree/develop-loosePartPipeline

### Regarding the uniqueness of directories:

Naming the default directory `templates`, with a low case 't', avoids conflicts with directories created by createResourceFile(Class forClass, ...), since Class names start with a capital letter, e.g. createResourceFile(Templates.getName(), ...) would create a `Templates` one, in the worst case. Even then, control of the source code for this project is tight so problems will be caught before they actually arise. The same `templates` directory can be used for template files (keypoints, descriptors, etc) created for other feature matching methods, e.g. SIFT, SURF. If needed, future image file names can be named in a way to avoid possible conflicts. 

@vonnieda , thank you for providing in depth guidance and many useful suggestions: 

I believe this concludes my part regarding the development of the LoosePartFeederPipeline by using the MatchTemplate method. The rest, e.g. naming stages, separating reusable code, etc. are really decisions that can be safely made by you, in accordance with the principles set forth for the whole project. Please feel free to change all points, names, etc, that you think will serve these principles better. I can submit a Pull Request, if necessary.

I'm looking forward to seeing this code make it into the test branch and have users try it and provide feedback. I will provide a tutorial on how to use this new feature, when things are finalized.

vonnieda, 2017-06-20:
Sounds good, thanks @dzach. I worked on the UI side of it last night and have that pretty much ready to go. Unfortunately the rest of my week is very busy so it won't be until next week that I get to work on this, but I'm going to plan on spending all of next week getting it integrated so hopefully by the end of next week we'll be able to do some testing.

Thank you again for all of your hard work and diligence on this feature!

dzach, 2017-06-20:
Thank you @vonnieda, it's been a pleasure getting this done!
Will post some news in the mailing list and a new video showing speed and orientation graphics, in case someone wants to test it ahead of time.

vonnieda, 2017-06-23:
Hi @dzach, I've done some work integrating your code. You can see the results here if you'd like: https://github.com/openpnp/openpnp/tree/dzach/develop-loosePartPipeline

So far the major work is:
* Moved the pipeline XMLs into the right place so they can be loaded automatically by the feeder.
* Updated ReferenceLoosePartFeeder to use the new pipelines.
* Added buttons to edit and reset training pipeline to ReferenceLoosePartFeeder.
* Reformatting and cleanup.
* Removed duplicated code.

I've still more to do before this can be released, I'll keep you updated.

dzach, 2017-06-23:
Thanks @vonnieda ! The buttons look nice and tidy.

vonnieda, 2017-06-28:
Hi @dzach, I am getting ready to merge this code but I need to check with you on licensing. In some of the files you have put:
```
/**
 * code written by dzach, @ https://github.com/dzach
 */
```
At the top of the file. For me to be able to merge the code the files need to include the OpenPnP license like what you can see at the top of, for instance: https://github.com/openpnp/openpnp/blob/develop/src/main/java/org/openpnp/Main.java

You can see more information about this at https://github.com/openpnp/openpnp/wiki/Developers-Guide#licensing-and-copyright. 

So, is it okay if I add that license to the top of the new files? I'd be happy to assign copyright to you in those files as mentioned in the above link.

Thanks,
Jason

dzach, 2017-06-28:
Yes, of course, go ahead and include the standard copyright and license info as it appears in `Main.java`. I added that note more as a reminder of whom to blame for a possibly bad implementation of the match template method.

vonnieda, 2017-06-28:
Thanks @dzach, I will plan to get this released tonight. One more question: I've been working on the assumption that this should replace the default code and pipelines for the ReferenceLoosePartFeeder. Do you feel it's in good enough shape to do that, or would you prefer it releases in more of a "beta" first? For instance, I could make a new feeder called AdvancedLoosePartFeeder and have it live there while leaving the stock one alone. I'm happy to do either way. Want to get your take on it.

dzach, 2017-06-28:
Great news @vonnieda , thanks a lot!

I'm very confident that this works fine and can replace the existing pipeline, but not without some changes to the way the stock pipeline works: this new one doesn't handle the non-polarized parts in the same way, i.e. it needs at least the body dimension for the part package to work like the stock one does, and a rather calibrated camera. People using the stock one will have to go a bit out of their way and do some extra steps to get the old functionality back.

So I'd rather go with the safer option and add the AdvancedLoosePartFeeder. This nice name sounds promotive enough to get people to try and used it.

One other concern I have is about the default values used in the stages. I've optimized these values to work with my colored background and test parts, but they may not be optimal for other setups and may need to be set to values known to be good averages for most users.

vonnieda, 2017-06-29:
Hi @dzach, this code is finally merged! I made note of the changes here:
https://github.com/openpnp/openpnp/blob/develop/CHANGES.md#2017-06-28

And here is the commit:
https://github.com/openpnp/openpnp/commit/69013f3ce5549bbbdc66d56bb5bde367489b6efb

I'll make a new thread on the mailing list to announce it and give people a place to start posting feedback. I'd appreciate it if you would create a Wiki page explaining how to use it. I've created a placeholder for it here:
https://github.com/openpnp/openpnp/wiki/AdvancedLoosePartFeeder

Thanks for all your hard work on this!

dzach, 2017-06-29:
Thank you for your guidance and patience @vonnieda !
I will post "how to use" instructions to the new wiki page soon.

[1 more reply]

closed 2017-06-30 by vonnieda