openpnp/openpnp #575: Final Rotation Alignment Ignored Fiducials

vonnieda, 2017-05-29:
See discussion in https://groups.google.com/d/msgid/openpnp/CAJGcfUhOtApPf5OkwpufzX4thGyO1MjCtFzb9QjeSr5q_v-FwA%40mail.gmail.com.

The new final rotation alignment code added in https://github.com/openpnp/openpnp/pull/571 is not taking board fiducials into account. The fiducial updates are handled in ReferencePnpJobProcessor:657. This should be moved into the planning stage and the offsets applied there, so that any code past the planning stage has access to the correct board location.

cri-s, 2017-05-29:
https://github.com/openpnp/openpnp/compare/develop...cri-s:patch-1
This patch could fix the fiducial bug.
As local fiducials are not working, maybe this sort of fix seems reasonable..

vonnieda, 2017-05-29:
That code does not compile.

commit de7e361e (2017-05-29): Fixed #575, pre-rotation now includes fiducial compensation. Refactored the fid compensation code into it's own function and applied it in both places it's used.
commit 0713b6c4 (2017-07-02): Merge branch 'develop' into feature/fid-pipeline

* develop: (105 commits)
  Temporarily disable the Bottom Vision pre-rotate feature until it's working correctly. See https://github.com/openpnp/openpnp/issues/401.
  Added the ability to set speeds for each nozzle tip changer movement. Defaults to 1.0, producing the same behavior as before.
  Changed label of System output checkbox fixing capitalization and removing superfluous "enable" word. Refactored code a bit to make it compatible with WindowBuilder.
  Changes suggested by @vonnieda to SystemLogger simplifying the print mechanics
  remove [... cut]
closed 2017-05-29 by vonnieda