openpnp/openpnp #580: Add 'tip diameter' to nozzle tip wizard

dzach, 2017-06-08:
# Feature Request
## Describe the Feature
Add an entry for setting the tip diameter for each nozzle and have it available to the pipeline and the rest of the system. It can be used in many places to represent the size of the tip nozzle, e.g. as a recticle, as a center mark in the pipeline image, as the COG in models, e.g. rotated rects, helping diagnose incompatibility of nozzle tip with part size:

[image]

## How Will It Look?
If the feature will include changes to the user interface, how would they look? Include drawings or mockups if possible.

[image]

## Why Do We Need It?
1. What problem does it solve?
2. Is it useful for everyone, or does it only solve a problem for your machine?

vonnieda, 2017-06-08:
Great idea!

dzach, 2017-06-08:
Ok, I've got my `infinite monkey coder (TM)`to go as far as getting the label and the text box to appear in the wizard. 

[image]

There is a chance I have also succeeded to properly include the new property `tipDiameter` to the `AbstractNozzleTip` class.

How do I get the value of the tip diameter, say from a pipeline script stage? `pipeline.getNozzle()` returns just `null`. Is this correct?

I can reach my `tipDiameter` with this:
```
pipeline.getCamera().getHead().getDefaultNozzle().getNozzleTip().getTipDiameter()
```
but this doesn't feel right. Or does it?

vonnieda, 2017-06-08:
Nozzle has to be set on the pipeline before it's processed. So, in whatever class you want to use this, you need to call pipeline.setNozzle(the nozzle) before calling pipeline.process().

dzach, 2017-06-08:
Yes, now that I think of it, it is logical. In my example with the pipeline: a pipeline does not have a direct connection to a nozzle, but it does have one with the camera that is grabbing the image it is processing. So, I guess, it does feel right to get the tipDiameter through the camera, after all.

Should I submit this little addition of the nozzle tip diameter with a PR?

vonnieda, 2017-06-08:
Sorry if I wasn't clear. CvPipeline does have a nozzle property (getNozzle, setNozzle) but the nozzle has to be set in whatever is calling the pipeline. For instance, in LoosePartFeeder, this happens around line 83. If the thing that is calling the pipeline doesn't set the nozzle, then the pipeline will not have access to a nozzle. It is not a good idea to use getDefaultNozzle since that will likely be wrong on a system with multiple nozzles.

For example, let's say your system has 2 nozzles. The system goes to pick the first nozzle, and you call getDefaultNozzle(). That works fine. But then for the second nozzle you call getDefaultNozzle and you end up with the data for the first nozzle.

So, in short, if you need access to the nozzle in the pipeline, you should use pipeline.getNozzle(), and for any pipeline that uses that it must call setNozzle beforehand.

vonnieda, 2017-06-08:
Oh, and yes, please submit a PR for the nozzle tip diameter change. It will need to be a Length value, and not a double, though. So make sure that's the case first. A double has no units associated with it, a Length does.

dzach, 2017-06-08:
> pipeline.getNozzle() returns just null. Is this correct?

I've already checked that but it returns `null`, while the pipeline is called by `Edit pipeline` of the LoosePartFeeder wizard.

vonnieda, 2017-06-08:
Yea, looks like those properties are not being set properly in the wizard.
They were added to the feeder, but not the wizard. I'd suggest adding

        pipeline
.setNozzle(MainFrame.get().getMachineControls().getSelectedNozzle());

at line 96 in ReferenceLoosePartFeederConfigurationWizard.

Jason

dzach, 2017-06-08:
I almost have it, but I don't want to fiddle with the tests:
```
testSampleJob(SampleJobTest): Error while saving machine.xml (Transform of class org.openpnp.model.Length not supported)
```

dzach, 2017-06-08:
It comes from here:
``` 
   public AbstractNozzleTip() {
        this.id = Configuration.createId("TIP");
        this.name = getClass().getSimpleName();
        this.tipDiameter = new Length(1.0, Configuration.get().getSystemUnits());
    }
```

vonnieda, 2017-06-08:
Can you push the code to a branch so I can look at it? That looks like it
should work fine.

vonnieda, 2017-06-08:
Oh, make sure you have it set as an @Element and not an @Attribute.

Jason

dzach, 2017-06-08:
> Oh, make sure you have it set as an `@Element`

That did it.

https://github.com/dzach/openpnp/tree/feature-NozzleTipDiameter

closed 2017-06-08 by dzach