openpnp/openpnp #583: Per part/package/feeder vacuum

pk7639, 2017-06-09:
# Feature Request
## Describe the Feature

The ability to change the vacuum sense level required for a pick on a per part basis

## How Will It Look?

It'd be great to see a 'disable vac check' option against a part. So the machine would just blindly pick it. The vision system will trip if it's not on the nozzle. This could just as easily be against a package or feeder. 

## Why Do We Need It?
1. What problem does it solve?

When picking parts with packages like Mini-MELF you never get the same vacuum levels that you get with a flat surfaced part. At present, the job stops.

2. Is it useful for everyone, or does it only solve a problem for your machine?

Everyone will have this problem

## Provide Examples
Is this something that exists in other pick and place software? Provide examples and references if possible.

Yes, our other machine has the option of setting the vac sense level for each feeder, and 'disable' is an option.

vonnieda, 2017-06-09:
@pk7639 Do you use the same nozzle tip for the Mini-MELF as you do other parts? In other words, could you just apply a different setting at the nozzle tip? Or does it really need to get into individual parts?

pk7639, 2017-06-09:
We have three nozzles on the head, and we use the smallest for the irregular packages. a much smaller nozzle tip would give higher vaccum, but we already see some inertia induced rotation of the part (it has a glass body) The round body means that it doesn't really matter anyway, we'll always see a lower vacuum level on it than we will with a flat part. 
It's not just MELF's either. We have a temperature/humidity sensor and a small smd header which have uneven surfaces that produce the same effect.

I guess we could have a phantom nozzle with a low or zero vacuum threshold and do a stubbed out tip change (ie no mechanical change) for that part. But it doesn't seem as neat as a the per part, package, or feeder vacuum sense level override option??

vonnieda, 2017-06-09:
Yep, agreed. Just wanted to suss out the possibilities. I think it will be a good addition.