openpnp/openpnp #584: Loading nozzle speed and unloading nozzle speed behaves different

lilltroll77, 2017-06-18:
I think the speed needs a fix for load unload nozzle under Machine setup.

If
 l_n = location for position n
 v_n = speed for ???
v_1 is during loading between l_1 -> l_2 ?
but during unloading l_2 -> l_1 gets speed v_2 !?

That need to be shifted to v_1, and so on for v_3...v_4, otherwise loading and unloading behaves different. 

Maybe it instead should be 3 or 5 Speed textboxes located between the location boxes, since the speed is between 2 locations.

# Read This First
To submit an issue for OpenPnP you must use this template or it will be deleted. Please fill out either the Problem Report or Feature Request section below and delete the other section.

-----------------------------------------------------------------------

# Problem Report
## Describe the Issue
Please describe the issue in as much detail as possible. Without a good description it's impossible for developers to understand the problem.

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP
2. Click the Power On button.
3. Jog the machine 10mm to the right.
4. The machine fails to jog and shows an error.

## Expected Result
What did you expect to happen? Be descriptive.

## Actual Result
What actually happened? Be descriptive.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
The more information you can provide about the issue the more likely it is that a developer will be able to respond.

Whenever possible, please use the Menu -> Help -> Submit Diagnostics feature and include the URL here. Submit Diagnostics makes it easy to include log files, screen capture, debug images, configuration files, etc.

-----------------------------------------------------------------------

# Feature Request
## Describe the Feature
What do you want it to do? Be as descriptive as possible.

## How Will It Look?
If the feature will include changes to the user interface, how would they look? Include drawings or mockups if possible.

## Why Do We Need It?
1. What problem does it solve?
2. Is it useful for everyone, or does it only solve a problem for your machine?

## Provide Examples
Is this something that exists in other pick and place software? Provide examples and references if possible.

vonnieda, 2017-06-18:
I like the idea of making it clear that the speed is to be used between specific locations.

lilltroll77, 2017-06-29:
I started to play with the java code.
I'm not new to GUI programming, eclipse or git - but I'm completely new to java and swing.
What about this basic idea (testing different label-positions) - or should I change the Formlayout to a non-grid variant. In such case, any proposals which one you prefer to make the graphics scale correctly.  
[image]
I have also added ToolTipText.

vonnieda, 2017-06-29:
I think the first variant with the label to the left looks good. I don't
think there is any reason to change the layout.

If you are new to Swing development in OpenPnP, make sure you are using
WindowBuilder: https://eclipse.org/windowbuilder/
This is how all of the UIs in OpenPnP are created and it is much easier
than trying to edit layout constraints by hand.

Jason

lilltroll77, 2017-06-29:
I suspected that something like WindowBuilder was involved.
I will check that out and try to fix the rest of the code as well.

lilltroll77, 2017-06-29:
I have working code now, based on WindowBuilder including the speed calculations.
It loads and unloads the nozzle without spitting them out on the table due to incorrect speeds.
[image]
But I do not have a clue how to make the code backward compatible with the (old) machine.xml

With my current naming it looks like this
```
<nozzle-tip class="org.openpnp.machine.reference.ReferenceNozzleTip" id="e092921a-2eef-449b-b340-aa3f40d8d791" name="507" allow-incompatible-packages="false">
                        <compatible-package-ids class="java.util.HashSet"/>
                        <changer-start-location units="Millimeters" x="149.4" y="77.3" z="-13.8" rotation="0.0"/>
                        <changer-start-to-mid-speed>0.01</changer-start-to-mid-speed>
                        <changer-mid-location units="Millimeters" x="149.4" y="77.0" z="-17.6" rotation="0.0"/>
                        <changer-mid-to-mid-2-speed>0.025</changer-mid-to-mid-2-speed>
                        <changer-mid-location-2 units="Millimeters" x="149.4" y="25.0" z="-16.5" rotation="0.0"/>
                        <changer-mid-2-to-end-speed>0.04</changer-mid-2-to-end-speed>
                        <changer-end-location units="Millimeters" x="149.4" y="17.0" z="-16.5" rotation="0.0"/>
                        <calibration angle-increment="15.0" enabled="false">
                           <pipeline>
                     
```
(I haven't read the google naming of variables enough yet.)

But how to handle the old way with 4 speed settings in machine.xml? Can it be controlled with compatibility package settings?

vonnieda, 2017-06-29:
@lilltroll77 First, thank you for running with this and for thinking about the issue of backwards compatibility!

The way I usually handle this kind of thing is to add the new variables, as you've done, and then add a method like this:

```
@Commit
public void commit() {
    // copy old variables into new and set old ones to null or 0
}
```

The commit method is called by the configuration serializer after it loads the config. So what will happen is that the first time a user with an old config loads their config with this new code, it will copy the variables into the new spots and zero out the old ones. Finally, I'll add a // TODO to remove the old stuff in 6 months or so.

You can also change the old variables from double to Double (primitive to object) and set them to null. This will remove them entirely from the old config once it's saved.

As for the mapping from old to new, I'll leave that to you to figure out. I think anything reasonable will be fine and we'll just let users know they should check their settings after installing the update.

lilltroll77, 2017-06-30:
Ok, I did changed from double to Double and added a check and convert for old variables to new ones.
Now it looks like this. It seems to work when I add old variables to my XML.
https://github.com/lilltroll77/openpnp/commit/f09c5c265688c4b4f249089f553c86b5d2538ddb

vonnieda, 2017-06-30:
@lilltroll77 Looks great! Only thing I see missing is failing to null changerStartSpeed, changerMidSpeed and changerStartSpeed2 after copying them into their new home.

lilltroll77, 2017-07-02:
Things went crazy here 2 days ago when I tried to handle git within eclipse and change the repo from openpnp to my fork. All changed files became empty of some reason. I think everything is in sync now on my side. I have made a pull request.

vonnieda, 2017-07-03:
Fixed in #596

closed 2017-07-03 by vonnieda