openpnp/openpnp #589: NullPointerException when adding actuator if simulated Up camera is present

lilltroll77, 2017-06-25:
If you follow the steps at
[https://github.com/openpnp/openpnp/wiki/Setup-and-Calibration%3A-Camera-Lighting]
and have a SimulatedUpCamera under machine setup, openpnp will throw a NullPointerException when you press your (new) button under Machine Controls -> actuators.

Maybe at least update the documentation that you need to remove simulated cameras prior to adding an actuator-control for lightning.

[image]

vonnieda, 2017-06-25:
@lilltroll77 What version did this happen on?

lilltroll77, 2017-06-26:
2017-06-16 version under Win10 x64

vonnieda, 2017-06-27:
Thanks for the report @lilltroll77, this is now fixed.

commit 53e8bbaf (2017-06-27): Fixes #589 by checking for null in SimulatedUpCamera when machine activity happens.
commit 99e978bc (2017-06-28): Merge branch 'develop' into feature/sim-cam-work

* develop:
  Fixes DrawTemplateMatches stage drawing at corner rather than centered.
  Added support to enable Linux to detect and use the /dev/serial/by-id device names to allow consistent port choice.
  * Removed extraneous calls to VisionUtils.getBottomVisionCamera(). * Fixed an error with calculating the speed the nozzle should move at. Part speed is already applied in ReferenceNozzle, the code would have caused it to move at part speed ^ -2. * Added comments about magic numbers. * Fixed minor text errors.
  Fix up formatting.
  Fixes #589  [... cut]
commit 2f354f16 (2017-06-28): Merge branch 'develop' into dzach/develop-loosePartPipeline

* develop:
  Fixes DrawTemplateMatches stage drawing at corner rather than centered.
  Added support to enable Linux to detect and use the /dev/serial/by-id device names to allow consistent port choice.
  * Removed extraneous calls to VisionUtils.getBottomVisionCamera(). * Fixed an error with calculating the speed the nozzle should move at. Part speed is already applied in ReferenceNozzle, the code would have caused it to move at part speed ^ -2. * Added comments about magic numbers. * Fixed minor text errors.
  Fix up formatting.
   [... cut]
commit b62c0ce8 (2017-06-29): Merge branch 'develop' into feature/CvPipeline_Variables

* develop:
  Fixes DrawTemplateMatches stage drawing at corner rather than centered.
  Added support to enable Linux to detect and use the /dev/serial/by-id device names to allow consistent port choice.
  * Removed extraneous calls to VisionUtils.getBottomVisionCamera(). * Fixed an error with calculating the speed the nozzle should move at. Part speed is already applied in ReferenceNozzle, the code would have caused it to move at part speed ^ -2. * Added comments about magic numbers. * Fixed minor text errors.
  Fix up formatting.
  Fix [... cut]
closed 2017-06-27 by vonnieda