openpnp/openpnp #595: Documentation is a little inconsistent regarding  "Getting Started with Eclipse" & "Your First Pull Request"

lilltroll77, 2017-07-02:
The older wiki post [https://github.com/openpnp/openpnp/wiki/Getting-Started-with-Eclipse](url) clones the Git Repository direct from https://github.com/openpnp/openpnp.git.
That procedure  will make it impossible(?) to later push the changes to GitHub without pointing to another clone.

The above mentioned wiki comes after the wiki [https://github.com/openpnp/openpnp/wiki/Your-First-Pull-Request](url) which instead makes a clone from a fork at your account.

**Conclusion:** If you follow the steps here [https://github.com/openpnp/openpnp/wiki/Developers-Guide](url) including installing eclipse you will end with 2 clones on your computer, and the one in eclipse cannot be pushed to Github.

My suggestion is that step 7 in step by step "Getting Started with Eclipse" should be rewritten.

vonnieda, 2017-07-02:
It's quite easy to change the remote that a repo is pointing to, so maybe just adding a footnote that you can change it to your own would suffice.

lilltroll77, 2017-07-02:
Maybe like this:
 !! I'm not native English speaking and has some word blindness , so check my grammar and syntax. !!

7. On the Source Git Repository screen enter https://github.com/openpnp/openpnp.git in the URI field(Note1) or enter the address to your fork at github if you later would like to contribute to the project(Note2) and click Next.

Note1: It is possible to later change so your GIT clone is pointing to your fork at github.
Note2: You cannot directly push your changes to openpnp, it must go through your own fork.

vonnieda, 2017-07-02:
Updated the Wiki as you suggested, thank you!

https://github.com/openpnp/openpnp/wiki/Getting-Started-with-Eclipse

closed 2017-07-02 by vonnieda