openpnp/openpnp #597: CvPipelineEditor often sets stage values to null expectedly

vonnieda, 2017-07-08:
I've noticed, and others have commented, that the pipeline editor will often change a property value to null seemingly randomly. I've noticed this mostly with the ConvertColor stage, for the conversion property and any stage that uses a Color property. 

There are two issues. The first is that this happens at all, which delete user data, and the second is that once it happens it's impossible to recover from. The stage must be deleted and re-added, often losing other settings.

So ideally, try to figure out why it's happening and fix it, and a a fallback, try to make it so that it's recoverable at least.

dzach, 2017-07-09:
Hi @vonnieda 
This happens to me all the time. I think it can be reproduced easily if one tries to change a parameter fast enough while the pipeline is still processing. If that's indeed the case, then disabling the pipeline stages from getting user input while processing might be a way to fix it.

vonnieda, 2017-07-09:
I figured out a way to reproduce this on demand:
1. Add at least two ConvertColor stages to a pipeline.
2. Select the first, select the conversion dropdown and then click away from the dropdown to have it fold back up.
3. Now press space bar, which will highlight the dropdown but not drop it down.
4. Click on the second ConvertColor stage and it's conversion value will get set to null.

vonnieda, 2017-07-09:
Note, the above seem to work with any stage that has a nullable first property.

vonnieda, 2017-07-30:
Fixed in https://github.com/openpnp/openpnp/pull/611

closed 2017-07-30 by vonnieda