openpnp/openpnp #599: BottomVision Pre-Rotate Fails on Shared C Heads

vonnieda, 2017-07-19:
PK reports that the new bottom vision pre-rotate feature fails on their machine which uses a shared motor for the C axes. This is confirmed by other users in the below thread:

https://groups.google.com/d/msgid/openpnp/392748bb-206d-47b8-bea9-351692d78ad6%40googlegroups.com?utm_medium=email&utm_source=footer

cri-s, 2017-07-25:
Marek have confirmed that the fix works for shared Axis. here the diff:
This fix should be applied only if machine have shared C axis

--- openpnp-develop/src/main/java/org/openpnp/machine/reference/ReferencePnpJobProcessor.java
+++ ReferencePnpJobProcessor.java
@@ -628,6 +628,10 @@
                         part,
                         boardLocation,
                         placement.getLocation(), nozzle);
+				// store the actual corrected rotation instead of the offset for shared C axis
+				if (plannedPlacement.alignmentOffsets.getPreRotated()) {
+					plannedPlacement.alignmentOffsets = new PartAlignment.PartAlignmentOffset(plannedPlacement.alignmentOffsets.getLocation().derive(null,null,null,nozzle.getLocation().getRotation()),true);
+				}
                 Logger.debug("Align {} with {}", part, nozzle);
             }
             else
@@ -670,14 +674,15 @@
                         preRotated means during alignment we have already rotated the component
                         - this is useful for say an external rotating stage that the component is
                         placed on, rotated to correct placement angle, and then picked up again.
-                 */
-                if (plannedPlacement.alignmentOffsets.getPreRotated()) {
-                    placementLocation = placementLocation.subtractWithRotation(
-                            plannedPlacement.alignmentOffsets.getLocation());
-                }
-                else {
-                    Location alignmentOffsets = plannedPlacement.alignmentOffsets.getLocation();
-                    // Rotate the point 0,0 using the alignment offsets as a center point by the angle
+			*/
+		    if (plannedPlacement.alignmentOffsets.getPreRotated()) {
+			    placementLocation = placementLocation.subtractWithRotation(plannedPlacement.alignmentOffsets.getLocation());
+				// restore the corrected rotation and override the offset calc for shared C axis
+                             if(Double.isNaN(placementLocation.getRotation()) {
+				    placementLocation = placementLocation.derive(null,null,null,plannedPlacement.alignmentOffsets.getLocation().getRotation());
+                            }
+		    }
+		    else {
+			    Location alignmentOffsets = plannedPlacement.alignmentOffsets.getLocation();
+			    // Rotate the point 0,0 using the alignment offsets as a center point by the angle
                     // that is
                     // the difference between the alignment angle and the calculated global
                     // placement angle.

cri-s, 2017-08-31:
do you have decided in the meantime, if you attribute this as bug to prerotate code or to missing customisation for shared machines ?

vonnieda, 2018-12-06:
Analysis of this bug in https://groups.google.com/d/msg/openpnp/9rb52Ip628c/rz-37M_4CwAJ

markmaker, 2019-03-11:
Hi Jason

had a quick look at that. Some observations:

1. It seems almost trivial to fix this. Just store 'angle' instead of 'Double.NaN'  in the AlignmentOffset. 
2. It will already be applied in the placementLocation.subtractWithRotation() code. 
3. The subtractWithRotation() should probably be done in the Wizard too. 
4. I don't think the second vision shot is needed. One could fix the X, Y offset at the same time as the angle (obviously transforming the X, Y offset by the angle offset)
5. The angle offset will be relatively small and even with considerable nozzle runout this should not significantly alter the offset in X, Y. And we now have proper nozzle runout compensation too. Alternatively the second vision shot could be made subject to the detected angle offset being awfully large. 
6. I don't see the validity of the strange rounding operation in findOffsetsPreRotate(). If this is some sort of rotational backlash compensation, then it should be properly and universally handled as such, like for X and Y. 

I'm irritated by the triviality of 1. Am I missing something? 

For reference, some key code locations:

Points 1, 4-6:
https://github.com/openpnp/openpnp/blob/72d7a9fe827103824b6e315b475da07feaecb58d/src/main/java/org/openpnp/machine/reference/vision/ReferenceBottomVision.java#L124-L131

Point 2:
https://github.com/openpnp/openpnp/blob/0b0698dda410382804b5b615b393b4067db3290a/src/main/java/org/openpnp/machine/reference/ReferencePnpJobProcessor.java#L735-L737

Point 3:
https://github.com/openpnp/openpnp/blob/5131a7483863394f55ce70249cff3a145bfed80e/src/main/java/org/openpnp/machine/reference/vision/wizards/ReferenceBottomVisionPartConfigurationWizard.java#L141-L152

_Mark

vonnieda, 2019-03-11:
Hi @markmaker, thanks for taking a look! 

Your points are all valid, although I'll have to give some thought to the solution. When I look at this last I felt like I needed to set the angle to the inverse to make everything work. It could be that that is due to an additional implementation that is not obvious from the code. See my notes at: https://github.com/openpnp/openpnp/blob/ed9b6d603d9d8a1dfc7f0cb816ef06cac7058ccb/src/main/java/org/openpnp/machine/reference/ReferencePnpJobProcessor.java#L722-L755

A few other thoughts:

1. Basically agree. It seems trivial. I've been working on getting some other fixes in place to make this easier to test. Since I don't have a shared C head to test on, it's hard for me to verify the problem and the potential solution. So, I've been working on unit tests that will be able to test the various types of bottom vision, prove the error, and then fix it.
2. Noted - need to check this when I have more time.
3. Noted.
4. The second vision shot has been discussed quite a bit recently. Will try to dig up the thread later tonight. As you surmised, it corrects for X/Y error introduced by rotation. See next...
5. The error can sometimes be rather large. In theory it should be small - in practice it's sometimes large. We've recently discussed making the second vision shot optional by configuration for users who don't need it and this would certainly apply if you are using nozzle tip calibration. Making it automatic based on the angle offset is a good idea.
6. The rounding was introduced by a PR from Cri S. I never understood why and there was never a good explanation. I did my best to guess at what it was doing and put comments in the code. I'd be perfectly find removing it and seeing if anyone complains. More info at https://github.com/openpnp/openpnp/pull/590#issuecomment-403222572

Fixing this bug is next on my list, so hopefully we'll have some progress in the next couple days.

markmaker, 2019-03-11:
> When I look at this last I felt like I needed to set the angle to the inverse to make everything work. 

Yes, the Job compensation does `subtractWithRotation( )` while the pre-rotate does `placementAngle + angle`. So we need to store `-angle` into the AlignmentOffset to get the same. 

I guess the confusion could be due to the bottom vision being a mirror image of the reality. X and Y are compensated by the camera's Flip Vertical or Flip Horizontal switch but the rotational "[handedness](https://en.wikipedia.org/wiki/Screw_thread#Handedness)" is thereby reversed. So a right-handed angle offset as observed from the bottom camera must be applied as a left-handed angle offset in the proper C axis i.e. it must be negated. 

Like I once added in the Wiki:

> A Bottom Camera image should appear as if you are looking onto a mirror laying on the floor reflecting a view up from the camera towards the nozzle. Again the top of the image should be away from you, the bottom of the image should be towards you. But because left and right should remain left and right (and it is actually a mirror image), you usually need to enable either Flip Vertical or Flip Horizontal depending on how you mounted the camera.

https://github.com/openpnp/openpnp/wiki/Setup-and-Calibration:-General-Camera-Setup#set-rotation-and-transforms

_Mark

markmaker, 2019-03-11:
Another thought:

7. The code currently assumes part outlines must always be in "landscape" orientation (width > height). In my view this is not a valid assumption. The industrie defines a "Component Zero Orientation" based on pin 1, and/or cathode orientation. Pre-rotate will almost certainly be wrong for high pin count SOIC, xSSOP packages, etc.. Instead the code should simply take the offset angle modulo 90° assuming the error is never near that. Note that [modulo preserves the sign](https://en.wikipedia.org/wiki/Modulo_operation#/media/File:Divmod.svg). 

See "Component Zero Orientation":
https://ohm.bu.edu/~pbohn/__Engineering_Reference/pcb_layout/pcbmatrix/Component%20Zero%20Orientations%20for%20CAD%20Libraries.pdf
(I actually wrote an Eagle ulp that does enforces this for existing libraries and boards). 

Code to be fixed:
https://github.com/openpnp/openpnp/blob/72d7a9fe827103824b6e315b475da07feaecb58d/src/main/java/org/openpnp/machine/reference/vision/ReferenceBottomVision.java#L106

_Mark

vonnieda, 2019-03-11:
@markmaker:

Here is the conversation regarding that, from when the PR was originally added:

> > Can you explain why you add 90 degrees when width < height? I understand
> > that this flips the rectangle up so that it's in "portrait" mode, but I
> > don't understand the reason for it.
> 
> No, it don't have to do with portrait mode, it have to do with opencv
> and how it calc
> angle on rotatedRect . It is a tecnical detail and previously the fact
> was uninportant and hidden because of mod 45 operation and how
> component was presented to vision.
> Here sign of rotation matter and so i have choose the -90 to +90 range
> using this formula
> and rotation directions.
> OpenCv don't have knowledge of components and it choose height and
> width arbitrary,
> and return always only one quadrant of rotation. -90 to -0 and if
> rotatedRect is at 0 degree
> the most of time it returns -90 . This line expand it to +-90 degree
> instead of the
> normally used -0 to -180 degree on opencv mailing lists and probably
> opencv books.

I still don't really understand it but since several people vouched for the PR I merged it. This is another piece I'd be willing to change or remove if we can prove it's not needed.

markmaker, 2019-03-12:
Ok, given this information (which is logical, come to think of it):
> OpenCv don't have knowledge of components and it choose height and
width arbitrary and return always only one quadrant of rotation. 

I'd say it should be reverted to said `mod 45°` operation. However it needs to be done with proper sign handling, as 45° .. 90° must first be mapped to -45° .. 0°. 

`angle = (angle >= 45.0 ? angle - 90.0 : angle) % 45.0;`

Note that the previous code is **unstable** for quadratic parts, as `width < height` will flicker. It explains this post:
https://groups.google.com/d/msg/openpnp/vVT7N7H1kfM/r6ndY1mxBwAJ

As to how to "prove" it, I guess the best will be to run tests with "tall" parts such as a large pin count xSSOP. 

[image]

_Mark

markmaker, 2019-03-13:
Hi Jason 

I'm trying to create a PR. 

Looked at 
https://github.com/openpnp/openpnp/blob/72d7a9fe827103824b6e315b475da07feaecb58d/src/main/java/org/openpnp/machine/reference/vision/ReferenceBottomVision.java#L105-L106
again.

To understand, one also needs to look up the `angleNorm()` which does the wraparound i.e. sign aware modulo of the angle nicely. With no `lim` param given, it defaults to the -45° .. +45° range which in my view is the only correct thing to do after vision, as explained above.
https://github.com/openpnp/openpnp/blob/72d7a9fe827103824b6e315b475da07feaecb58d/src/main/java/org/openpnp/machine/reference/vision/ReferenceBottomVision.java#L257-L267

With this knowledge I think the following is true:

1. The `(rect.size.width < rect.size.height) ? 90 + rect.angle : rect.angle` simply does nothing inside `angleNorm()` because `angleNorm()` will simply subtract these 90° again (`clip = 2*lim`). 
2. So what remains is effectively a **doubling** of the angle!

This code will simply over-compensate the rotation error by a factor of two. The part will still be off, just the other way around. I can't imagine how this will align parts properly. For angle errors > 22.5° it becomes erratic, as the doubled angle will even be wrapped around.

Was this really tested?

For a sanity check, see the equivalent `findOffsetsPostRotate()` code:
https://github.com/openpnp/openpnp/blob/72d7a9fe827103824b6e315b475da07feaecb58d/src/main/java/org/openpnp/machine/reference/vision/ReferenceBottomVision.java#L163-L175

Now, I will go on and create a PR to the best of my knowledge, then I will try to test that on my machine with deliberately large pickup angle errors. 

In the meantime you could perhaps think this through as well. 
  ;-)

_Mark

markmaker, 2019-03-13:
Hi Jason

why is this camera re-getting needed?

https://github.com/openpnp/openpnp/blob/72d7a9fe827103824b6e315b475da07feaecb58d/src/main/java/org/openpnp/machine/reference/vision/ReferenceBottomVision.java#L154-L155

_Mark

vonnieda, 2019-03-14:
Hi @markmaker, thanks for looking at this, and for the PR. It will probably be about a week before I can look at it; I'm swamped with work. I will definitely look at it as soon as I can, though.

Regarding the camera re-get. I believe the intention was to allow the selected camera to be changed from a custom pipeline stage. It was added as part of the PR I referenced above, I believe. Personally I think it's hacky and should be removed, or implemented in a most obvious fashion if it's really needed.

markmaker, 2019-03-15:
Hi @vonnieda 

same problem here, swamped with work. No hurry.

Just a remark in the meantime.

I was wrong: Bottom vision being a mirror image (camera Flip X or Flip Y being set) does indeed also negate the angle, so rotation looks the same from both cameras. 

The problem is another one, namely that OpenCV has coordinate system where the Z axis is pointing away from the observer and angles work clockwise as opposed to the OpenPNP camera view, where the Z axis is pointing towards the observer and angles work counter-clock-wise: 
[image]

https://docs.opencv.org/2.4/modules/calib3d/doc/camera_calibration_and_3d_reconstruction.html 

Consequently the angle must be negated for the down camera too, as is evident in the loose part feeder: 
https://github.com/openpnp/openpnp/blob/5131a7483863394f55ce70249cff3a145bfed80e/src/main/java/org/openpnp/machine/reference/feeder/AdvancedLoosePartFeeder.java#L96

If you agree, I propose adding a VisionUtils method for angles. An analog to `VisionUtils.getPixelCenterOffsets()`  that is already used for the X and Y coordinates and where - no surprise! - the Y axis is actually inverted, reflecting the coordinate transformation between OpenCV and OpenPNP. 
https://github.com/openpnp/openpnp/blob/5131a7483863394f55ce70249cff3a145bfed80e/src/main/java/org/openpnp/util/VisionUtils.java#L47-L62

Similarly a `VisionUtils.getPixelAngle()` method could encapsulate the same transformation, simply negating the angle. 

So no more head scratching in the application code. For instance I wondered why the heck the supposed angle need to be **added** to the observed angle to find the (negative) offset. With proper transformation the offset is the intuitive **difference** between observed and supposed angle. 

Found only 4 places where it would be used (2 x Bottom Vision and 2 x Loose Parts), 

_Mark

markmaker, 2021-04-25:
Again, I think it was fixed in #815. Can be closed.

closed 2021-04-25 by vonnieda