openpnp/openpnp #600: Documentation / wiki/Setup-and-Calibration%3A-Driver-Setup

johngrabner, 2017-07-19:
# Problem Report
## Describe the Issue
Instruction to setup a driver shows an image that does not match actual.
Step "Set Serial Port and Baud rate" shows image 
[image]

but actual running program shows
[image]

I am a newbe, so this deviation confuses the learning process.
It appears that the documentation is outdated with the introduction of sub-drivers.  I would offer a fix, but being a newbe, I am more likely to mess things up.

## Steps to Reproduce
1. follow instruction under "Choosing a Driver"
1.1. select GcodDriver
1.2 click apply
1.3 quit the application and relaunch

## Expected Result
image in documentation match current version of code. 
[image]

## Actual Result
See above

johngrabner, 2017-07-19:
To make things easier, here is the clean image that I believe should be used. 
[image]

vonnieda, 2017-07-20:
Looks good, thanks @johngrabner. I've updated the Wiki with the image you provided. If it looks good to you, please close this issue.

closed 2017-08-16 by vonnieda

Duyvinh0307, 2018-10-02:
https://youtu.be/99Tx5tPeeZY
Hepl

vonnieda, 2018-10-02:
Hi @Duyvinh0307 - this is a closed issue not related to the problem you are experiencing. Please post your question to http://groups.google.com/group/openpnp.