openpnp/openpnp #601: Panelization Dialog Fiducial Dropdown Failure

vonnieda, 2017-07-22:
On Windows, at least, you can't drop the global fiducial dropdown in the panelization panel with the mouse, you must use the keyboard. This is likely due to a bad value in the dropdown but I haven't had time to look yet.

See further discussion in https://groups.google.com/d/msgid/openpnp/b6f2e5e7-dfe6-4426-8690-4b37ecee5469%40googlegroups.com?utm_medium=email&utm_source=footer

vonnieda, 2017-07-22:
@BendRocks Tagging you in hopes you will take a look at this :)

vonnieda, 2017-07-22:
This may be related to http://bugs.java.com/bugdatabase/view_bug.do;jsessionid=d238b89a70e0051d5edca359b731?bug_id=4106334

BendRocks, 2017-07-22:
OK, I'm on it. The dropdown not happening was noted in #501. I'll re-visit and report back

vonnieda, 2017-07-22:
@BendRocks If you'll just focus on the dropdown and ignore #602 I'd appreciate it. I have a few hours to blow so I am fixing the stuff in #602 now, and since this one (the dropdown) doesn't happen on Mac it's harder for me to fix.

BendRocks, 2017-07-22:
OK, got it.

vonnieda, 2017-07-22:
Potential fix pushed in https://github.com/openpnp/openpnp/commit/92465bf4e4df7c61201fd299a6b1d1c55699eafd, waiting for user feedback.

BendRocks, 2017-07-22:
Confirmed this fixes the dropdown issue on windows. But why? There are other places with dropdowns that  don't have this issue and don't need this code.

vonnieda, 2017-07-22:
@BendRocks See http://bugs.java.com/bugdatabase/view_bug.do;jsessionid=d238b89a70e0051d5edca359b731?bug_id=4106334. The fix contains two fixes, really. The first makes the dropdown menu a heavy weight component, which means it's a Windows native component. This doesn't have the limitation of not being able to extend outside the dialog box. The second is that it limits the length of the dropdown to 7 items, which further ensures it won't try to extend outside the dialog box. The root problem is that dropdowns can't extend outside dialogs, and since this is the only (or one of the only) place we use a dropdown in a dialog, we only see it here.

BendRocks, 2017-07-22:
Strange. One of the first things I tried was to make the jcombox hold strings instead, and then added a single or two strings which would have fit for sure. But still no drop action. Even if you set the setMaximumRowCount to 2 (which would fit) it still doesn't drop. In any case, the Java x-platform aspects are all new to me so I appreciate the lesson.

vonnieda, 2017-07-22:
Yea, just did some further testing myself and on my box it seems the
lightweight line is the only one that matters. Maybe @cri-s will chime in
and explain the other two lines.

Jason

cri-s, 2017-07-22:
setEditable is to not allow users to edit/add new values that are not
present as part.
As the documentation of java class don't specify the default state
when the class is created,
i have included it.
SetMaximumRowCount is to limit it to 7 lines in order it don't go over
window borders when
more then 7 entry are present.

vonnieda, 2017-07-22:
Thanks @cri-s. JComboBox is non-editable by default, as seen here: https://docs.oracle.com/javase/7/docs/api/javax/swing/JComboBox.html#isEditable()

But I think it's fine to leave that line in just to be sure.

vonnieda, 2017-07-22:
Fix confirmed by everyone involved, looks good. Thanks @cri-s for the fix and @BendRocks for looking into this!

closed 2017-07-22 by vonnieda

cri-s, 2017-07-22:
Usually i don't use that part at all, i have made the fix specifically
for one user (+2 later)  and i have
never tested this code on real machine. I know from that user that the
feature is unusable
as rotation is wrong, it modify board origin, fiducial code don't work
exept for the simplest
case.
From memory, but  probably i could be wrong ,
if the board is rotated, it automatically modify board origin and that
a big issue. making it
not usable for rotated boards.
Several wizards and scripts returned wrong coordinate when using this feature.
Using only panel fiducials don't worked at all.
Even using panel fiducials and pcb fiducials, the machine was not able
to localize pcb fiiducials because looked on wrong location.
There was other issues that i don't remember.
For just step and repeat pcb with no rotation and if pcb itself have
fiducials, it works.

BendRocks, 2017-07-23:
> I know from that user that the feature is unusable as rotation is wrong, it modify board origin, fiducial code don't work exept for the simplest case

Hi @cri-s , are you referring to panelization? If so, #456 has a sample panel to try with all the parameters needed. It should take just a few minutes to confirm it works. I just again with the new fid pipeline and it works as expected. If there are other panels you know of that are causing a problem, then rather than working on them alone in secret why not just open a bug with the bad panel as a jpg (silk + copper + board outline) and we can all try to fix it? Or send the gerber and I'll make the jpg

I have been confused by the new fiducial pipeline sometimes...where it prompts a rectangular pattern instead of circle if you don't have a package fully defined. Maybe that is the root of the issue?

cri-s, 2017-07-23:
Yes, i look to panelization, and i have seen the photos of wrong
placements using it,
the wrong rotation and board zero its assume, the failure when only
panel fiducials
exists and the failure of  panelisation code, and it was before new
fiducial codes was merged, further the coordinate errors returned.
Again, i never have tested it, always remove this code and it was just
for 2 users that
requested it and Marek that wanted test it out and i have it just
included it into the
jar that i was sending him. Further the errors was just 15 minutes
fix, i have not looked
on the renderer code and other things, just fixed the gui errors
knowing that with some
locales and settings this fix don't work as it was only a partial fix
not fixing all gui errors.
SMdude can you tell a bit about rotation errors as when hi have tested
prerotation the
rotation problem on panelisation came up.
The 5 minutes are loose time: If you instead have real pcb's from this
panels, or other panels, then with 15 minutes and real machine, you
can test something.
simulation is really different as real data.

BendRocks, 2017-07-23:
OK, so it sounds like there are two actionable issues above: The first is that 0 degree PCB rotation is assumed. I will file a bug on that. I wish I could just fix it quickly, but I'm slowly learning that anything to do with rotation on OpenPNP takes a while for me to figure out because there are so many things to consider. 

The second is that fiducials are required. This isn't true. You can always enter zeros for the panel fid locations and the untick the "check panel fiducials" checkbox. And note, too, that the code doesn't require explicit panel fiducials. You can easily use two PCB fiducials, you just need to express their location relative to panel origin. And I guess with the new fid recognizer, we can use an pad from an R or C as a fiducial, couldn't we? Again, you just need the coordinates relative to panel 0. I'll note this in the help.

I think those are the two actionable points you've made. One is a bug, the other is by design. 

Note that **I have run real PCBs on this**. The test image you see in the pull request obviously came from a real PCB. And I just re-ran panel alignment on current dev build with PCB eyeballed to 0 degree alignment with the new fiducial pipeline and it worked as expected, reporting ~3 degrees of rotation. 

Finally, if you can send a panel that is causing you a problem and I'll try to fix it.

BendRocks, 2017-07-23:
On second look at this, the code works correctly if you place a board at about 45 degrees and indicate the first board has a rotation of 45 degrees. That 45 degree will propagate to all the panels. And then when you run the panel fiducial recognizer, it will find the fiducials of the panel and update the rotation (which will be 45 +/- some amount). 

So, I don't think there's anything you've written that is actionable thus far. 

I guess if you specify a panel orientation of 45 degrees and then place the panel at 30 degrees it could run into an issue....But that is true in the PCB case too, right? The panel fids work just like the PCB fids.

cri-s, 2017-07-23:
I know that for simple panels  and 0 rotation it works, and there is
probably a misunderstanding.

For fiducials, sample image from internet, the fidiucials have arrows
nerar it in this image.
https://static1.squarespace.com/static/54982a02e4b02e9f5e5d9ca7/t/55ae453fe4b02de83b54c2b4/1437484353478/?format=1000w
There is a single panel with panel fiducials and the pcb don't have
fiducials at all.
Assuming this panel is placeable using you'r code beside the fiducial
issues , actually it's not.
The problem with this is that fiducial alignment is not working for
this panel except for the
first pcb i think/remember.
That was that i mean about single pcb inside panels require fiducials,
otherwise it don't work.

Annother big issue is wizard/script that from nozzle/camera getting
the coordinate back,
that simple not work at all because of several problems, rotation,
missing two levels of
fiducial correction,

cri-s, 2017-07-23:
just try to rotate real board 270 degree, populate without real
component both side, top and bottom. If that works, i'm totally wrong.

2017-07-23 22:34 GMT+02:00, Cri S <phone.cri@gmail.com>:

BendRocks, 2017-07-23:
> just try to rotate real board 270 degree, populate without real component both side, top and bottom. If that works, i'm totally wrong.

Just tried at 270 and -90, both work. And I've spent about 3 hours of a Sunday chasing these assertions you've made. If you have reproducible bugs, then please file them with explicit steps. But all this "I talked to a guy, and his brother said that it sometimes didn't work" stuff is really time consuming to chase. 

I've updated the wiki on panelization. Maybe your friend is having an issue because the panel origin isn't as expected??? I added a drawing that might help.

BendRocks, 2017-07-23:
Also, I'm not sure why there would be an issue with these fiducials. I cannot really understand the point you are trying to make. But I'm pretty certain that using any 2 of these fids would be fine.

cri-s, 2017-07-24:
sorry for have stolen you'r time.
The panel origin is not the problem, the problem is that a lot of
things don't works
as it should, and during checking the coordinates, i had to restart
openpnp several times
because of gui errors. No big deal hovewer as after restart the X and
Y was editable
again as example. If you state that all is ok, i surely don't want
start a flame or discussion
with you.
board origin 100x100
Origin=100x 100y
boards are x10 y20 spacing 6mm

+---------------+
|.+-+..+-+..+-+.|
|.|#|..|#|..|#|.|
|.|4|..|5|..|6|.|
|.o-+..o-+..o-+.|
|.+-+..+-+..+-+.|
|.|#|..|#|..|#|.|
|.|1|..|2|..|3|.|
|.o-+..o-+..o-+.|
O---------------+
^board 0 position

rotating 90 degree Clockwise
Origin now need to be 100x and 132Y, and yes,
this create problems if board zero moves,
especially if machine have automatic pcb transport.
I was not refering to that.

O------------+
|.o--+..o--+.|
|.|#1|..|#4|.|
|.+--+..+--+.|
|.o--+..o--+.|
|.|#2|..|#5|.|
|.+--+..+--+.|
|.o--+..o--+.|
|.|#3|..|#6|.|
|.+--+..+--+.|
+------------+
^board 0 position

top
boards nr 1 and 2
100 132
100 116

placement x=4 y=3 for board 1 and 2
103 128
103 112

bottom
board is same as top
placement x=4 y=3 for board 1 and 2
103 126
103 110

It seems ok. Nothing to complain
The problem hovewer is, when you mirror the board, ie assembly bottom
layer as coordinates above, real board is this:
+------------+
|.o--+..o--+.|
|.|#3|..|#6|.|
|.+--+..+--+.|
|.o--+..o--+.|
|.|#2|..|#5|.|
|.+--+..+--+.|
|.o--+..o--+.|
|.|#1|..|#4|.|
|.+--+..+--+.|
O------------+
^board 0 position

This is just one example where simulation and real differ.
the placement coincide from this example just because this is just
plain step and repeat,
if as example boards need to be rotated 90 deg for connector reasons,
the placement coincidence in that fictive case is missing.
As you don't have it, and only do simple step and repeat, and
communicate "why you would bother if the origin change" everything
seems ok.

Hovewer if one do something advanced, not just the plain step/repeat
as example excluding board 1 and 5, then the result is that four board
are populated only half side, including two that should not be
populated.

Further if you check again the coordinate and real position of boards
it should be clear that if you run some script or correct/learn
fiducials on real panel, the coordinate is wrong and
it depends on the thing you do if it run or produce errors.
If you do only single side assembly, everything is ok.

I don't go further with fiducials, because probably it is just a waste
of time for both of us.

Cri.S

This is just the simplest false assumption that this code use,
and for explaing it don't require in depth code review in contrast to
fiducials and i hope
the  is response is  explained enought. If not ask/tell me.

cri-s, 2017-07-24:
For fiducials, if single boards don't have fiducials, and panel have,
this actually don't work
during PnpJobProcessor run.
code:
   if (job.isUsingPanel() && job.getPanels().get(0).isCheckFiducials()){
                Panel p = job.getPanels().get(0);

                BoardLocation boardLocation = job.getBoardLocations().get(0);

                Location location = locator.locateBoard(boardLocation,
p.isCheckFiducials());
                boardLocationFiducialOverrides.put(boardLocation, location);
                Logger.debug("Panel Fiducial check for {}", boardLocation);
        }

        for (BoardLocation boardLocation : job.getBoardLocations()) {
            if (!boardLocation.isEnabled()) {
                continue;
            }
            if (!boardLocation.isCheckFiducials()) {
                continue;
            }
            Location location = locator.locateBoard(boardLocation);
            boardLocationFiducialOverrides.put(boardLocation, location);
            Logger.debug("Fiducial check for {}", boardLocation);
        }
    }

It have several issues, but you have right, if  before running
jobprocessor one manually
call the gui button of panel fiducials, then it works, just because
you have hardcoded
things and it ignore settings given from gui. and the fiducial
position must be found out
before doing panelize, otherwise the position is wrong because of
board position issue described on previous mail if doing bottom side.

Summary, it works when the bugs are known and full understood
permitting to using
workarounds. Without knowing the issues of failures, it  fails really badly.

2017-07-24 4:44 GMT+02:00, Cri S <phone.cri@gmail.com>:

cri-s, 2017-07-24:
During pnpJobProcessor, it checks the panel fiducials if enabled first.
It is applied only to the first board, not on all boards, and then it
checks the local
fiducials of board 1 again.

Seeing this behavior  or coding, there is really no issue here in
checking panel fiducials,
as there are not used here except for finding fiducials for first
board, but that is all.

I think this should be enought and now the failings are clear and at
the same time the
workarounds.

2017-07-24 12:44 GMT+02:00, Cri S <phone.cri@gmail.com>: