openpnp/openpnp #606: Z-axis movement calibration and an API request

terryjen, 2017-07-25:
# Feature Request 1
I'm working on a study in the Lab to use the openPNP system to put a resistor on it, and spray some glue to cover pins using another machine. I have two Feature Request, need your help. 
Feature Request 1:
Calibrate X, Y errors caused by Z movement.  Because of hardware problem, the Z-axis movement may cause X, Y errors. 
When moving Z, I found the center of the camera moves.

Describe the Feature
-----------------------------------------------------------------------
In machine settings, We can calibrate the errors of X, Y because of Z movement caused by machine.
So that pick and place movement can be more accurate, by using interpolation method. And when Z moves, X, Y errors can be calculated by interpolation method, and move X, Y automatically.

How Will It Look?
-----------------------------------------------------------------------
The interface may be similar to Machine Setup->Heads->ReferenceHead H1->Configruation->Location , with textboxes and buttons. 
There may be Z-min and Z-max, like this:
Z-min[Textbox]:     X[Textbox]  ,Y [ Textbox ]   and buttons   (a New Button to Add Interpolation)
Zi       [Textbox]:     X[Textbox]  ,Y [ Textbox ]   and buttons
Zi+1   [Textbox]:     X[Textbox]  ,Y [ Textbox ]   and buttons
Z-max[Textbox]:    X[Textbox]  ,Y [ Textbox ]    and buttons

From Z-min to Z-max, users can add interpolation.

Steps to Calibrate
-----------------------------------------------------------------------
1. Z-min, move camera center from home to a fiducial, we get the Xmin, Ymin.
2. Move Z-min to Zi, and move X, Y to let the camera center back to fiducial, we get Xi, Yi.
2. Move Zi to Zi+1, and move X, Y to let the camera center back to fiducial, we get Xi+1, Yi+1
3. Move Zi+1 to Z-max and move X, Y to let the camera center back to fiducial, we get Xmax, Ymax

Then, the program can calculate how much x, y to compensate the error when Z is between Z-min and Z-max. After applying this calibration, when we move Z-axis, the system will move X, Y automatically to correct the error. Ideally, in the center of camera won't move away through out Z-min to Z-max.

But it would be a problem if the camera is not totally vertical. so we'd better check it by using the smallest nozzle to pick and place several times to check. Or we can change the value manually to get better result.
We also tried to use a laser pen to scale up the distance it moved, we believe the accuracy would be better than other methods.

Why Do We Need It?
-----------------------------------------------------------------------
When using camera to calibrate the nozzle position to pick and place, errors may come out, especially when they are in different height levels, or they need to rotate.

# Feature Request 2
My Professor wants the Glue Spray system and OpenPNP to work together alternately. So I have to code a program to communicate between OpenPNP and Glue Spray system(.net based). All what I need to do on OpenPNP is to tell OpenPNP "Yes, go next Pick and Place.", and OpenPNP returns me "Done", and then I can tell Glue Spray system to work, and return to me after done spray, and call OpenPNP again to next step. 

It's not complex,but I'm only a primary user of visual studio.net, My knowledge about java is quite far from understanding your whole program.

I tried JavaScript, no information about how to control (Process one step of the Job and Pause) yellow button, and it seems JavaScript can not run independently.

I tried to download the source file of OpenPNP, and compile(build) it in IntelliJ Idea (I cannot import it to Eclipse). try to RUN this program, error: "org.codehaus.mojo:buildnumber-maven-plugin:1.3 or one of its dependencies could not be resolved." Tried many different ways from google.

I already find out that "fsm.send(Message.Step)" and "FiniteStateMachine" may control the yellow button to Process one step of the Job and Pause. So I'd like to ask that how can I use it like an API to control the process?  

Any advise will be appreciated. 
Thanks very much
Terry

cri-s, 2017-07-28:
it is normal that x/y moves if you change Z. OpenPnP uses always the same Z height if you configure
part height correctly and as consequence, no x/y offset is introduced.
Normally nozzle offset is measured without uplooking camera. It could be made meaning that nozzle 0
have offset 0 and camera have offset relative to nozzle 0. 
You must be aware that webcam is always 
not centered and as result offsets are valid only for the height used during calibration/setup.

vonnieda, 2017-07-30:
Please submit only one change / feature request per issue. Combining multiple requests in a single issue makes it difficult to track what is happening with each.

closed 2017-07-30 by vonnieda