openpnp/openpnp #607: UI Bug: In Parts list in the Job tab. Deleting items with "Part/Side/Type" selected overwrites values in the next item.

yaddatrance, 2017-07-27:
# Read This First
To submit an issue for OpenPnP you must use this template or it will be deleted. Please fill out either the Problem Report or Feature Request section below and delete the other section.

-----------------------------------------------------------------------

# Problem Report
## Describe the Issue

This is a "unexpected behavior" bug. Deleting an item in the Parts menu (under the Pick and Place tab of the Jobs tab) with the wrong element highlighted can cause the next item to inherit the first item's element.

Also seems to affect Side and Type columns.

Running on the development release of 2017-07-02_15_55-02.d43f075

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:

1. In the Job tab, load up a job with parts defined. 
2. Click on a line item using the "Part/Side/Type" column such that you see the 'v' 
3. If you click on the red X to delete the part, 
4. the deleted item's "Part" gets set as the active part for the next item in the list.

## Expected Result
Item should be deleted cleanly without affecting other items.

## Actual Result
What actually happened? Be descriptive.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue

Image 1: Selecting the item with the Part column in focus, in this case a battery
[image]

Image 2: Deleting with the red X ends up overwriting the Part of the next item. R4 incorrectly shows CR-2032
[image]

Whenever possible, please use the Menu -> Help -> Submit Diagnostics feature and include the URL here. Submit Diagnostics makes it easy to include log files, screen capture, debug images, configuration files, etc.

-----------------------------------------------------------------------

# Feature Request
## Describe the Feature
What do you want it to do? Be as descriptive as possible.

## How Will It Look?
If the feature will include changes to the user interface, how would they look? Include drawings or mockups if possible.

## Why Do We Need It?
1. What problem does it solve?
2. Is it useful for everyone, or does it only solve a problem for your machine?

## Provide Examples
Is this something that exists in other pick and place software? Provide examples and references if possible.

vonnieda, 2017-07-30:
Reproduced as described. I think this is probably similar to the bug in #597 and a similar fix might work.

vonnieda, 2019-02-21:
One of the recent UI fixes fixed this. Can no longer reproduce.

closed 2019-02-21 by vonnieda