openpnp/openpnp #613: Problem Report: First time movement after it connected or homed, timeout.

terryjen, 2017-07-31:
# Problem Report
## Describe the Issue
When I connect OpenPnP to the machine, and click move (X 0.1mm, for example) for the first time, it will move continuously until to the end. This happens after I homed, too.  And because I set X, Y home to max and Z home to min in smoothieboard, X, Y will move to min, and Z will not move.  After that, when I press to move the same axis, everything is fine.

## Steps to Reproduce
If you want to test this in your machine, try to do like this if you are using the smoothieboard:
1. Set in your machine configuration, alpha, beta, gamma "home_to_max"
2. Click the Power On button.
3. Home your machine.
4. Jog the machine 1mm to the left.
5. The machine start to move to more than 1mm, until to the end.  And "timeout error" comes out.
(Actually, I pressed emergency stop button when it moves over 1mm, so I didn't find out actually where it stopped.).
6. It happens sometimes when I connected to the machine, and start the first move, too.

## Expected Result
It should move for only 1mm.

## Actual Result
It moves much more than 1mm.

## Probably reason
I compared the log on OpenPnP with the console on pronterface, and found the difference between the command sent to machine. In pronterface, it sents G91 everytime before G0, and G90 after G0. I believe this is to change the set Relative/Absolute Positioning to the machine. And I tried this in OpenPnP as following steps:
1. Set in your machine configuration, alpha, beta, gamma "home_to_max"
2. Click the Power On button.
3. Home machine.
4. Goto->machine->driver->console-> send command"G91"
5. Jog the machine 1mm to the left.
6. The machine moves 1mm and stops.
Everything goes right!

## Suggestion
If OpenPnP command is all based on Relative Positioning system, OpenPnP can send a G91 command right after the machine started to solve this problem. But if OpenPnP can send G91 before G0 every time, it might be better, and it maybe doesn't matter if OpenPnP send G90 after G0.

terryjen, 2017-07-31:
Take care of the setting for "home_to_max", especially for Z-axis.

vonnieda, 2017-07-31:
OpenPnP uses absolute positioning for all movements. You need to do the following:

1. Add G90 to your Gcode Driver CONNECT_COMMAND. This will be sent after connect and will set the controller to absolute.
2. In your axis configuration in machine.xml, set the home-coordinate value to your max values as configured in Smoothie. For X, set the value you have in alpha_max, for Y beta_max, etc. This will tell OpenPnP what the machine's coordinates are after homing. Without this, after you home, it assumes that you are at 0,0,0,0 so a 1mm move will send a G0X1. Since your machine is actually at alpha_max, beta_max, etc. it needs to send G0X299 (for instance). 

If you attach your Smoothie config and machine.xml I can help further.

terryjen, 2017-07-31:
Thanks so much von Nieda! I will try right now, and report here later.

terryjen, 2017-07-31:
1. There is G90 in CONNECT_COMMAND, so no need to change.
2. My files are attached below.
Thanks so much!

[config.txt](https://github.com/openpnp/openpnp/files/1188942/config.txt)
[machine.xml.txt](https://github.com/openpnp/openpnp/files/1188943/machine.xml.txt)

vonnieda, 2017-07-31:
Here is the updated machine.xml. The lines that were changed are line 871 and 876. I changed home-coordinate to alpha_max and beta_max, respectively. 

[machine.xml.txt](https://github.com/openpnp/openpnp/files/1188960/machine.xml.txt)

Try this file and let me know if it works better. If not, please include your OpenPnP.log file so I can see what is happening.

terryjen, 2017-08-01:
Thanks Mr. Jason von Nieda, it works much better now.
a small problem is, after home, when I press the first move, it won't move, for X, Y & Z, and after that, it works fine.
Here is the log file only with 3 actions, step 1: Home; step 2, move X249; Step 3. move X248.
step 2 will not move.
Is it because the settings below? it is already in 249 after home?
_alpha_homing_retract_mm                         1                # distance in mm
beta_homing_retract_mm                           1                
gamma_homing_retract_mm                      1_                

[Log file.txt](https://github.com/openpnp/openpnp/files/1190518/Log.file.txt)

Thanks!

vonnieda, 2017-08-01:
Your controller appears to be responding normally. I'd suggest looking at the Gcodes that OpenPnP is sending in the log and typing those into Pronterface. See if it behaves the same, and if so, try to figure out what is happening. I would also suggest increasing the retract to 5mm to see if that helps. It could be that 1mm is not enough to release the switch and it still thinks it's on the endstop.

terryjen, 2017-08-01:
Thanks Mr. Jason von Nieda for your reply.
I think I know where the problem is now. 
1. In my setting, home is (250, 190, 0), it clearly shows in the **Green Status Bar** after home.
2. And I move X 1mm, that is G0 X249.  No movements, but the  **Green Status Bar** shows (249, 190, 0)
3. It seems because of retract after home, the actual position of the system is (249,189,1).
4. And if I move G0 X248 for the first step, it works fine. 
5. When I set the retract distance to 5 mm. if I press X.Left button for 1mm after home, it moves to right. because the command is G0 X249, but the actual position is X245.

So, it is not a big problem. I'm not sure if it is a problem that when actual position after home is (245,185,5), and OpenPnP knows this position, but it shows in  **Green Status Bar** is (250,190, 0).

And it seems that if I send command in the console, such as G0 X200,  the machine will move, but the coordinates in the  **Green Status Bar** will not change, this might be a small bug.
Ideally, **Green Status Bar** shows coordinates actually where it is might be better.

Thanks again!
Terry

vonnieda, 2017-08-01:
Retract should not affect home position. You can verify this by sending M114 to see where Smoothie thinks it is. It should always be at alpha_max, beta_max, etc. after homing, no matter what the retract is set to.

The green status bar (DROs) just show what OpenPnP thinks the controller's position is. It doesn't query the controller. That's why you have to set the home-coordinate. This is how you tell OpenPnP where it is after homing. 

So, another way to make sure it's right is to get your Smoothie settings how you want them, then in Pronterface do home, then M114. The values from M114 are what you should put in the home-coordinate values in machine.xml. Then everything will line up.

closed 2017-08-01 by vonnieda