openpnp/openpnp #614: A small suggestion about Timeout

terryjen, 2017-08-01:
# Problem Report
## Describe the Issue
The power of the stepper motor of PnP system that I'm working on is small, so the movement speed is low. when I press to move for 100mm, everything is good, except there will be an error :
_Timeout waiting for response to M400; Wait for moves to complete before returning._

## Steps to Reproduce
1. Machine Setup->Driver->GcodeDriver->General Settings->Command Timeout
2. Change the settings to a smaller number according to your system speed, for example, 1000ms.
3. Jog the machine 100mm to any direction.
4. The timeout error will comes out.

## Suggestion
I know we can set the timeout to a high enough data, such as 100000ms to prevent this error.
And I know we can delete the M400 in the Gcode-> MOVE_TO_COMMAND, thanks for your reply yesterday, ^^. 
But then I was wondering why we need this timeout function?
Based on that OpenPnP knows the speed of the machine and the movement distance, it can easily calculate how much time it may take for this movement. For example, move 100mm, the speed is 1000mm/min, it needs 6000ms,  and then give 10-1000ms tolerance, which we can set in the Command Timeout settings. 
In normal condition, there won't be any timeout error, only if there really has some error?
Will it makes timeout function more meaningful?

Thanks,
Terry

vonnieda, 2017-08-01:
OpenPnP cannot know the details of how long a move will take. Only the controller knows that. It has to take into account acceleration, linear interpolation, axis coordination, and any startup and stop delays involved. The best thing to do is just measure your longest transit time on your hardware and set your timeout slightly higher.

closed 2017-08-01 by vonnieda

terryjen, 2017-08-01:
Yes, it seems very complex when considering these matters. Thanks.